GPUOpen-Drivers/AMDVLK #3: SteamVR vrcompositor crash

ChristophHaag, 2017-12-23:
RX 480.

With some trash output removed that's the output of vrmonitor.sh
```
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/internalMemMgr.cpp:351:AllocateGpuMemNoAllocLock)
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/internalMemMgr.cpp:351:AllocateGpuMemNoAllocLock)
...
AMD-PAL: Error: Assertion failed: palResult == Pal::Result::Success | Reason: Unknown (/home/chris/oldhome/build/amdvlk-git-debug/src/xgl/icd/api/vk_image.cpp:794:CreatePresentableImage)
AMD-PAL: Error: Assertion failed: palResult == Pal::Result::Success | Reason: Unknown (/home/chris/oldhome/build/amdvlk-git-debug/src/xgl/icd/api/vk_image.cpp:803:CreatePresentableImage)
*** Error in `/home/chris/.local/share/Steam/SteamApps/common/SteamVR/bin/linux64/vrcompositor': malloc(): memory corruption: 0x000000000228ff70 ***
```
Backtrace:
```
Core was generated by `/home/chris/.local/share/Steam/SteamApps/common/SteamVR/bin/linux64/vrcomposito'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f0ce179a860 in raise () from /usr/lib/libc.so.6
[Current thread is 1 (Thread 0x7f0ce39ef780 (LWP 29097))]
(gdb) bt
#0  0x00007f0ce179a860 in raise () from /usr/lib/libc.so.6
#1  0x00007f0ce179bec9 in abort () from /usr/lib/libc.so.6
#2  0x00007f0ce17dd437 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007f0ce17e3d34 in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007f0ce17e6f8d in _int_malloc () from /usr/lib/libc.so.6
#5  0x00007f0ce17e960a in calloc () from /usr/lib/libc.so.6
#6  0x00007f0cd75fbbf4 in amdgpu_va_range_alloc (dev=0x178aa90, va_range_type=amdgpu_gpu_va_range_general, size=4194304, va_base_alignment=2097152, 
    va_base_required=0, va_base_allocated=0x7ffd98485180, va_range_handle=0x7ffd98485190, flags=0) at amdgpu_vamgr.c:260
#7  0x00007f0cd86dc925 in Pal::Linux::Device::AssignVirtualAddress (this=0x178ca10, pGpuMemory=0x228ff60, pGpuVirtAddr=0x7ffd98485220)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/os/lnx/lnxDevice.cpp:3048
#8  0x00007f0cd86e27d7 in Pal::Linux::GpuMemory::AllocateOrPinMemory (this=0x228ff60, baseVirtAddr=0, pPagingFence=0x0, 
    virtualAccessMode=Pal::VirtualGpuMemAccessMode::Undefined) at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/os/lnx/lnxGpuMemory.cpp:120
#9  0x00007f0cd86b6562 in Pal::GpuMemory::Init (this=0x228ff60, createInfo=..., internalInfo=...)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/gpuMemory.cpp:537
#10 0x00007f0cd86aebd5 in Pal::Device::CreateInternalGpuMemory (this=0x178ca10, createInfo=..., internalInfo=..., pPlacementAddr=0x228ff60, ppGpuMemory=0x7ffd98485530)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/device.cpp:2810
#11 0x00007f0cd86e4300 in Pal::Linux::Image::CreatePresentableMemoryObject (pDevice=0x178ca10, pImage=0x228faa8, pMemObjMem=0x228ff60, ppMemObjOut=0x7ffd984856f8)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/os/lnx/lnxImage.cpp:247
#12 0x00007f0cd86e3e92 in Pal::Linux::Image::CreatePresentableImage (pDevice=0x178ca10, createInfo=..., pImagePlacementAddr=0x228faa8, 
    pGpuMemoryPlacementAddr=0x228ff60, ppImage=0x7ffd984859a0, ppGpuMemory=0x7ffd984859c0)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/os/lnx/lnxImage.cpp:150
#13 0x00007f0cd86d9612 in Pal::Linux::Device::CreatePresentableImage (this=0x178ca10, createInfo=..., pImagePlacementAddr=0x228faa8, 
    pGpuMemoryPlacementAddr=0x228ff60, ppImage=0x7ffd984859a0, ppGpuMemory=0x7ffd984859c0)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/pal/src/core/os/lnx/lnxDevice.cpp:1415
#14 0x00007f0cd8251973 in vk::Image::CreatePresentableImage (pDevice=0x19592b8, pCreateInfo=0x7ffd98485c60, pAllocator=0x16eb268, imageUsageFlags=24,
    presentMode=Pal::PresentMode::Windowed, pImage=0x26b8088, imageFormat=VK_FORMAT_B8G8R8A8_SRGB, sharingMode=VK_SHARING_MODE_EXCLUSIVE, concurrentQueueFlags=1,
    pDeviceMemory=0x26b80a0) at /home/chris/oldhome/build/amdvlk-git-debug/src/xgl/icd/api/vk_image.cpp:848
#15 0x00007f0cd827b84c in vk::SwapChain::Create (pDevice=0x19592b8, pCreateInfo=0x7ffd98487030, pAllocator=0x16eb268, pSwapChain=0x7ffd984871c0)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/xgl/icd/api/vk_swapchain.cpp:299
#16 0x00007f0cd822372e in vk::Device::CreateSwapchain (this=0x19592b8, pCreateInfo=0x7ffd98487030, pAllocator=0x16eb268, pSwapChain=0x7ffd984871c0)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/xgl/icd/api/vk_device.cpp:1789
#17 0x00007f0cd822528b in vk::entry::vkCreateSwapchainKHR (device=0x19592b0, pCreateInfo=0x7ffd98487030, pAllocator=0x0, pSwapchain=0x7ffd984871c0)
    at /home/chris/oldhome/build/amdvlk-git-debug/src/xgl/icd/api/vk_device.cpp:2425
#18 0x00007f0ce2c329d4 in ?? () from /usr/lib/libvulkan.so.1
#19 0x0000000000474305 in vr::CHmdWindowSDL::CreateVulkanSwapChain(SDL_Window*, unsigned int, unsigned int, vr::CHmdWindowSDL::SwapChainData_t*, VkSurfaceKHR_T*, bool)
    ()
#20 0x0000000000475b9c in vr::CHmdWindowSDL::CHmdWindowSDL(vr::CHmdDistort*, int, int, unsigned int, unsigned int) ()
#21 0x0000000000448d66 in vr::CHmdDistort::Initialize(vr::CVRNotificationManager*, CVRSharedState*, void*, bool) ()
#22 0x000000000047bcce in CVRCompositor::Init() ()
#23 0x000000000047f3e5 in RealMain(unsigned int, ParsedArguments_t&) ()
#24 0x0000000000440cda in main ()
```

davidmaoamd, 2018-01-09:
@ChristophHaag 
Thanks for reporting this. 
I have debugged this a little bit. 
The vrcompositor did not use the correct usageFlag when creating the swapchain. 
We have informed Valve about this, and they will look into that.

The usageFlags getting from the vkGetPhysicalDeviceSurfaceCapabilitiesKHR is the union of all supported usage flags supported for any particular presentable format.
Application still need to call vkGetPhysicalDeviceImageFormatProperties  to get the exact supported usageFlags for the specific image format it choose to create the swapchain.

Btw: What steamvr version are you running? SteamVR(Main) or SteamVR?
Thanks.

ChristophHaag, 2018-01-09:
There is only one SteamVR publicly available. And of this one, the beta version

> SteamVR Version 2018-01-09 (1515459962)

ChristophHaag, 2018-02-27:
0eb739638710b51fd37b4d92ef12b75d259e646a makes the SteamVR compositor work, thanks. VR applications like hellovr_vulkan don't seem to work though and there are graphical issues like this corrupted texture on the Vive controller:
[image]

davidmaoamd, 2018-02-27:
Thanks for reporting this.
We will work on this asap. 
Btw: Besides the openvr examples, what other demos/games you would play with for VR use cases?

ChristophHaag, 2018-02-27:
There are only few available. A list collected by users is at https://steamcommunity.com/app/250820/discussions/5/133257959064016658/

Games that I can test:

Locomancer recently got an update with Unity 2017.3 which hangs the GPU on radv (latest git) so I was curious to see how it would work on another driver.

Serious Sam VR/The Talos Principle VR are Vulkan games.

Astrokill (not featured in the list yet) is an Unreal Engine 4 game that is frequently updated to the latest Unreal Engine, runs with OpenGL by default but also runs with Vulkan with the -vulkan switch.

High Fidelity, Doom 3 BFG Fully Possessed, Vivecraft, Polynomial 2 linux test branch, are all OpenGL applications so don't need to run under amdvlk themselves, they only need the external memory extensions to work.

davidmaoamd, 2018-03-01:
Thanks for the information! 
We will evaluate those applications one by one.

ChristophHaag, 2018-03-11:
A good start would probably be the hellovr_vulkan example from the openvr repository. It causes the kernel to spam

    Mär 12 00:00:32 c-pc kernel: [drm:amdgpu_cs_ioctl [amdgpu]] *ERROR* Failed in the dependencies handling -22!

into the dmesg log.

davidmaoamd, 2018-03-12:
Yes. The issue is reproduced in my side as well.
I am working on this.
Thanks.

jinjianrong, 2018-05-15:
@ChristophHaag we can't reproduce the issue now.  could you help confirm if the VR examples run well with latest driver?  Thanks

ChristophHaag, 2018-05-15:
No, I'm using the steamvr beta from today, just built amdvlk dev from today and vrcompositor doesn't start and/or crashes.
```
Core was generated by `/home/chris/.local/share/Steam/SteamApps/common/SteamVR/bin/linux64/vrcomposito'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f4506cd21e0 in __memcpy_ssse3 () from /usr/lib/libc.so.6
[Current thread is 1 (Thread 0x7f45094b7f00 (LWP 29365))]
(gdb) bt
#0  0x00007f4506cd21e0 in __memcpy_ssse3 () from /usr/lib/libc.so.6
#1  0x00007f45002852a5 in ?? () from /usr/lib/amdvlk64.so
#2  0x000000000050481f in CVulkanVRRenderer::AllocateMemory(VkDevice_T*, unsigned long, bool, bool, bool, VkImage_T*, VkBuffer_T*, unsigned int, unsigned int*) ()
#3  0x0000000000504acc in CVulkanVRRenderer::CreateBufferHelper(VkDevice_T*, int, unsigned int, VkDeviceMemory_T**, bool, unsigned long*, unsigned int*, unsigned int*) ()
#4  0x000000000050bd06 in CVulkanVRRenderer::Initialize(vr::IHmdWindow*, vr::IVRIPCResourceManagerClient*, VkInstance_T*, VkPhysicalDevice_T*, VkDevice_T*, VkQueue_T*, unsigned int, VkQueue_T*, unsigned int, vr::IVRResources*, vr::IVRCompositorSystemInternal*) ()
#5  0x000000000043b06b in CGenericResources::Init(vr::IGraphicsDevice::CreateStruct_t const&) ()
#6  0x0000000000412c5e in vr::CHmdDistort::Initialize(vr::CVRNotificationManager*, CVRSharedState*, vr::IVRIPCResourceManagerClient*, void*, bool) ()
#7  0x000000000044dea7 in CVRCompositor::Init() ()
#8  0x00000000004509dc in RealMain(unsigned int, ParsedArguments_t&) ()
#9  0x000000000040c29c in main ()
```

jinjianrong, 2018-07-24:
@ChristophHaag With latest steamVR beta and amdvlk, we are not seeing the issue any more. could you help confirm it? thanks

ChristophHaag, 2018-07-24:
I can confirm that SteamVR Compositor + hellovr_vulkan now work fine.

I saw you added VK_EXT_direct_mode_display but it doesn't seem work (tried with SteamVR beta).

What also doesn't work is SteamVR with OpenGL clients like hellovr_opengl.

If you have time, you can also try looking into running SteamVR with amdvlk and applications with radv, currently that results in a garbled image in SteamVR. Of course that should be a really low priority issue because nobody is going to seriously run it like this, I was just curious.

AnzhongHuang, 2018-07-27:
@ChristophHaag The extension VK_EXT_direct_mode_display depends on XRandR 1.6, it's probably released with XServer 1.2 or later. You can check the XRandR's version from the file /usr/include/xcb/randr.h,
  #define XCB_RANDR_MAJOR_VERSION 1
  #define XCB_RANDR_MINOR_VERSION 6
 
If you version is lower than 1.6, you can use the ppa https://launchpad.net/~kisak/+archive/ubuntu/steamvr to upgrade your system. To be honest, I only run the direct mode on this ppa.

ChristophHaag, 2018-07-27:
Yes, [libdrm 2.4.89, xcb-proto and libxcb 1.13, xorgproto 2018.3, xorg 1.20, and optionally xf86-video-amdgpu master](https://haagch.frickel.club/#!drmlease.md) :)

direct mode is running fine with radv, but with amdvlk the steamvr compositor was opened in the extended mode window last time I tried a couple of days ago. I'm not entirely sure but I thought steamvr *beta* might need VK_EXT_display_control before enabling direct mode? Not sure since SteamVR doesn't really believe in logging errors...

Right now with an amdvlk build from July 25 I get a GPU hang on my RX 480 whenever the steamvr compositor starts.

```
Jul 27 11:04:06 c-pc kernel: amdgpu 0000:08:00.0: GPU fault detected: 147 0x09f08401 for process vrcompositor pid 6333 thread vrcomposit:cs0 pid 6340
Jul 27 11:04:06 c-pc kernel: amdgpu 0000:08:00.0:   VM_CONTEXT1_PROTECTION_FAULT_ADDR   0x0871753E
Jul 27 11:04:06 c-pc kernel: amdgpu 0000:08:00.0:   VM_CONTEXT1_PROTECTION_FAULT_STATUS 0x06084001
Jul 27 11:04:06 c-pc kernel: amdgpu 0000:08:00.0: VM fault (0x01, vmid 3, pasid 32800) at page 141653310, read from 'TC7' (0x54433700) (132)
Jul 27 11:04:16 c-pc kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=25532, emitted seq=25534
```

AnzhongHuang, 2018-07-27:
Did you build the amdvlk by yourself, I suppose you have enabled this extension in amdvlk during building, since this extension is not enabled by default. 

How do you know the steamvr is working on the extended window, I mean on XServer 1.2, If you connected the VIVE HMD to a Ubuntu system, the HMD is not activated, you can see the LED is red on the HMD, until you use the command like   "xrandr --output HDMI-1 --mode 2160x1200 --right-of DP-1 " to active the HMD. So if the HMD is activated, it can't be a extended window, This behavior is guaranteed by OS instead of amdvlk. 

Can you confirm that: 1) the HMD is not activated at the beginned, if not, I think you need to upgrade the OS, 2) Is that extension enabled during driver building?

ChristophHaag, 2018-07-27:
The current build is not my own, it's from an archlinux repo that *should* use the same build config as this aur package: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=amdvlk-git#n43

So it should not use special build config options but in `vulkaninfo` output I see

    VK_EXT_direct_mode_display          : extension revision  1

so it must be enabled in the build...

> How do you know the steamvr is working on the extended window

Well, in extended mode the SteamVR compositor is just a fullscreen window. When the Vive display is activated, SteamVR automatically puts the window on this display, but if the Vive display is off, the window appears somewhere else, I think usually the leftmost monitor. Without activating the Vive display, I saw the SteamVR compositor with hellovr_vulkan working on my main monitor.

Unfortunately I can't test again right now because I get this GPU GPU fault and GPU hang immediately when starting SteamVR with amdvlk, which freezes the display and requires a reboot.

With radv it's working fine currently (with direct mode).

AnzhongHuang, 2018-07-27:
In amdvlk, the direct mode depends on both VK_EXT_DIRECT_MODE_DISPLAY and VK_EXT_ACQUIRE_XLIB_DISPLAY extensions. I think the released amdvlk driver doesn't enable the VK_EXT_ACQUIRE_XLIB_DISPLAY extension, since we thought this extension is based on XRandR1.6, but the mainstream Linux system doesn't support it, so we didn't enable it either. If you want to use direct mode with amdvlk, you have to rebuild the amdvlk driver and enable the two extensions.

gusthe, 2018-12-24:
I tried direct mode after building with the VK_EXT_ACQUIRE_XLIB_DISPLAY extension enabled to test steamVR, and direct mode still does not work. Is XRandR1.6 required to make it work? I also couldn't get extended mode to work either. The games seemed to run like normal but I would just get a black screen on the head set.

AnzhongHuang, 2018-12-25:
@gusthe, did you enable BUILD_XLIB_XRANDR_SUPPORT?
option(BUILD_XLIB_XRANDR_SUPPORT "Build Xlib with xrandr 1.6 support" OFF)

gusthe, 2018-12-25:
Yeah I enabled BUILD_XLIB_XRANDR_SUPPORT. vulkaninfo reports that I have.
`VK_EXT_acquire_xlib_display         : extension revision  1`
and
`VK_EXT_direct_mode_display          : extension revision  1`
I'm also on xorg 1.20, and direct mode works with radv.

Here is my vrcompositor log.
[vrcompositor.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2708844/vrcompositor.txt)

beniwtv, 2019-05-04:
This still does not seem to work - Steam doesn't think direct mode is available. Works on RADV.

JacobHeAMD, 2019-05-05:
Looks like "VK_EXT_display_surface_counter" is not enabled, please try  AMDVLK_ENABLE_DEVELOPING_EXT="VK_EXT_display_surface_counter".

beniwtv, 2019-05-05:
I originally tried with "all" as option to that variable - going with "VK_EXT_display_surface_counter" makes no difference however. vulkainfo does show this extension to be enabled, but weridly, Steam still refuses - I attached my vrcompositor log.

[vrcompositor.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/3145541/vrcompositor.txt)

JacobHeAMD, 2019-05-05:
Could you please debug it and find out why "Pal::Linux::Dri3WindowSystem::AcquireScreenAccess" is failed?

beniwtv, 2019-05-05:
Yes, this might be Steam not AMDVLK, trying to debug now to see what's up

yshui, 2019-05-29:
@JacobHeAMD It seems to get a BadCrtc from xserver.

(I dump the X traffic and searched for RandR CreateLease requests).

yshui, 2019-05-29:
It's requesting a lease with a crtc that is 0.

yshui, 2019-05-30:
The latency of AMDVLK is much worse than radv :'(

ChristophHaag, 2019-06-24:
Finally I was able to build amdvlk again.

Unfortunatels in vrcompositor.txt I get `Direct mode features: not present` even though all the extensions seem to be provided. Maybe I can test later with our OpenXR runtime (Monado) since it's open source and has working direct mode on radv.

```
Mon Jun 24 2019 13:33:28.815790 - Instance extensions available:
Mon Jun 24 2019 13:33:28.815827 -  - VK_KHR_surface, version 25
Mon Jun 24 2019 13:33:28.815839 -  - VK_KHR_xcb_surface, version 6
Mon Jun 24 2019 13:33:28.815848 -  - VK_KHR_xlib_surface, version 6
Mon Jun 24 2019 13:33:28.815857 -  - VK_KHR_wayland_surface, version 6
Mon Jun 24 2019 13:33:28.815865 -  - VK_KHR_get_physical_device_properties2, version 1
Mon Jun 24 2019 13:33:28.815873 -  - VK_KHR_get_surface_capabilities2, version 1
Mon Jun 24 2019 13:33:28.815881 -  - VK_KHR_external_memory_capabilities, version 1
Mon Jun 24 2019 13:33:28.815888 -  - VK_KHR_device_group_creation, version 1
Mon Jun 24 2019 13:33:28.815897 -  - VK_KHR_external_semaphore_capabilities, version 1
Mon Jun 24 2019 13:33:28.815904 -  - VK_KHR_external_fence_capabilities, version 1
Mon Jun 24 2019 13:33:28.815912 -  - VK_EXT_debug_report, version 9
Mon Jun 24 2019 13:33:28.815920 -  - VK_KHR_display, version 21
Mon Jun 24 2019 13:33:28.815928 -  - VK_EXT_acquire_xlib_display, version 1
Mon Jun 24 2019 13:33:28.815935 -  - VK_EXT_direct_mode_display, version 1
Mon Jun 24 2019 13:33:28.815943 -  - VK_KHR_get_display_properties2, version 1
Mon Jun 24 2019 13:33:28.815950 -  - VK_EXT_debug_utils, version 1
Mon Jun 24 2019 13:33:28.815968 - Direct mode features: not present
Mon Jun 24 2019 13:33:28.815977 - Requesting 8 instance extensions:
Mon Jun 24 2019 13:33:28.815985 -  - VK_KHR_surface
Mon Jun 24 2019 13:33:28.815992 -  - VK_KHR_xcb_surface
Mon Jun 24 2019 13:33:28.815999 -  - VK_KHR_display
Mon Jun 24 2019 13:33:28.816007 -  - VK_EXT_direct_mode_display
Mon Jun 24 2019 13:33:28.816014 -  - VK_EXT_acquire_xlib_display
Mon Jun 24 2019 13:33:28.816021 -  - VK_KHR_external_memory_capabilities
Mon Jun 24 2019 13:33:28.816028 -  - VK_KHR_external_semaphore_capabilities
Mon Jun 24 2019 13:33:28.816036 -  - VK_KHR_get_physical_device_properties2
Mon Jun 24 2019 13:33:28.830228 - HMD deviceUUID is a
Mon Jun 24 2019 13:33:28.830264 - CHmdWindowSDL: Failed to create direct mode surface
```

So testing with extended mode, hellovr_vulkan works fine.

hellovr_opengl however looks very... interesting.
https://youtu.be/Y6cc3nE-qs4
That's with radeonsi from latest mesa git as OpenGL driver on a Vega 64.

yshui, 2019-06-24:
@ChristophHaag

> Direct mode features: not present

That is new. That wasn't the error previously.

ChristophHaag, 2019-06-24:
I can confirm amdvlk's direct mode implementation generally works.

You can try it with those similar open source code bases:

https://gitlab.com/lubosz/xrgears is an older demo that can run in direct mode on the first available non-desktop output, but this demo is not very actively maintained. Right now it renders completely white for me on both radv and amdvlk, but it does demonstrate that direct mode works.

https://gitlab.freedesktop.org/monado/monado is an OpenXR runtime featuring a Vulkan compositor with direct mode, reusing some code from xrgears, but the runtime is actively maintained and developed.
From the OpenXR-SDK I tried `hello_xr -g Vulkan`, which runs okay-ish in direct mode with some rendering glitches (might be the fault of hello_xr or the compositor).

So it seems the issue has to do with specifically how SteamVR does direct mode.

JacobHeAMD, 2020-07-16:
I assume this issue has been fixed by https://github.com/GPUOpen-Drivers/pal/pull/29

closed 2020-07-16 by JacobHeAMD