GPUOpen-Drivers/AMDVLK #4: Reversed Z-buffer not working

mikedilger, 2018-01-04:
This was on ArchLInux, with an R7 360.  I haven't tried other cards or OS builds.

Using the latest code from the amdvlk-git AUR repo for ArchLinux.

I have tried other drivers (vulkan-radeon) where my code works fine.

I would provide code but its in rust and might not be your cup of tea.  The code is very straightfoward, two piercing triangles.  In reversing the z-buffer, the following changes were flipped from 1.0 to 0.0 or the opposite:   viewport min_depth and max_depth;  clear value; depth_compare_op.  LunarG validation reports no issues.

Oh, I also get a SIGSEGV on exit (gdb seems to think it is in libpthread) which doesn't happen with the other drivers.

procedural, 2018-01-04:
I think it would be helpful if you could compile a minimal repro executable in Rust with debug symbols enabled and share it here...

mikedilger, 2018-01-04:
I've stripped out everything easy to strip out.  Without rewriting it carefully to be minimal, it is still 41MB.

I am on a different machine today with a different graphics card vendor, so I can't verify that this doesn't work on the AMD card, but I will check later today (if the power is restored to my home).

[siege-client.gz](https://github.com/GPUOpen-Drivers/AMDVLK/files/1604666/siege-client.gz)

procedural, 2018-01-05:
Is this what I should see?

[image]

A desired result would be the white triangle on top?

procedural, 2018-01-05:
[image]

Crash happens after this instruction, check line `154` of `main.rs`, would be nice if you could share the source code too, if you can.

procedural, 2018-01-05:
Vulkan trace calls of the application, if they can be helpful: https://pastebin.com/raw/KVUeTTyB

procedural, 2018-01-05:
I see no vkCmdSetViewport calls in the trace log...

procedural, 2018-01-05:
I guess viewport information is passed with `vkCreateGraphicsPipelines`...

procedural, 2018-01-05:
Trace file generated from `vktrace` compiled from https://github.com/LunarG/VulkanTools/commit/6620d66, if someone wants to debug it outside of Linux (if it's portable enough): https://mega.nz/#!YcZXWCjK!7ZNOMARxGeAaD46kzlvk25DWb4cSdsWYjzWm9wM9rW8

procedural, 2018-01-05:
Ok, so your application calls `vkCreateGraphicsPipelines` two times which both set `minDepth` to `1` and `maxDepth` to `0`...

[image]

mikedilger, 2018-01-06:
It's not efficient of me to call vkCreateGraphicsPipelines multiple times, but that should be ok.

min_depth to 1 and max_depth to 0 is what I intend, that's the reversed z-buffer. Clear value is 0 (max depth) and compare_op is greater_or_equal. It should behave very similar to a normal z-buffer configuration, except to give better floating point resolution at the far plane (since that is nearer to 0.0)

source code:
[siege-client-amdvlk.tar.gz](https://github.com/GPUOpen-Drivers/AMDVLK/files/1609371/siege-client-amdvlk.tar.gz)

Output on other drivers:
[image]

DominikWitczakAMD, 2018-01-08:
We'd appreciate if you could confirm you have run the application with validation enabled and no errors are being reported by any of the validation layers. Thanks.

shadiramadan, 2018-01-08:
Hi @DominikWitczakAMD, I believe @mikedilger stated in his first post that:
>  LunarG validation reports no issues.

But a confirmation would help clear things up.

mikedilger, 2018-01-09:
Yes that's right @DominikWitczakAMD @shadiramadan the validation layer does not report any errors (and I have gotten errors from it before so I believe it would report if it had anything to say).  I'm on a different computer in a different office right now, but when I get back to the AMD card I can confirm again.

procedural, 2018-01-09:
I can confirm too, no validation errors.

JacobHeAMD, 2018-01-10:
Hi @mikedilger, the depth issue is already fixed by the latest drop. Could you please try again with the latest AMDVLK driver?
I'm debugging the SIGSEGV issue.

mikedilger, 2018-01-10:
@JacobHeAMD confirmed!  The depth issue is resolved.

JacobHeAMD, 2018-01-10:
Hi @mikedilger , there is an issue of the application. The memory is allocated for two images, however, the alignment is got according to the first image. And the memoffset for vkBindImageMemory of the second image is not aligned to the alignment required by the second image. And vkBindImageMemory of the second image is failed as a result.
vkBindImageMemory(device = 0x7f8058caf000, image = 0x7f80583d2200, memory = 0x7f80578b9c00, memoryOffset = **2707456**)
memoryOffset should be aligned to 0x8000 (which is returned by vkvkGetImageMemoryRequirements of the second image).

procedural, 2018-01-10:
Can confirm too, the issue with depth is solved.

[image]

JacobHeAMD, 2018-01-11:
Hi @mikedilger, the SIGSEGV issue I mentioned (vkBindImageMemory) before is different from the issue you reported. And it only happens to the "vkreplay -o trace.vktrace". There is no this issue with siege-client. So please ignore it.

JacobHeAMD, 2018-01-11:
Hi @procedural, thanks for your confirmation.

mikedilger, 2018-01-11:
Ok so I take it the memory alignment of siege-client was ok.  In particular, from the pastebin trace:
* alignment returned was 4096 (0x1000)
* offset of 2707456 is ok (661 * 4096)
I am then under the impression that under "vkreplay" the card (possibly a different card) requested a different (coarser) alignment, but the replay ignored that and replayed faithfully, giving a SEGV... so not an actual bug.

mikedilger, 2018-01-12:
Side note: I'm kind of an idiot.  I was using a D16_UNorm depth buffer, not a floating point based one. Reversing the Z without using actual floating point values is pointless.  Anyhow, something to laugh at.

JacobHeAMD, 2018-01-22:
Another internal ticket is already created for the segfault issue.

closed 2018-01-22 by JacobHeAMD