GPUOpen-Drivers/AMDVLK #5: Installation instructions recommend nuking all Vulkan ICD files

mdiluz, 2018-01-08:
See [here](https://github.com/GPUOpen-Drivers/AMDVLK#remove-previous-installed-json-files).

This is destructive and could cause users to unknowingly hurt other driver installations on the same system. In particular, this will remove RADV's icd file and require manual fixing from the user.

Perhaps a change to the below commands would be safer:

### Remove Previous Installed JSON Files
```
sudo rm /etc/vulkan/icd.d/amd_icd*.json
sudo rm /usr/share/vulkan/icd.d/amd_icd*.json
```

Or simply re-working the advice to reflect the risks.

JacobHeAMD, 2018-01-09:
Hi @mdiluz , thanks for finding this. Will change it.

JacobHeAMD, 2018-01-09:
See [10c8459](https://github.com/GPUOpen-Drivers/AMDVLK/commit/10c8459db9ebac2c93e06f152957035f897324ea)

closed 2018-01-09 by JacobHeAMD

mdiluz, 2018-01-09:
Thanks Jacob.

For what it's worth - with this change it still recommends removing the RADV icd file, which can happily live side by side with AMDVLK. Though, it's worth noting, not all games will be able to switch between the two effectively and will simply pick one.

jinjianrong, 2018-01-09:
Hi Marc, the problem is  that there will be conflict if two JSON files co-exist in the system for  GPUs from same vendor.

mdiluz, 2018-01-09:
I think that's fair enough for the general user.

It's worth noting that the conflict appears to be entirely superficial - in testing we've found no fundamental issues with having both json files there, vkEnumeratePhysicalDevices simply returns two devices, and the program can choose whichever one it wants. It's actually really useful to have them both installed.