GPUOpen-Drivers/AMDVLK #7: PatchDescriptorLoad segfault when compiling SPIR-V binary

alan-baker, 2018-02-01:
Assembled the following shader using SPIR-V Tools (attached):
```
               OpCapability Shader
         %1 = OpExtInstImport "GLSL.std.450"
              OpMemoryModel Logical GLSL450
              OpEntryPoint GLCompute %main "main"
              OpExecutionMode %main LocalSize 1 1 1
              OpSource GLSL 450
              OpSourceExtension "GL_GOOGLE_cpp_style_line_directive"
              OpSourceExtension "GL_GOOGLE_include_directive"
              OpName %main "main"
              OpName %B "B"
              OpMemberName %B 0 "x"
              OpName %Buf "Buf"
              OpMemberDecorate %B 0 Offset 0
              OpDecorate %B BufferBlock
              OpDecorate %Buf DescriptorSet 0
              OpDecorate %Buf Binding 1
      %void = OpTypeVoid
         %3 = OpTypeFunction %void
     %float = OpTypeFloat 32
         %B = OpTypeStruct %float
%_ptr_Uniform_B = OpTypePointer Uniform %B
       %Buf = OpVariable %_ptr_Uniform_B Uniform
       %int = OpTypeInt 32 1
     %int_0 = OpConstant %int 0
   %float_0 = OpConstant %float 0
%_ptr_Uniform_float = OpTypePointer Uniform %float
      %main = OpFunction %void None %3
         %5 = OpLabel
        %14 = OpAccessChain %_ptr_Uniform_float %Buf %int_0
              OpStore %14 %float_0
              OpReturn
              OpFunctionEnd
```
[glsl_store.spv.gz](https://github.com/GPUOpen-Drivers/AMDVLK/files/1686785/glsl_store.spv.gz)

Compilation succeeds when starting from the assembly, but segfaults when compiling from the assembled binary. The crash stacktrace follows:

```
Stack dump:
0.	Program arguments: vulkandriver/drivers/xgl/builds/Debug64/llpc/amdllpc glsl_store.spv -S -val -o glsl_store.s 
1.	Running pass 'Patch LLVM for descriptor load operations' on module ''.
#0 0x00005602515d0844 llvm::sys::PrintStackTrace(llvm::raw_ostream&) vulkandriver/drivers/llvm/lib/Support/Unix/Signals.inc:398:0
#1 0x00005602515d08d7 PrintStackTraceSignalHandler(void*) vulkandriver/drivers/llvm/lib/Support/Unix/Signals.inc:462:0
#2 0x00005602515ced9e llvm::sys::RunSignalHandlers() vulkandriver/drivers/llvm/lib/Support/Signals.cpp:49:0
#3 0x00005602515d01c8 SignalHandler(int) vulkandriver/drivers/llvm/lib/Support/Unix/Signals.inc:252:0
#4 0x00007f3dfad470c0 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x110c0)
#5 0x000056025054619e llvm::Value::getType() const vulkandriver/drivers/llvm/include/llvm/IR/Value.h:245:0
#6 0x000056025054674f llvm::GetElementPtrInst::Create(llvm::Type*, llvm::Value*, llvm::ArrayRef<llvm::Value*>, llvm::Twine const&, llvm::Instruction*) vulkandriver/drivers/llvm/include/llvm/IR/Instructions.h:870:0
#7 0x00005602505dd162 Llpc::PatchDescriptorLoad::visitCallInst(llvm::CallInst&) vulkandriver/drivers/xgl/icd/api/llpc/patch/llpcPatchDescriptorLoad.cpp:457:0
#8 0x00005602505e14f6 llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::delegateCallInst(llvm::CallInst&) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:285:0
#9 0x00005602505e0b2d llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visitCall(llvm::CallInst&) vulkandriver/drivers/llvm/include/llvm/IR/Instruction.def:190:0
#10 0x00005602505e002f llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit(llvm::Instruction&) vulkandriver/drivers/llvm/include/llvm/IR/Instruction.def:190:0
#11 0x00005602505df774 void llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit<llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>, false, false> >(llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>, false, false>, llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>, false, false>) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:91:0
#12 0x00005602505df180 llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit(llvm::BasicBlock&) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:108:0
#13 0x00005602505debf8 void llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit<llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::BasicBlock, true, false, void>, false, false> >(llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::BasicBlock, true, false, void>, false, false>, llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::BasicBlock, true, false, void>, false, false>) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:91:0
#14 0x00005602505de5a0 llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit(llvm::Function&) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:104:0
#15 0x00005602505de26c void llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit<llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Function, true, false, void>, false, false> >(llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Function, true, false, void>, false, false>, llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Function, true, false, void>, false, false>) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:91:0
#16 0x00005602505ddea2 llvm::InstVisitor<Llpc::PatchDescriptorLoad, void>::visit(llvm::Module&) vulkandriver/drivers/llvm/include/llvm/IR/InstVisitor.h:100:0
#17 0x00005602505db120 Llpc::PatchDescriptorLoad::runOnModule(llvm::Module&) vulkandriver/drivers/xgl/icd/api/llpc/patch/llpcPatchDescriptorLoad.cpp:71:0
#18 0x0000560252237177 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) vulkandriver/drivers/llvm/lib/IR/LegacyPassManager.cpp:1591:0
#19 0x000056025223782b llvm::legacy::PassManagerImpl::run(llvm::Module&) vulkandriver/drivers/llvm/lib/IR/LegacyPassManager.cpp:1694:0
#20 0x0000560252237a23 llvm::legacy::PassManager::run(llvm::Module&) vulkandriver/drivers/llvm/lib/IR/LegacyPassManager.cpp:1726:0
#21 0x00005602505dac06 Llpc::Patch::Run(llvm::Module*) vulkandriver/drivers/xgl/icd/api/llpc/patch/llpcPatch.cpp:159:0
#22 0x000056025052d215 Llpc::Compiler::BuildComputePipeline(Llpc::ComputePipelineBuildInfo const*, Llpc::ComputePipelineBuildOut*) vulkandriver/drivers/xgl/icd/api/llpc/context/llpcCompiler.cpp:1211:0
#23 0x00005602505247bd BuildPipeline(Llpc::ICompiler*, CompileInfo*) vulkandriver/drivers/xgl/icd/api/llpc/tool/amdllpc.cpp:851:0
#24 0x0000560250525920 main vulkandriver/drivers/xgl/icd/api/llpc/tool/amdllpc.cpp:1189:0
#25 0x00007f3dfa49c2b1 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x202b1)
#26 0x000056025052262a _start (vulkandriver/drivers/xgl/builds/Debug64/llpc/amdllpc+0x18262a)
Segmentation fault
```

alan-baker, 2018-02-01:
I was incorrect about succeeding when compiling the assembly. The compiler exits with a successful return code, but the output is meaningless. The debug print outs indicate a Fragment shader is being compiled and the final disassembly is a nop.

linqunzju, 2018-02-02:
It seems you use standalone compiler. please try to add option -auto-layout-desc.

alan-baker, 2018-02-02:
Thank you that appears to fix the issue.

closed 2018-02-02 by alan-baker