GPUOpen-Drivers/AMDVLK #8: Request for adding 2 Vulkan extensions: variable pointers, conservative rasterization..

oscarbg, 2018-02-03:
Hi,
don't know if it's the correct place to ask for new Vulkan extensions to be supported on AMD drivers like AMDVLK but anyway I will ask here:
first is VK_KHR_variable_pointers I ask because is needed by clspv tool for converting OpenCL kernels to Vulkan compute shaders and is supported almost everywhere (NV drviers, Intel ANV,Adreno) (even by AMD RADV driver):
http://vulkan.gpuinfo.org/listreports.php?extension=VK_KHR_variable_pointers
second is VK_EXT_conservative_rasterization a new ext. reléased last month.. 
supported right now by Nvidia dev driver and also seen this string on Intel HD Graphics Vulkan  Windows driver so seems support is coming soon for it..
it's a D3D12 optional feature so makes sense to expose also to Vulkan devs.. should be good to VKD3D project devs eventually (a D3D12 Vulkan wrapper)..

jinjianrong, 2018-02-09:
Thanks for the request. We added it into our TODO list

oscarbg, 2018-10-25:
closing it.. these two are supported..

closed 2018-10-25 by oscarbg