GPUOpen-Drivers/AMDVLK #10: DOOM 2016 triggers an assertion

notaz, 2018-02-04:
DOOM 2016 (when ran through wine) triggers an assertion:
```
amdvlk/drivers/llvm/lib/IR/Instructions.cpp:1942: void llvm::ExtractValueInst::init(llvm::ArrayRef<unsigned int>, const llvm::Twine&): Assertion `!Idxs.empty() && "ExtractValueInst must have at least one index"' failed.
Aborted
```

The issue can be also reproduced using this renderdoc capture:
https://drive.google.com/file/d/1pfyFQFBdYvQ2LFBoqwMyOQbfhsDMmMNB/view?usp=sharing

GPU: POLARIS10 (rx470)

notaz, 2018-05-11:
I've noticed this has been marked "fixed", but the latest version of the game still doesn't run, with current code drop it fails with
"LLVM ERROR: Unsupported calling convention."

The game works with radv and -pro drivers. I can do another renderdoc capture, if that helps.

jinjianrong, 2018-05-14:
The original issue reported here (assertion) should have been fixed. There are other issues with running Doom on Wine: https://github.com/GPUOpen-Drivers/xgl/issues/24.  we are working on it.

closed 2018-09-04 by jinjianrong

massatt212, 2019-06-14:
Doom doesnt enter the menu after loads up to 100%
works on mesa not on AMDVLK

VileLasagna, 2019-08-01:
Bumping to confirm I'm experiencing the same issue as @massatt212 as from a few days ago. Was running fine but last updates in arch (mesa to 19.1.3, linux to 5.2) seem to have brought this on. RADV works on my machine but it's weirdly really slow (can't maintain 30FPS) I don't know why and it leads me to suspect other things on my machine, but I was having a great time with AMDVLK until this came about (except on VR, for which I'm looking at #3 to see if I can gleam anything)

I'm happy to try and help you get more info, @jinjianrong , I'm running a Radeon VII on Arch (everything up to date). Should this issue be reopened or a new one filed?

jinjianrong, 2019-08-02:
@VileLasagna do you mean you didn't see the issue before last updates in arch (mesa to 19.1.3, linux to 5.2) using same amdvlk driver?

VileLasagna, 2019-08-02:
@jinjianrong So the main issue of the thread is a crash on DOOM, that is allegedly fixed and, while I exeperienced a few crashes, those were very infrequent.
Massat brings up a different issue which I started to experience as well, also in DOOM:
The application loads normally but graphics and input freeze once the menu is 100% loaded and tries to come up.

This is happening 100% of the time for me as of between one and two weeks ago.
And yeah, I was running it normally up to that point.