GPUOpen-Drivers/AMDVLK #11: Patches to run build and run on Xorg and Xwayland with kernel 4.15.0 and mesa 17.3.2

FelixBellaby, 2018-02-07:
The attached patches allow the amdvlk.so driver to build and run on both the Xorg and Xwayland platforms.

I have only tested these patches on a Ryzen system with a single Vega card.

I tested with gcc 7.2.0, the mainline 4.15.0 kernel and Mesa 17.3.2 (Ubuntu 17.50 + ppa:paulo-miguel-dias/mesa + ppa:kernel-ppa/ppa) and the VulkanSDK 1.0.65.0. The examples at https://github.com/SaschaWillems/Vulkan run correctly (excluding those using geom or tess shaders).

~~~

The link command from icd/CMakeFiles/xgl.dir/link.txt fails on gcc-7.2, because the libpal.a archive appears without the --whole-archive flag BEFORE appearing with the --whole-archive flag. The attached
patch reverses the order of these link arguments to allow the build to complete without losing the link time optimisation:

[cmakelists patch.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/1705181/cmakelists.patch.txt)

~~~

The amdvlk64.so driver segfaults when run on a wayland session, because the Xwayland server does not support DRI2. This segfault can be avoided by replacing the DRI2 calls with DRM calls. The attached  patch adds necessary DRM functions to the drmLoader.proc file so that the  drmLoader.{h,cpp} can be rebuilt using the drmLoader.py script:

[drmloader patch.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/1705166/drmloader.patch.txt)

The attached  patch alters the OpenDri3() function to use these DRM calls:

[opendri3 patch.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/1705168/opendri3.patch.txt) 

~~~

The amdvlk.so driver does not recognise when the DRI3 and DRM devices connect to the same GPU under Xwayland, because the drmGetNodeTypeFromFd(fd) and drmGetBusid(fd) functions are broken in this case. However, the attached patch fixes this problem by simplifying the IsSameGpu() function:

[IsSameGpu patch.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/1705175/IsSameGpu.patch.txt)

JacobHeAMD, 2018-02-08:
For the issue of IsSameGpu, it was fixed internally (not in github yet).
For other patches, would you please send a pull request per patch?

JacobHeAMD, 2018-02-08:
Hi @FelixBellaby, I just found that the issue of Xwayland was already fixed internally with another solution. The fix will be included in next drop of master branch. Regarding to the GCC7 issue, there was already a workaround in dev branch.  Master branch will have it in next drop. So the pull requests are not necessary.

FelixBellaby, 2018-02-08:
Good to know that you are addressing compatibility with Xwayland. Increasing the range of supported platforms should pull in more contributions to the code base.
I will look to submit patches as pull requests against the dev branch.

JacobHeAMD, 2018-03-01:
Hi @FelixBellaby , all of the patches get in dev and master branch now. However, looks like your solution for GCC7 issue is better than the workaround. Would you mind send a pull request for this patch (cmakelists patch.txt)?

closed 2018-03-06 by JacobHeAMD