GPUOpen-Drivers/AMDVLK #12: BUG ASUS N551zu

APakrohk, 2018-02-11:
Hardware is wrong.
https://user-images.githubusercontent.com/27810360/35762947-81f65142-08b6-11e8-8f13-6f9c4d52a59a.png
--------------
Asus website
https://www.asus.com/Laptops/N551ZU/
--------------
lspci -mm

`00:00.0 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Root Complex" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Root Complex" 00:00.2 "IOMMU" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) I/O Memory Management Unit" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) I/O Memory Management Unit" 00:01.0 "VGA compatible controller" "Advanced Micro Devices, Inc. [AMD/ATI]" "Kaveri [Radeon R7 Graphics]" "ASUSTeK Computer Inc." "Kaveri [Radeon R7 Graphics]" 00:01.1 "Audio device" "Advanced Micro Devices, Inc. [AMD/ATI]" "Kaveri HDMI/DP Audio Controller" "ASUSTeK Computer Inc." "Kaveri HDMI/DP Audio Controller" 00:02.0 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Device 1424" "" "" 00:02.1 "PCI bridge" "Advanced Micro Devices, Inc. [AMD]" "Device 1425" "" "" 00:03.0 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Device 1424" "" "" 00:03.1 "PCI bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Root Port" "" "" 00:03.2 "PCI bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Root Port" "" "" 00:04.0 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Device 1424" "" "" 00:10.0 "USB controller" "Advanced Micro Devices, Inc. [AMD]" "FCH USB XHCI Controller" -r09 -p30 "ASUSTeK Computer Inc." "FCH USB XHCI Controller" 00:10.1 "USB controller" "Advanced Micro Devices, Inc. [AMD]" "FCH USB XHCI Controller" -r09 -p30 "ASUSTeK Computer Inc." "FCH USB XHCI Controller" 00:11.0 "SATA controller" "Advanced Micro Devices, Inc. [AMD]" "FCH SATA Controller [IDE mode]" -r40 -p01 "ASUSTeK Computer Inc." "FCH SATA Controller [IDE mode]" 00:12.0 "USB controller" "Advanced Micro Devices, Inc. [AMD]" "FCH USB OHCI Controller" -r11 -p10 "ASUSTeK Computer Inc." "FCH USB OHCI Controller" 00:12.2 "USB controller" "Advanced Micro Devices, Inc. [AMD]" "FCH USB EHCI Controller" -r11 -p20 "ASUSTeK Computer Inc." "FCH USB EHCI Controller" 00:13.0 "USB controller" "Advanced Micro Devices, Inc. [AMD]" "FCH USB OHCI Controller" -r11 -p10 "ASUSTeK Computer Inc." "FCH USB OHCI Controller" 00:13.2 "USB controller" "Advanced Micro Devices, Inc. [AMD]" "FCH USB EHCI Controller" -r11 -p20 "ASUSTeK Computer Inc." "FCH USB EHCI Controller" 00:14.0 "SMBus" "Advanced Micro Devices, Inc. [AMD]" "FCH SMBus Controller" -r16 "ASUSTeK Computer Inc." "FCH SMBus Controller" 00:14.2 "Audio device" "Advanced Micro Devices, Inc. [AMD]" "FCH Azalia Controller" -r01 "ASUSTeK Computer Inc." "FCH Azalia Controller" 00:14.3 "ISA bridge" "Advanced Micro Devices, Inc. [AMD]" "FCH LPC Bridge" -r11 "ASUSTeK Computer Inc." "FCH LPC Bridge" 00:14.4 "PCI bridge" "Advanced Micro Devices, Inc. [AMD]" "FCH PCI Bridge" -r40 -p01 "" "" 00:18.0 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Function 0" "" "" 00:18.1 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Function 1" "" "" 00:18.2 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Function 2" "" "" 00:18.3 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Function 3" "" "" 00:18.4 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Function 4" "" "" 00:18.5 "Host bridge" "Advanced Micro Devices, Inc. [AMD]" "Family 15h (Models 30h-3fh) Processor Function 5" "" "" 01:00.0 "Display controller" "Advanced Micro Devices, Inc. [AMD/ATI]" "Bonaire PRO [Radeon R9 M270X]" -rff -pff "" "" 02:00.0 "Network controller" "Intel Corporation" "Wireless 7260" -rbb "Intel Corporation" "Dual Band Wireless-N 7260" 03:00.0 "Unassigned class [ff00]" "Realtek Semiconductor Co., Ltd." "RTL8411B PCI Express Card Reader" -r01 "ASUSTeK Computer Inc." "RTL8411B PCI Express Card Reader" 03:00.1 "Ethernet controller" "Realtek Semiconductor Co., Ltd." "RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller" -r12 "ASUSTeK Computer Inc." "RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller"`

---------------
**There is no problem in Windows**
---------------
Distribute me Xubuntu

davidmaoamd, 2018-02-11:
Thanks for reporting the issue.
The DID for M280X is 0x6646, and M270X is 0x6647.
I cannot find the DID information from the pasted log captured from "lspci -mm". 
Could you try to run the lspci again to see if you could get the device id?

APakrohk, 2018-02-12:
Thank you for answering
`00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) I/O Memory Management Unit
00:01.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Kaveri [Radeon R7 Graphics]
00:01.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Kaveri HDMI/DP Audio Controller
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1424
00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 1425
00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1424
00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Root Port
00:03.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Root Port
00:04.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1424
00:10.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB XHCI Controller (rev 09)
00:10.1 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB XHCI Controller (rev 09)
00:11.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [IDE mode] (rev 40)
00:12.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller (rev 11)
00:12.2 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller (rev 11)
00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller (rev 11)
00:13.2 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller (rev 11)
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 16)
00:14.2 Audio device: Advanced Micro Devices, Inc. [AMD] FCH Azalia Controller (rev 01)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 11)
00:14.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] FCH PCI Bridge (rev 40)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 5
01:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Bonaire PRO [Radeon R9 M270X] (rev ff)
02:00.0 Network controller: Intel Corporation Wireless 7260 (rev bb)
03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTL8411B PCI Express Card Reader (rev 01)
03:00.1 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 12)
`

davidmaoamd, 2018-02-12:
It's weird. I still cannot find the DID. 
Could you send me the log of vulkaninfo on the AMDVLK driver.

APakrohk, 2018-02-12:
I do not have enough information.
How should do I?
Help me

notaz, 2018-02-13:
You can run "lspci -nn" instead, should show the device id.

APakrohk, 2018-02-18:
**lspci -nn**
`00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Root Complex [1022:1422]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) I/O Memory Management Unit [1022:1423]
00:01.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Kaveri [Radeon R7 Graphics] [1002:130c]
00:01.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Kaveri HDMI/DP Audio Controller [1002:1308]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1424]
00:02.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:1425]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1424]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Root Port [1022:1426]
00:03.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Root Port [1022:1426]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1424]
00:10.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB XHCI Controller [1022:7814] (rev 09)
00:10.1 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB XHCI Controller [1022:7814] (rev 09)
00:11.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [IDE mode] [1022:7800] (rev 40)
00:12.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:12.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:13.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB OHCI Controller [1022:7807] (rev 11)
00:13.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller [1022:7808] (rev 11)
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:780b] (rev 16)
00:14.2 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] FCH Azalia Controller [1022:780d] (rev 01)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:780e] (rev 11)
00:14.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] FCH PCI Bridge [1022:780f] (rev 40)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 0 [1022:141a]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 1 [1022:141b]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 2 [1022:141c]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 3 [1022:141d]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 4 [1022:141e]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 15h (Models 30h-3fh) Processor Function 5 [1022:141f]
01:00.0 Display controller [0380]: Advanced Micro Devices, Inc. [AMD/ATI] Bonaire PRO [Radeon R9 M270X] [1002:6647] (rev ff)
02:00.0 Network controller [0280]: Intel Corporation Wireless 7260 [8086:08b1] (rev bb)
03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTL8411B PCI Express Card Reader [10ec:5287] (rev 01)
03:00.1 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 12)
`

JacobHeAMD, 2018-08-28:
It's not an issue of AMDVLK.

closed 2018-08-28 by JacobHeAMD