GPUOpen-Drivers/AMDVLK #15: vkAcquireNextImageKHR returning VK_TIMEOUT w/o waiting the timeout duration

mikedilger, 2018-02-27:
I've been able to cause a condition where vkAcquireNextImageKHR returns VK_TIMEOUT.  It's not the first frame, but rather near the first frame (maybe the second, third, or fourth).

The timeout parameter passed to that function has been 4000000000 nanoseconds (4 seconds).  Yet my renderer comes up and crashes all within about half a second, so it's clearly not waiting 4 seconds.

I can get the renderer to work in other conditions, and I'm not concerned about the condition that causes this, just about it's handling.  After all, it's quite possible that I haven't quite fulfilled the synchronization requirements, in which case it would technically be my fault.

However, IMHO, vkAcquireNextImageKHR shouldn't return VK_TIMEOUT unless it was actually a timeout. It should return an appropriate error if it was an error.

Vulkan layers report no warnings or errors.

mikedilger, 2018-02-27:
I would attach a trace but I can't seem to get the behavior to repeat when I run it under vktrace.

davidmaoamd, 2018-02-27:
Hi Dilger,
You might run into the case that the semaphore wait got interrupted or returned with some unexpected error.
I agree we’d better to improve this to handle those corner case correctly.
We will make the change.

Thanks.
Best Regards,
David

jinjianrong, 2018-05-15:
@mikedilger could you verify if the issue is fixed in latest code? thanks

closed 2018-05-15 by jinjianrong