GPUOpen-Drivers/AMDVLK #16: vsync not working correctly on xorg

aufkrawall, 2018-02-28:
In an Xorg session, try out any application which is suitable to spot tearing (e.g. vkquake, set host_maxfps to a value well above display refreshrate) and enable vsync in it.
There is a small stripe of tearing very close to the top of the screen, which is not there with any OGL application and RadeonSI.

Side note: RadV has a similar issue, but only in windowed mode. amdvlk has it both in windowed and fullscreen.

davidmaoamd, 2018-03-01:
AMDVLK uses copy to present even for fullscreen application.
That is why we would have different behavior compared to Radv in the fullscreen case. 
I would suspect it is the issue in the X server side, but not AMDVLK. 
Anyway, thanks for reporting this. 
I think the next step might be to file bugzilla in the freedesktop.

aufkrawall, 2018-03-01:
Thanks for your response, I created a bug entry in the xf86-video-amdgpu section of Xorg freedesktop tracker:
https://bugs.freedesktop.org/show_bug.cgi?id=105316

aufkrawall, 2018-03-04:
Michel Dänzer informed me that tearing in windowed mode on Xorg can in no way get reliably prevented without a compositor.
However, he shared some ideas why amdvlk behavior in fullscreen is different to radeonsi and radv, so I hope maybe something specific is possible regarding fullscreen vsync (please read the short freedesktop tracker conversation).

davidmaoamd, 2018-03-05:
As I mentioned before, for some reason, AMDVLK did not enable page flip for fullscreen application for now.
But we will try to enable it in the near future.
Thanks.

aufkrawall, 2018-03-06:
Thanks for the clarification, I'll gladly test it as soon it goes live.

JacobHeAMD, 2018-09-10:
For testing, you can set useFlipHint true in xgl/icd/settings.cfg.

aufkrawall, 2018-11-21:
Can I set this also with precompiled binaries?

JacobHeAMD, 2018-11-21:
It's not necessary to rebuild the binary, just add following line into /etc/amd/amdPalSettings.cfg
UseFlipHint,1

aufkrawall, 2018-11-21:
Thank you, seems to work like a charm. :)

closed 2018-11-26 by JacobHeAMD

aufkrawall, 2018-12-04:
Btw.: Is there a reason why this is not enabled by default?
I think the expectation is to completely eliminate tearing when turning on vsync in the application, without having to configure the driver first.

Alexander88207, 2018-12-04:
normally "TearFree" should help do it... 

> Is there a reason why this is not enabled by default?

is still good or not? because most of them take it in order for example to have no input delays etc ..

aufkrawall, 2018-12-04:
TearFree is not enabled by default for single display configurations.
If lag is an issue, why enable vsync in the first place?

Alexander88207, 2018-12-05:
depends on the game in my opinion for example i would never vsync on shooter's or games like need for speed.

aufkrawall, 2019-03-15:
@JacobHeAMD Could you please provide an update why this is not enabled by default? I don't experience any drawbacks with that option set.

Oschowa, 2019-04-21:
I also think it might be good to enable this by default, until i found this report i almost gave up on amdvlk because i thought the tearing was unfixable.

Oschowa, 2019-04-21:
At least please document this setting in the readme.

aufkrawall, 2019-05-29:
Hm, `UseFlipHint,1` screws up frametime variance in Hitman 2 DXVK + vsync, perceived smoothness on Xorg is really terrible vs. TearFree.

Please implement proper vsync in amdvlk, and of course also FreeSync at this occasion. :-1:  It's both available in RADV.
I'd reopen this ticket, but I can't.

aufkrawall, 2019-06-10:
Just in case: Please test vsync with either amdgpu.dc=0 (if you are on Polaris or older) or use the modesetting Xorg DDX driver, as there is this issue screwing up vsync in at least many Wine games with amdgpu.dc=1 + xf86-video-amdgpu-git:
https://bugs.freedesktop.org/show_bug.cgi?id=110659

aufkrawall, 2019-06-15:
I've opened a new ticket for it.