GPUOpen-Drivers/AMDVLK #17: Firmware feature level for Cape Verde is insufficient.

difron, 2018-03-03:
Feature level of firmware found in https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git for Cape Verde is 23. Minimum required feature level is 25. Is there a way to use amdvlk on this card?

JacobHeAMD, 2018-03-06:
What does "Minimum required feature level" refer to? amdvlk enables some feature according to the feature level, but it could work even with those feature disabled.
BTW, Verde is not enabled by kernel or upstreamed amdgpu.ko. To run amdvlk on Verde, custom kernel should be built with CONFIG_DRM_AMDGPU_SI enabled.

difron, 2018-03-06:
'Minimum requirements' refers to PAL src/core/hw/gfxip/gfx6/gfx6Device.cpp lines 2508-2511 and 'feature level' to what is returned by amdgpu_query_firmware_version from libdrm_amdgpu.so as last parameter.
Even if kernel is build with amdgpu enabled for Cape Verde specific version of firmware seams to be required.

difron, 2018-03-07:
A solution described in https://github.com/GPUOpen-Drivers/AMDVLK/issues/18.
Should I close this issue or wait until you upstream the firmware, if you are going to?

JacobHeAMD, 2018-03-07:
Please close it. I'm not sure if it will be up-streamed at this point. :(

difron, 2018-03-07:
OK then. But could you write instructions on how get old chips working somewhere in ReadMe, please?

closed 2018-03-07 by difron