GPUOpen-Drivers/AMDVLK #18: VK_ERROR_OUT_OF_HOST_MEMORY on radeon HD7770

boomboompsh, 2018-03-06:
Running vulkaninfo on my fully-updated archlinux machine results in the above-mentioned error. This error also appeared running on ubuntu, though I assumed I had done something wrong in the install process. The card has 1 GB of VRAM and the system has 8GB of RAM. According to [this forum post](https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/amd-linux/997201-amdvlk-vs-radv-vs-amdgpu-pro-17-50-vulkan-performance?p=997346#post997346) a similar issue exited in the closed-source opencl driver, which could be fixed by setting GPU_FORCE_64BIT_PTR=1

boomboompsh, 2018-03-06:
It's definitely enabled on my machine. I spent a really long time getting that set up properly. I had to add amdgpu.si_support=1 to the kernel parameters to get it to work. The radeon driver is blacklisted and dmesg only mentions amdgpu.

JacobHeAMD, 2018-03-06:
Good! Do you mind to debug why VK_ERROR_OUT_OF_HOST_MEMORY is reported? In the meanwhile, I'll download and build a custom kernel to reproduce this issue on Ubuntu.

boomboompsh, 2018-03-06:
It's almost midnight here, and I have work in the morning, so it'll have to wait.

JacobHeAMD, 2018-03-06:
It's ok, thank you very much!

difron, 2018-03-06:
The cause of this is https://github.com/GPUOpen-Drivers/AMDVLK/issues/17. I have the same card and initially got this VK_ERROR_OUT_OF_HOST_MEMORY.

boomboompsh, 2018-03-07:
I got as deep into debugging the issue as my knowledge of gdb would allow. The error occurs in fpCreateInstance. exploring it in gdb results in:
`The value of 'fpCreateInstance' is of type 'PFN_vkCreateInstance' which is a typedef of type 'VkResult (*)(const VkInstanceCreateInfo *, const VkAllocationCallbacks *, VkInstance *)'
'fpCreateInstance' is a pointer to a value of type 'VkResult (const VkInstanceCreateInfo *, const VkAllocationCallbacks *, VkInstance *)'`

Here's a backtrace of the line that calls fpCreateInstance:
`#0  0x00007ffff7634aca in loader_create_instance_chain (
    pCreateInfo=0x7fffffffe210, pAllocator=0x0, inst=0x5555557bc970, 
    created_instance=0x7fffffffe200)
    at /home/andrew/opt/Vulkan-LoaderAndValidationLayers/loader/loader.c:4668
#1  0x00007ffff763aa94 in vkCreateInstance (pCreateInfo=0x7fffffffe300, 
    pAllocator=0x0, pInstance=0x7fffffffe430)
    at /home/andrew/opt/Vulkan-LoaderAndValidationLayers/loader/trampoline.c:396
#2  0x0000555555557758 in AppCreateInstance (inst=0x7fffffffe430)
    at /home/andrew/opt/Vulkan-LoaderAndValidationLayers/demos/vulkaninfo.c:766
#3  0x00005555555605e1 in main (argc=3, argv=0x7fffffffe748)
    at /home/andrew/opt/Vulkan-LoaderAndValidationLayers/demos/vulkaninfo.c:2687`

With my limited knowledge of GDB, I can't get it to step inside fpCreateInstance. I would guess that's where it moves into the actual driver code, but I don't really know. I do know that it returns VK_ERROR_OUT_OF_HOST_MEMORY, which is what results in the error seen. faking success of the function (by setting res to VK_SUCCESS in gdb) results in a VK_ERROR_INITIALIZATION_FAILED.

boomboompsh, 2018-03-07:
I did a debug build of AMDVLK, the error output from vulkaninfo is as follows:

Vulkan API Version: 1.0.69

WARNING: [loader] Code 0 : loader_icd_scan: ICD JSON /usr/share/vulkan/icd.d/amd_icd64.json does not have an 'api_version' field.
AMD-PAL: Error: Unconditional Assert | Reason: Unknown (/home/andrew/opt/amdvlk-git/src/pal/src/core/hw/gfxip/gfx6/gfx6Device.cpp:2550:DetermineIpLevel)
AMD-PAL: Info: Vulkan error: VK_ERROR_OUT_OF_HOST_MEMORY(-1), from Pal error: Pal::Result::ErrorOutOfMemory(-4) (/home/andrew/opt/amdvlk-git/src/xgl/icd/api/vk_conv.cpp:1006:PalToVkError)
/home/andrew/opt/Vulkan-LoaderAndValidationLayers/demos/vulkaninfo.c:771: failed with VK_ERROR_OUT_OF_HOST_MEMORY

JacobHeAMD, 2018-03-07:
Hi @boomboompsh, thanks for your debugging. Looks like this issue is same as [#17](https://github.com/GPUOpen-Drivers/AMDVLK/issues/17) just as @difron mentioned. I'm trying to find a newer firmware. Will get back to you if any lucky.

JacobHeAMD, 2018-03-07:
Could you please try following steps to get the firmwares?
1. Download https://www2.ati.com/drivers/linux/ubuntu/amdgpu-pro-17.40-514569.tar.xz 
2. xz -d amdgpu-pro-17.40-514569.tar.xz && tar xvf amdgpu-pro-17.40-514569.tar
3. cd amdgpu-pro-17.40-514569
4. dpkg -x amdgpu-pro-dkms_17.40-514569_all.deb dkms
5. verde firmwares locates under dkms/usr/src/amdgpu-17.40-514569/firmware/radeon

difron, 2018-03-07:
Thank you,  @JacobHeAMD! This fixed my problem. Now firmware exposes feature level 29, and no more VK_ERROR_OUT_OF_HOST_MEMORY occurs.
Note: I think you need to regenerate initrd after copying firmware to your system folder.

boomboompsh, 2018-03-08:
Thank you! This fixed the problem. Closing the issue, although installing firmware from the amdgpu-pro packages should not be necessary in future. But you know that, and are probably getting it fixed as I'm typing this.

closed 2018-03-08 by boomboompsh

John-Gee, 2018-06-19:
This fixed the same issue on my 280X, it'd be nice if that upstreamed.

It makes me wonder what other difference there might be.