GPUOpen-Drivers/AMDVLK #19: vkGetInstanceProcAddress returning null when getting vkEnumerateDeviceGroups

edrickhong, 2018-03-17:
As the title says. I think this may be because I did not specify the device group creation extension when creating the VkInstance. However, I am unclear as to what it means to have these extensions promoted to core. I assume that I can expect it on all 1.1 confirming implementations. Or does it mean the extension is no longer considered a KHX extension and I still have to query for it?

JacobHeAMD, 2018-03-19:
Assuming the issue you report here is "vkGetInstanceProcAddr returning null when getting vkEnumeratePhysicalDeviceGroupsKHR" since I didn't see the issue with vkEnumeratePhysicalDeviceGroups.
VK_KHR_device_group_creation extension is still exposed by AMDVLK 1.1 driver, so you could get vkEnumeratePhysicalDeviceGroupsKHR by vkGetInstanceProcAddr if VK_KHR_device_group_creation extension is enabled while creating instance. And you are right, VK_KHR_device_group_creation extension is promoted to 1.1 core, that means you can get the vkEnumeratePhysicalDeviceGroups by vkGetInstanceProcAddr even without enabling VK_KHR_device_group_creation extension. Just don't confuse vkEnumeratePhysicalDeviceGroupsKHR  and vkEnumeratePhysicalDeviceGroups. vkEnumeratePhysicalDeviceGroupsKHR is for VK_KHR_device_group_creation extension, and vkEnumeratePhysicalDeviceGroups is for 1.1 core.

edrickhong, 2018-03-19:
Well, I'm querying for "vkEnumeratePhysicalDeviceGroups" . I see there's a new build up now and I'll try that this weekend. If the issue still persists, I'll let you know

edrickhong, 2018-03-21:
It's working. I must have pulled the wrong build. Thanks for the help

closed 2018-03-21 by edrickhong