GPUOpen-Drivers/AMDVLK #20: _ZN2vk6secure5entry24vkGetDeviceProcAddr_nameE VK_ERROR_INCOMPATIBLE_DRIVER

BoBXygene, 2018-03-25:
Just trying to build on updated ubuntu x64 while following instructions.

When i build, the make process halts on a missing ~/vulkan-driver/drivers/xgl/icd/api/strings/strings.cpp
I had to create an empty one (repo synch don't find any missing file an this git cant be browsed).

It then builds the amdvlk64.so but vulkan loader crashes because it doesn't find an exported method, even if it's there :

~/vulkan-driver//home/franck/vulkan-sdk/VulkanSDK/1.1.70.1/x86_64/bin/vulkaninfo 
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.1.70

ERROR: [Loader Message] Code 0 : /usr/lib/x86_64-linux-gnu/amdvlk64.so: undefined symbol: _ZN2vk6secure5entry24vkGetDeviceProcAddr_nameE
Cannot create Vulkan instance.
/tmp/SDK-DEV/Build/Vulkan-LoaderAndValidationLayers/demos/vulkaninfo.c:768: failed with VK_ERROR_INCOMPATIBLE_DRIVER

~/vulkan-driver/nm /usr/lib/x86_64-linux-gnu/amdvlk64.so|grep _ZN2vk6secure5entry24vkGetDeviceProcAddr_nameE
                 U _ZN2vk6secure5entry24vkGetDeviceProcAddr_nameE

So exports exists but wont get recognized.
Looks like that there is a mess in the datatypes between expected prototypes and built prototype.
(Cmake dind't emit any error before building)

JacobHeAMD, 2018-03-26:
For the string.cpp, please refer to https://github.com/GPUOpen-Drivers/xgl/blob/dev/icd/api/strings/strings.cpp . Empty string.cpp isn't expected to work.
The files included by strings.cpp should be generated by generate_strings.py during compiling.

JacobHeAMD, 2018-04-21:
Close it since no responding.

closed 2018-04-21 by JacobHeAMD