GPUOpen-Drivers/AMDVLK #21: Support for APU Graphics Device - Richland [Radeon HD 8610G]

rgh2, 2018-03-26:
Does the Radeon open source driver have support for APUs like the Richaland Radeon HD 8610G. This is the graphics device that I have installed on my laptop.  If the open source driver doesn't support this family of GPUs, what is the best way for me to experiment with Vulkan on my laptop ? Also, when is it likely that support for APUs like mine will be added ?  I think on the main page it said the drivers have support for the 8000M series, but I'm not sure if that includes APUs like the one I have installed.

ms178, 2018-04-03:
The Radeon HD 8610G is based on the older VLIW4-architecture wheras Vulkan is supported only on GCN-based GPUs. I am afraid that Vulkan won't be available for your laptop. I've got a 6770M in mine which is also a VLIW4 chip, so we are in the same boat and need to upgrade sooner or later.

JacobHeAMD, 2018-04-08:
Yes, @ms178 is right. Thanks, @ms178 .

closed 2018-04-08 by JacobHeAMD