GPUOpen-Drivers/AMDVLK #22: 29-02 release notes mention extensions not exposed yet..

oscarbg, 2018-04-01:
it says:
Implementation for below AMD extensions: 
AMD_shader_fragment_mask
AMD_gcn_shader
AMD_shader_trinary_minmax
AMD_shader_explicit_vertex_parameter
AMD_shader_ballot

but my 2.0.21 Vega report (https://vulkan.gpuinfo.org/displayreport.php?id=2884#extensions)
shows still missing:

> AMD_shader_fragment_mask
> AMD_gcn_shader
> AMD_shader_trinary_minmax
> AMD_shader_explicit_vertex_parameter
> AMD_shader_ballot
> 

also release notes mention:

> Hide VK_AMD_negative_viewport_height in Vulkan 1.1

but it still shows in extensions list..

jinjianrong, 2018-04-02:
@oscarbg These extensions are implemented (partly) but have not been enabled yet.  VK_AMD_negative_viewport_height is hidden for Vukan 1.1 but still exposed as 1.0 extension, please create a 1.1 instance to check if the extension still shows.

oscarbg, 2018-04-02:
thanks for clarification..
hope complete implementation comes soon..
for VK_AMD_negative_viewport_height was just using vulkaninfo tool  from 1.1.70 SDK that shows 1.1 version and the negative viewport extensión altough the tool may be querying a 1.0 instance..

jinjianrong, 2018-04-02:
These extension will be enabled in 1-2 weeks

closed 2018-04-03 by jinjianrong

oscarbg, 2018-04-20:
ping.. 
seems not present on v2.0.25.. 
cah share info on:
any big delays?
will be in next week code drop?
thanks..

jinjianrong, 2018-04-20:
@oscarbg The extensions listed above have been enabled in latest code expect AMD_shader_ballot. Please check it and let us know if any exension is still missed.  AMD_shader_ballot is delayed because of  LLVM backend issue, wil be in next week drop.

oscarbg, 2018-04-23:
yes,
just tested today with 2.0.27 and it's also reported:
https://vulkan.gpuinfo.org/displayreport.php?id=3066#extensions

jinjianrong, 2018-04-23:
@oscarbg  could you help check if below line is in your local xgl/icd/api/vk_physical_device.cpp file
                   availableExtensions.AddExtension(VK_DEVICE_EXTENSION(AMD_GCN_SHADER));