GPUOpen-Drivers/AMDVLK #27: Possible bug when writing the vertex fetch instructions

godlikepanos, 2018-04-11:
Hi,

I have a vertex attribute with format equal to VK_FORMAT_A2B10G10R10_SNORM_PACK32. My program fails on pipeline creation. Did some digging and I hit an assertion deep into LLVM code.

The issue starts in:

llpcVertexFetch.cpp:1163

// Insert alpha channel: %vf0 = insertelement %vf0, %a, 3
vertexFetch[0] = InsertElementInst::Create(vertexFetch[0],
    pAlpha,
    ConstantInt::get(m_pContext->Int32Ty(), 3),
    "",
    pInsertPos);

The pAlpha is of type FloatTyID and vertexFetch[0] is VectorTyID. 

Later on there is this assert:

assert(isValidOperands(Vec, Elt, Index) &&
    "Invalid insertelement instruction operands!");

which states that "Second operand of insertelement must be vector element type" which is probably not.

If you do a quick code inspection, is this piece of code correct? If yes I'll dig some more and create a repro case.

linqunzju, 2018-04-12:
It seems it is a bug in our driver. vertexFetch[0] shouldbe VectorTyID with integer type. but pAlpha is float now. please try to change the type to Int32Ty in previous bitcast instruction

// %a = bitcast %a to i32
pAlpha = new BitCastInst(pAlpha, m_pContext->Int32Ty(), "", pInsertPos);

please let me know if it can work in your case.

godlikepanos, 2018-04-21:
After some more digging it seems something is still a bit off. 

As I mentioned before I use VK_FORMAT_A2B10G10R10_SNORM_PACK32 (for tangents where w component holds the sign of the bitangent). I've done an experiment where I hardcode 0b11 to the alpha component of this format when building my assets. In theory, that will return -1.0 inside the vertex shader. Nevertheless the shader reports +1.0. 

In Mesa GL and nVidia GL the behaviour seems the expected. In AMDVLK and -oddly enough- in RADV it seems it doesn't.

By looking at your comments in llpcVertexFetch.cpp I see that there are some annoyances around the sign of alpha.

linqunzju, 2018-04-25:
sorry for delay response, I agree this code is too tricky. it is easy to cause bugs after LLVM optimization.

Can you try following code in "else if (pFormatInfo->nfmt == BUF_NUM_FORMAT_SNORM)"

pAlpha = new BitCastInst(pAlpha, m_pContext->FloatTy(), "", pInsertPos);
pAlpha = BinaryOperator::CreateMul(pAlpha,
                                                            ConstantFP::get(m_pContext->FloatTy(), 3.0f),
                                                             "", pInsertPos);
auto pCond = new ICmpInst(pInsertPos,
                                               CmpInst::FCMP_UGT,
                                               pAlpha,
                                               ConstantFP::get(m_pContext->FloatTy(), 1.0f),
                                               "");
pAlpha = SelectInst::Create(pCond, ConstantFP::get(m_pContext->FloatTy(), -1.0f), pAlpha, "", pInsertPos);
pAlpha = new BitCastInst(pAlpha, m_pContext->Int32Ty(), "", pInsertPos);

ps: above code isn't tested yet. please let me know if doesn't work.

godlikepanos, 2018-04-25:
I'm afraid that asserts in:

pAlpha = BinaryOperator::CreateMul(pAlpha, ConstantFP::get(m_pContext->FloatTy(), 3.0f), "", pInsertPos);

linqunzju, 2018-04-26:
Got it, I will write a test to test it and let you know when it is done.

linqunzju, 2018-04-26:
please try again, thanks!

            else if (pFormatInfo->nfmt == BUF_NUM_FORMAT_SNORM)
            {
                // %a = bitcast %a to f32
                pAlpha = new BitCastInst(pAlpha, m_pContext->FloatTy(), "", pInsertPos);
                // %a = mul %a, 3.0f
                pAlpha = BinaryOperator::CreateFMul(pAlpha,
                                                   ConstantFP::get(m_pContext->FloatTy(), 3.0f),
                                                   "",
                                                   pInsertPos);
                // %cond = ugt %a, 1.5f
                auto pCond = new FCmpInst(pInsertPos,
                                          FCmpInst::FCMP_UGT,
                                          pAlpha,
                                          ConstantFP::get(m_pContext->FloatTy(), 1.5f),
                                          "");
                // %a = select %cond, -1.0f, pAlpha
                pAlpha = SelectInst::Create(pCond,
                                            ConstantFP::get(m_pContext->FloatTy(), -1.0f),
                                            pAlpha,
                                            "",
                                            pInsertPos);
                // %a = bitcast %a to i32
                pAlpha = new BitCastInst(pAlpha, m_pContext->Int32Ty(), "", pInsertPos);
            }
            else if (pFormatInfo->nfmt == BUF_NUM_FORMAT_SSCALED)
            {
               // %a = bitcast %a to float
                pAlpha = new BitCastInst(pAlpha, m_pContext->FloatTy(), "", pInsertPos);
                // %a = fptosi %a to i32
                pAlpha = new FPToSIInst(pAlpha, m_pContext->Int32Ty(), "", pInsertPos);
                // %a = shl %a, 30
                pAlpha = BinaryOperator::CreateShl(pAlpha,
                                                   ConstantInt::get(m_pContext->Int32Ty(), 30),
                                                   "",
                                                   pInsertPos);
                // %a = ashr a, 30
                pAlpha = BinaryOperator::CreateAShr(pAlpha,
                                                    ConstantInt::get(m_pContext->Int32Ty(), 30),
                                                    "",
                                                    pInsertPos);
                // %a = sitofp %a to float
                pAlpha = new SIToFPInst(pAlpha, m_pContext->FloatTy(), "", pInsertPos);
                // %a = bitcast %a to i32
                pAlpha = new BitCastInst(pAlpha, m_pContext->Int32Ty(), "", pInsertPos);

godlikepanos, 2018-04-26:
Thanks again. Now it appears correct.

jinjianrong, 2018-05-10:
The fix is included May 9 code drop. Please check it.

closed 2018-05-10 by jinjianrong