GPUOpen-Drivers/AMDVLK #28: Request to provide precompiled builds at http://repo.radeon.com/amdvlk/ soon?

oscarbg, 2018-04-12:
Hi,
just exploring ROCM repo "repo.radeon.com" found and amdvlk folder..
altough is empty right now, that is a hint that precompiled builds are coming soon there?
for ex. builds for Ubuntu (similar to ROCM) are going to be provided soon?
thanks..

jinjianrong, 2018-04-12:
yes, we are going to provide builds on repo.radeon.com

oscarbg, 2018-05-09:
seems they are live!
I hope every weekly drop or so of code gets a new build on the repo!
thanks &closing..

JacobHeAMD, 2018-05-09:
Please hold on, the repo doesn't work yet because I was interrupted by other urgent issue. Will keep you posted once it works.

JacobHeAMD, 2018-05-09:
It's supposed to be working now, please try it. 
wget -qO - http://repo.radeon.com/amdvlk/apt/debian/amdvlk.gpg.key | sudo apt-key add -
sudo sh -c 'echo deb [arch=amd64] http://repo.radeon.com/rocm/apt/debian/ xenial main > /etc/apt/sources.list.d/amdvlk.list'
apt update
apt-get install amdvlk

nadrolinux, 2018-09-01:
Hello,
Have you got any plans to update this repository?
Patryk,

JacobHeAMD, 2018-09-03:
Sorry, it is in "pending" status. Hopefully I'll have time on this this week.

ernstp, 2018-11-12:
Ping for v2018.4.1 ?

ernstp, 2018-11-12:
Right, so why not add them to the repository at http://repo.radeon.com/amdvlk/apt/debian/pool/main/a/amdvlk/ also? So people can start using that...

CrackedCrafterz, 2018-11-13:
Hello @JacobHeAMD 

I regularly build arch packages (.pkg.tar.xz) from amdvlk. Would be possible if someone would view [this](https://github.com/CrackedCrafterz/AMDVLK-Arch) ?, maybe it would then enlarge the pool and nobody needs to waste over 2 hours on building 😃

JacobHeAMD, 2018-11-15:
The source for Ubuntu 18.04 is deployed to repo.radeon.com, please try it.
sudo wget -qO - http://repo.radeon.com/amdvlk/apt/debian/amdvlk.gpg.key | sudo apt-key add -
sudo sh -c 'echo deb [arch=amd64] http://repo.radeon.com/amdvlk/apt/debian/ bionic main > /etc/apt/sources.list.d/amdvlk.list'
apt update
apt-get install amdvlk

ernstp, 2018-11-17:
That works fine... 32-bit libraries? Not 32-bit Ubuntu support of course, just the multilib stuff...

JaxLinAMD, 2020-07-24:
32bit pakcage is provided since v-2020.Q3.1

closed 2020-07-24 by JaxLinAMD