GPUOpen-Drivers/AMDVLK #31: Requests for multiple extensions..

oscarbg, 2018-05-03:
Hi,
I was to open a new issue asking for descriptor indexing extension (VK_EXT_descriptor_indexing) after learning in detail in Vulkan Dev Day in Montreal slides: https://www.khronos.org/assets/uploads/developers/library/2018-vulkan-devday/10-DescriptorIndexing.pdf
Better I'm consolidating it all requests here so others can add to this thread and not polluting as new issue requests:

Thanks..

jinjianrong, 2018-05-07:
@oscarbg thanks for the requests. we have been working on these extensions.

oscarbg, 2018-08-07:
hi,
just updated list of missing exts vs AMD RADV Mesa driver or EXT,KHR ones missing that are supported on both Intel & Nvidia GPUs..
no pressure, just keep tracking of differences between drivers..

jinjianrong, 2018-08-07:
@oscarbg  thanks for the update

oscarbg, 2018-09-04:
Hi seems Nvidia added Vulkan HDR support (Windows only) to his latest dev driver via 2 exts 
    VK_EXT_swapchain_colorspace 
    VK_EXT_hdr_metadata 
so just asking for Win support here & adding to the list..

oscarbg, 2018-10-25:
Small update..
Adding request for VK_EXT_transform_feedback used in DXVK 0.9 needed for correct render of Witcher 3 and other games..
Done:
VK_EXT_swapchain_colorspace
VK_EXT_hdr_metadata

CrackedCrafterz, 2018-11-14:
I need
VK_KHR_get_physical_device_properties2
VK_KHR_surface
VK_KHR_win32_surface

for Battlefield Bad Company 2

jinjianrong, 2018-11-21:
@CrackedCrafterz   VK_KHR_get_physical_device_properties2 adn VK_KHR_surface are already supported by the open source driver. Regarding VK_KHR_win32_surface, it is supported by AMD windows driver, but currently the open source driver only supports Linux.

CrackedCrafterz, 2018-11-21:
> @CrackedCrafterz VK_KHR_get_physical_device_properties2 and VK_KHR_surface are already supported by the open source driver. 

Ok and why then dxvk reports in the log that they are missing? nevermind i just go back to radv, thanks anyway 👍

Degerz, 2019-01-25:
Can we please get priority for VK_EXT_transform_feedback ? Can we also have the _geometryStreams_ feature enabled and the properties such as _Queries_, _StreamsLinesTriangles_, _RasterizationStreamSelect_, and _Draw_ supported as well ?

oscarbg, 2019-01-25:
same as @Degerz :
+1 on transform feedback, with recent support on Mesa master branch for Intel iGPUs (Anvil) and NV supp. in stable Linux drivers like 415.27, AMDVLK is the remaining one..

Degerz, 2019-01-30:
I see now that you guys added in transform feedback [here](https://github.com/GPUOpen-Drivers/xgl/commit/6a629e8286ed9d2a8a3799131a274ee58590765d) which makes your Vulkan driver capabilities more at par with the D3D12 driver but I still see that _RasterizationStreamSelect_ is not yet supported. Hopefully, support for the property will be added later on to cover some more cases and to match the D3D12 stream-output functionality ...

jinjianrong, 2019-01-30:
@Degerz We will add RasterizationStreamSelect  support later

Degerz, 2019-01-31:
@jinjianrong That's good to hear and it would also be nice as well if your team could consider adding in support for bindless storage buffers [via](https://github.com/KhronosGroup/Vulkan-Docs/issues/890) VK_EXT_buffer_device_address. NV_shader_buffer_load also provides [similar](https://www.khronos.org/registry/OpenGL/extensions/NV/NV_shader_buffer_load.txt) functionality as well and there's a multi-vendor [equivalent](https://github.com/KhronosGroup/GLSL/pull/49) as well for GL ...

oscarbg, 2019-05-05:
can we get some priority for supporting VK_EXT_host_query_reset?
from DXVK 1.1.1 release notes

```
In order to improve performance in Unreal Engine 4 titles and others, Queries were re-implemented to allow for better GPU utilization. Since this is a widely used API feature, this change may affect many games.

Note: For best performance, both your Wine version and graphics driver must support VK_EXT_host_query_reset.
```

AnzhongHuang, 2019-05-05:
@oscarbg It is implemented but it is not exposed for open source driver. I think we will expose it soon.

oscarbg, 2019-06-04:
Hi @AnzhongHuang , thanks it seems VK_EXT_host_query_reset got implemented in latest AMDVLK 2.0.91.. still missing in Windows driver but will be hopefully soon..

just adding new interesting extensions (missing support yet in AMDVLK) to first post.. from my POV in order of more to less interest to me:
*VK_EXT_fragment_shader_interlock
*VK_EXT_buffer_device_address
*VK_EXT_pipeline_creation_feedback
especially interested in new fragment_shader_interlock ext. as that exposes ROV functionality which is a new feature in AMD Vega GPUs and equivalent functionaly has been already present (for years!) in D3D12 and Metal2 APIs..
hopefully all of them got interest from AMD devs to implement :-)

thanks..

Degerz, 2019-08-08:
Could we get these extensions prioritized for our use case ?

`VK_EXT_fragment_shader_interlock`

There's a similar feature supported in AMD's D3D drivers and it would be great if we can get the equivalent functionality exposed in Vulkan so that it mirrors the feature set as seen in AMD's D3D drivers. Also, I would like both the feature's `SampleInterlockEXT` and especially `PixelInterlockEXT` to be enabled as well. 

`VK_EXT_index_type_uint8`

Can we have uint8 indices enabled as well ? This feature is only available in AMD's OpenGL drivers so it would be nice if it could be supported on AMDVLK too. uint8 indices are natively supported as well in hardware [since](https://github.com/mesa3d/mesa/blob/master/src/amd/registers/amdgfxregs.json#L1412) GCN3.

It would be much appreciated if the team can take on this request, please and thank you.

edrickhong, 2019-09-06:
Can we get VK_EXT_full_screen_exclusive? From gpuinfo, it looks like it has only seen adoption on Windows. Is the issue somewhere else on the stack?

JacobHeAMD, 2019-09-06:
VK_EXT_full_screen_exclusive is Windows-only extension, amdvlk doesn't support Windows. On Linux, it can't be acheived because of DRI3 protocal. An alternative on Linux with Xwindow is to acquire a display from Xserver (VK_EXT_acquire_xlib_display).

b1ackviking, 2019-11-10:
Hello,
Is there any progress on VK_KHR_push_descriptor? Will it be released soon on Windows or Linux?
It is widely supported by other drivers, including RADV for AMD on Linux since 2017. This extension is equivalent to D3D12 Root descriptors functionality and it is required for porting D3D12 applications to Vulkan.
Thanks.

Degerz, 2019-11-10:
@b1ackviking It will never happen ... 

Push descriptors are a slow path on AMD hardware so you'll have to emulate this behaviour on the CPU which is likely what drivers for AMD GPUs was doing anyways ...

AMD also explicitly recommended that you avoid using root descriptors as well on D3D12 ...

AnzhongHuang, 2019-11-11:
@b1ackviking Vulkan also has root descriptors, e.g. 
VK_DESCRIPTOR_TYPE_UNIFORM_BUFFER_DYNAMIC, VK_DESCRIPTOR_TYPE_STORAGE_BUFFER_DYNAMIC and push constant.

push_descriptor indicates the way to update descriptors, rather than where descriptors should be located. For AMD, the "root registers" are limited, so we can't put everything to "root", even the BUFFER_DYNAMIC and push constant I mentioned, they also could not be in the root if there is not enough registers.

b1ackviking, 2019-11-11:
Hello @Degerz, @AnzhongHuang
Thanks for your answers and clarification about the root registers.

Indeed, I was referring to the way to update descriptors rather than storing them.

Although storing data in a large buffer and providing dynamic offset is a common practice, which is the case with VK_DESCRIPTOR_TYPE_UNIFORM_BUFFER_DYNAMIC and VK_DESCRIPTOR_TYPE_STORAGE_BUFFER_DYNAMIC, there are scenarios when it is necessary to update the descriptor itself between drawcalls.

It is definetly doable on the CPU side by creating several versions of descriptor sets and binding them normally, but still it would be nice to have VK_KHR_push_descriptor in the driver, given that this extension is quite common among other drivers.

Degerz, 2019-11-12:
@b1ackviking Sure AMD could implement that functionality in their Vulkan drivers but that defeats one of Vulkan's main design philosophy which is supposed to be finding a way to hit the fast paths in hardware. Letting the driver manage this memory is inefficient compared to doing it yourself ... 

If you only need to update the descriptors then consider refactoring your codebase to use VK_KHR_descriptor_update_template instead ... 

If you absolutely must have to create a new descriptor set then just do it on the CPU ...

It's in AMD's interests to discourage bad practices like this to avoid hitting the slow paths in their drivers/hardware ...

b1ackviking, 2019-11-12:
@Degerz I totally agree 👍

Degerz, 2019-12-09:
@jinjianrong Is there a specific reason why your team won't expose VK_EXT_Conditional_Rendering ? Conditional rendering is also available on AMD's D3D11/12 drivers which is known as "predication". Does conditional rendering/predication trigger a slow-path in AMD's drivers or is there no performance benefit ? 

Also do you plan on supporting buffer references via VK_KHR_buffer_device_address ? IMO it's more convenient than using descriptor indexing to bind the SSBOs in the descriptor set.

oscarbg, 2020-05-08:
reviving this thread to request VK_EXT_pipeline_creation_cache_control support (win drv supports it already) for new Cemu 1.19 async shader & pipeline compilation feature ..

Degerz, 2020-05-09:
@oscarbg Not trivial since AMDVLK on Linux uses the LLVM compiler while AMDVLK on Windows uses the proprietary compiler ...

Also when can we expect VK_EXT_custom_border_color to be exposed soon since both D3D11/D3D12 have variable border colour ?

oscarbg, 2020-05-09:
@Degerz good to know.. 
and yes I forgot about VK_EXT_custom_border_color now used/supported in latest:
* Cemu 1.19
* Yuzu EA
* DXVK git
* VKD3D master branch

RinMaru, 2020-08-19:
Looks like fragment shader interlock isnt happening #108

oscarbg, 2020-10-18:
in addition to earlier requested VK_EXT_custom_border_color Yuzu seems to use also:
***VK_EXT_index_type_uint8**
so it would be nice to have..
note propietary Intel and Windows drivers have it already..

yuzu log, where I have seen this, using latest Windows 20.9.2 Win driver:
AMDVLK should be similar:
```
Render.Vulkan <Info> video_core\renderer_vulkan\vk_device.cpp:Create:315: **Device doesn't support viewport swizzles**
Render.Vulkan <Info> video_core\renderer_vulkan\vk_device.cpp:Create:339: **Device doesn't support uint8 indexes**
Render.Vulkan <Info> video_core\renderer_vulkan\vk_device.cpp:Create:365: **Device doesn't support custom border colors**
Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:433: Driver: AMD proprietary driver 2.0.158
[   4.933861] Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:434: Device: Radeon RX Vega
[   4.933862] Render.Vulkan <Info> video_core\renderer_vulkan\renderer_vulkan.cpp:Report:435: Vulkan: 1.2.152
```

LendoK, 2020-11-11:
**VK_EXT_conditional_rendering** would be really nice

nyanmisaka, 2020-11-26:
Bring **`VK_KHR_push_descriptor`** to Windows please.

oscarbg, 2021-01-03:
Hi,
please support VK_VALVE_mutable_descriptor_type extension..
it's recommended for VKD3D 2.1 (so for running DX12 games under Wine), and it's currently needed for Cyberpunk 2077 support (**)..
Note RADV currently supports it..
>   Cyberpunk 2077 (**)
> 
> (**): Currently only runs correctly on AMD hardware with RADV and VK_VALVE_mutable_descriptor_type.

baryluk, 2021-04-06:
From all the missing ones, I would myself vote on `VK_EXT_conditional_rendering`. Primary reason is that running zink on amdvlk would allow it to support OpenGL 4.6, instead of now OpenGL 2.1. sic! All other extensions and features looks to be supported.

jinjianrong, 2021-04-07:
@baryluk  VK_EXT_conditional_rendering is already supported by amdvlk. please check it.

nyanmisaka, 2021-04-07:
AFAIK the `VK_KHR_push_descriptor` is still not supported on Windows. (Adrenalin 20.3.2)

baryluk, 2021-04-07:
> @baryluk VK_EXT_conditional_rendering is already supported by amdvlk. please check it.

I did check yesterday, and it did not list in `vulkaninfo` on my Fury X (FIJI, GFX8). While RADV/ACO did.

I will download and compile `amdvlk` again, and report back later.

jinjianrong, 2021-04-09:
@baryluk  Currently AMDVLK doesn’t support VK_EXT_conditional_rendering on gfx8 due to no CP microcode support for 32bitMemoryPredication

mfleisz, 2021-06-01:
Any plans on when Vulkan Video (https://www.khronos.org/blog/an-introduction-to-vulkan-video) support will be available?

Degerz, 2021-06-18:
I think this issue can be closed now. For other features like video decode/encode we could file new issues instead ?

jinjianrong, 2021-06-18:
let's use new individual issue to track other new features

closed 2021-06-18 by jinjianrong

oscarbg, 2022-05-28:
sorry for posting here, but just pointing new missing extensions for my own future tracking..
22.5.2 Win driver:
*VK_KHR_push_descriptor
*VK_KHR_ray_tracing_maintenance1
*VK_AMD_shader_early_and_late_fragment_tests

latest available on Nvidia VK dev driver (or even as Mesa RADV possible still open Merge Requests):

*VK_KHR_fragment_shader_barycentric
*VK_EXT_image_2d_view_of_3d
*VK_EXT_graphics_pipeline_library
*VK_EXT_primitives_generated_query

oscarbg, 2022-08-26:
+VK_NV_compute_shader_derivatives: for UE5 Nanite games under VKD3D: https://github.com/GPUOpen-Drivers/AMDVLK/issues/289

oscarbg, 2022-09-02:
Mesh shaders also please..

amdrexu, 2022-09-05:
> Mesh shaders also please..

Mesh shader support will be released in the near future. Please wait.