GPUOpen-Drivers/AMDVLK #32: Writing to MRT with masked color write fails

Themaister, 2018-05-09:
MRT with 4 RTs, and color write mask setup:
0: 0x0
1: 0xf
2: 0xf
3: 0xf

The driver is confused and RT 1 outputs are written to RT 0, and so on, off-by-one.

amdrexu, 2018-05-10:
Could you attache the shader please? So we can have a check.

Themaister, 2018-05-10:
[fossilize.json.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/1991974/fossilize.json.txt)
[gbuffer.asm.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/1991976/gbuffer.asm.txt)

To replay the pipeline creation with all info:
```
git clone git://github.com/Themaister/Fossilize
cd Fossilize
git submodule update --init
mkdir build
cd build
cmake ..
make -j16
./cli/fossilize-replay /tmp/fossilize.json.txt --filter-graphics 1
```

amdrexu, 2018-05-11:
Reproduced. We will have a look and give you update once the root cause is pinpointed. Thank you.

amdrexu, 2018-05-14:
We have found the root cause and fix it internally. Will be available in next code base update. Thank you.

Themaister, 2018-06-01:
Pulled latest driver and seems to work fine now.

closed 2018-06-04 by jinjianrong