GPUOpen-Drivers/AMDVLK #33: DXVK flickering garbage

notaz, 2018-05-11:
When running games through wine+DXVK, some games are suffering from flickering garbage. The same setup works correctly when using the -pro driver or when using DXVK on AMD under Windows. It also works on a nvidia machine under Linux.

Here is a RenderDoc capture (requires nightly build of RenderDoc because DXVK uses recent extensions):
https://drive.google.com/file/d/1MZ3GaxQaKgdJ-FZHRj7hnpmimv9bKodU/view?usp=sharing
(visible best with "renderdoccmd replay")

GPU: POLARIS10/rx470 (others are known to be affected)

DXVK issue, for reference:
https://github.com/doitsujin/dxvk/issues/10

libcg, 2018-05-17:
Hit the same issue with Trackmania2: Stadium. GPUOpen-Drivers/xgl#21

exolyte, 2018-05-21:
This was fixed in the mesa driver by https://patchwork.freedesktop.org/patch/224448/

jinjianrong, 2018-07-12:
The  issue is fixed in latest code update. please check it.  thanks

closed 2018-07-12 by jinjianrong