GPUOpen-Drivers/AMDVLK #34: Ps4 Vulkan

Astromatik, 2018-05-13:
Hello. 
I have an PS4 with archlinux on it. Do you think if its possible to run VULKAN?
I m an beginer on linux scene, and my linux experience is to short. 
Marcan an developer run it in 1 years ago. 
Lot of player turning linux on it but is not respond... 
Can you help me ?

Astromatik, 2018-05-14:
Maybe I can help you to make an report of the devices

jinjianrong, 2018-05-15:
it is supposed that PS4 GPUs are not supported by amdgpu KMD and Vulkan driver. could you help check what GPU is used in your PS4 and run  vkinfo to see what error is reported?

libcg, 2018-05-15:
@Astromatik the PS4 linux port uses the radeon kernel driver instead of amdgpu. until someone ports it to amdgpu Vulkan support will not be possible.

Astromatik, 2018-05-15:
I m on Archlinux.Not Ubuntu or RedHat... .when I install vulkan-Radeon in AUR that make :

[root@psxitarch psxita]# pacman -S vulkan-radeon
résolution des dépendances...
recherche des conflits entre paquets...
:: vulkan-radeon et mesa-git sont en conflit. Supprimer mesa-git ? [o/N] o
erreur : la préparation de la transaction a échoué (la satisfaction des dépendances a échoué)
:: gst-plugins-base-libs : supprimer mesa-git casse la dépendance « mesa »
:: gtk3 : supprimer mesa-git casse la dépendance « mesa »
:: lib32-mesa-git : supprimer mesa-git casse la dépendance « mesa-git »
:: libglvnd-git : supprimer mesa-git casse la dépendance « mesa »
:: libglvnd-git : supprimer mesa-git casse la dépendance « opengl-driver »
:: retroarch : supprimer mesa-git casse la dépendance « mesa »
:: xf86-video-ati-git : supprimer mesa-git casse la dépendance « mesa »
:: xorg-server-devel-git : supprimer mesa-git casse la dépendance « mesa »

Astromatik, 2018-06-04:
Hello We have some news on the vkinfo:
https://pastebin.com/u/mircoho

Please can you check that

Astromatik, 2018-06-04:
We very Need help... Please, Vulkan run at 1fps...

JacobHeAMD, 2018-06-05:
Hi @Astromatik , according to vulkaninfo, it's running on Radv vulkan driver, which is owned by Mesa. To enable AMDVLK on Bonaire, please do as following,
1. Blacklist radeon.ko
2. Rebuild the kernel with CONFIG_DRM_AMDGPU_CIK enabled.
3. add amdgpu.cik_support=1 to the kernel parameters
4. Install AMDVLK driver and run vulkaninfo again.

Astromatik, 2018-06-05:
Hello Thank for your Quick ans error, I m on sea island the process is the same? 
Best regard

JacobHeAMD, 2018-06-05:
For SI, it's CONFIG_DRM_AMDGPU_SI and amdgpu.si_support=1 respectively.

Supatier, 2018-06-10:
Out of curiosity, what kernel version are you using.

Astromatik, 2018-06-10:
Its 4.1.14
We very Need help...

mirh, 2018-08-28:
https://twitter.com/marcan42/status/817235485689020417
At least with that 4.10, you needed a [good amount](https://github.com/fail0verflow/ps4-radeon-patches/tree/master/libdrm-2.4.74-amdgpupro-patchset) of libdrm patches to make it work. 

So, before complaining about user space (which should be next to trivial) somebody should upstream properly that.

JacobHeAMD, 2018-11-26:
Hi @Astromatik , did you re-open this ticket for any issues?

Astromatik, 2018-11-26:
Yes I reopen it for somes advances about Vulkan on PS4 linux. 

https://mobile.twitter.com/RancidoPS3ita/status/1061747667582271489?s=19

https://github.com/Ps3itaTeam/ps4linux-video-drivers

We Need Need somes Light about This, if AMDVLK can help This issues. We are near of the final objectif: Set the PS4 like an real PC linux.

JacobHeAMD, 2018-12-06:
Hi @Astromatik, looks like your project is based on Mesa. Mesa Vulkan (RADV) and AMDVLK are two different vulkan driver, it's beyond our capability to support RADV. Furthermore, you should report a bug/issue in detail so that we can try to reproduce it (if it's a AMDVLK issue).

closed 2018-12-06 by JacobHeAMD

mirh, 2018-12-07:
His issue is exactly that AMDVLK *isn't* supporting the ps4 gpu to begin with. 

I understand it's certainly funny for you, and a surprise to be sure, to be in the position *community has the hardware to test* while you don't...
Still, the change could possibly just be [this](https://github.com/fail0verflow/ps4-radeon-patches/blob/master/mesa-13.0.2-liverpool.patch)