GPUOpen-Drivers/AMDVLK #35: All vkQueueSubmit calls fail

edrickhong, 2018-05-28:
I remember working around this issue before on Ubuntu 16.04 but I can't remember exactly how. Upgraded to 17.10 and I'm running into the same problem again.

JacobHeAMD, 2018-05-30:
Please provide more information about how to reproduce this issue? Such as GPU, kernel, drm?

edrickhong, 2018-05-30:
Ubuntu 17.10 
Rx 560
Kernel version 4.13

JacobHeAMD, 2018-05-30:
It's not seen at my end. Is there any error message in kernel log? You can check it with "dmesg".

masahi, 2018-06-02:
I think I'm having the same problem. After building amdvlk, vulkaninfo works but the cube sample from Lunar SDK failed after vkQueueSubmit.

`$ ./cube
cube: /home/masa/VulkanSDK/1.1.73.0/examples/cube.c:641: demo_flush_init_cmd: Assertion `!err' failed.
`

JacobHeAMD, 2018-06-04:
Thanks, @masahi . There is no error message in dmesg log. Could you please help me to do some debug? Tracing Pal::Linux::Device::SubmitRaw and Pal::Linux::Device::Submit what is the "result"? I need to know the error code returned from kernel at least.

masahi, 2018-06-04:
ok, the return value of Pal::Linux::Device::Submit is -9, i.e. Result::ErrorInvalidValue

JacobHeAMD, 2018-06-05:
Would you mind to provide following information?
1. The content of pIbsRequest, and the value of numberOfRequests while Pal::Linux::Device::Submit is failed.
2. Run "dpkg -l | grep drm" in the console.
3. Run "cat /etc/lsb-release" in the console.

masahi, 2018-06-05:
numberOfRequests: 1
pIbsRequest->flags: 0
pIbsRequest->ip_type: 0
pIbsRequest->ip_instance: 0
pIbsRequest->ring: 0
pIbsRequest->number_of_dependencies: 0
pIbsRequest->number_of_ibs: 5
pIbsRequest->seq_no: 0

$ dpkg -l | grep drm
ii  libdrm-amdgpu1:amd64                       2.4.83-1\~16.04.1                             amd64        Userspace interface to amdgpu-specific kernel DRM services -- runtime
ii  libdrm-common                              2.4.83-1\~16.04.1                             all          Userspace interface to kernel DRM services -- common files
ii  libdrm-dev:amd64                           2.4.83-1\~16.04.1                             amd64        Userspace interface to kernel DRM services -- development files
ii  libdrm-intel1:amd64                        2.4.83-1\~16.04.1                             amd64        Userspace interface to intel-specific kernel DRM services -- runtime
ii  libdrm-nouveau2:amd64                      2.4.83-1\~16.04.1                             amd64        Userspace interface to nouveau-specific kernel DRM services -- runtime
ii  libdrm-radeon1:amd64                       2.4.83-1\~16.04.1                             amd64        Userspace interface to radeon-specific kernel DRM services -- runtime
ii  libdrm2:amd64                              2.4.83-1\~16.04.1                             amd64        Userspace interface to kernel DRM services -- runtime

$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"

JacobHeAMD, 2018-06-06:
libdrm-amdgpu1: 2.4.83 requires the number_of_ibs <= 4, however it is 5 for this case. There are two options to fix this issue:
1. Upgrade libdrm-amdgpu1 to 2.4.91
2. Add one line "MaxNumCmdStreamsPerSubmit,4" to /etc/amd/amdPalSettings.cfg

masahi, 2018-06-06:
hmm, I already have "MaxNumCmdStreamsPerSubmit,4" in /etc/amd/amdPalSettings.cfg. Don't know why I am still getting number_of_ibs begin 5.

$ cat /etc/amd/amdPalSettings.cfg
MaxNumCmdStreamsPerSubmit,4

Anyway, I will try upgrading libdrm-amdgpu.

masahi, 2018-06-06:
@JacobHeAMD When I print the value of `Settings().maxNumCmdStreamsPerSubmit` [here](https://github.com/GPUOpen-Drivers/pal/blob/dev/src/core/os/lnx/lnxDevice.cpp#L609), it says 4. So I am sure my setting is correct. But I am still getting number_of_ibs == 5.

Looking at lnxQueue.cpp, it seems to me that the driver is not respecting the overwritten MaxNumCmdStreamsPerSubmit. For example, [here](https://github.com/GPUOpen-Drivers/pal/blob/dev/src/core/os/lnx/lnxQueue.cpp#L1189) it only checks m_numIbs < AMDGPU_CS_MAX_IBS_PER_SUBMIT (16), before adding a new command.

Am I missing something?

JacobHeAMD, 2018-06-07:
Hi @masahi , sorry that setting MaxNumCmdStreamsPerSubmit is not enough now. Please add "CommandBufferCombineDePreambles,1" to amdPalSettings.cfg.

masahi, 2018-06-07:
@JacobHeAMD Thanks!! It works now.

closed 2018-06-08 by JacobHeAMD