GPUOpen-Drivers/AMDVLK #37: GPUOpen-Drivers/llvm does not build in Debug or Release modes

dnovillo, 2018-06-01:
I'm trying to build GPUOpen-Drivers/llvm to chase down a bug, but I cannot build it in Debug mode.  
The build works fine if I build in RelWithDebugInfo or Release.  Has anyone seen this failure?

This is the trace I get:

```
+ cd bld
+ cmake -G Ninja -DLLVM_BUILD_TESTS=ON -DCMAKE_BUILD_TYPE=Debug '-DLLVM_TARGETS_TO_BUILD='\''X86;AMDGPU'\''' ../llvm
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for dlfcn.h
-- Looking for dlfcn.h - found
-- Looking for errno.h
-- Looking for errno.h - found
-- Looking for fcntl.h
-- Looking for fcntl.h - found
-- Looking for link.h
-- Looking for link.h - found
-- Looking for malloc.h
-- Looking for malloc.h - found
-- Looking for malloc/malloc.h
-- Looking for malloc/malloc.h - not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for signal.h
-- Looking for signal.h - found
-- Looking for sys/ioctl.h
-- Looking for sys/ioctl.h - found
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for sys/param.h
-- Looking for sys/param.h - found
-- Looking for sys/resource.h
-- Looking for sys/resource.h - found
-- Looking for sys/stat.h
-- Looking for sys/stat.h - found
-- Looking for sys/time.h
-- Looking for sys/time.h - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for termios.h
-- Looking for termios.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for valgrind/valgrind.h
-- Looking for valgrind/valgrind.h - not found
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Looking for fenv.h
-- Looking for fenv.h - found
-- Looking for FE_ALL_EXCEPT
-- Looking for FE_ALL_EXCEPT - found
-- Looking for FE_INEXACT
-- Looking for FE_INEXACT - found
-- Looking for mach/mach.h
-- Looking for mach/mach.h - not found
[... 283 lines cut ...]
#7 0x000055e9667c1b3e bindingsErrorHandler(void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) <dir>/llpc//llvm/lib/Support/ErrorHandling.cpp:231:0
#8 0x000055e96653056f llvm::TypeInfer::ValidateOnExit::~ValidateOnExit() <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:819:0
#9 0x000055e96652bdff llvm::TypeInfer::MergeInTypeInfo(llvm::TypeSetByHwMode&, llvm::TypeSetByHwMode const&) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:328:0
#10 0x000055e96654def6 llvm::TreePatternNode::UpdateNodeType(unsigned int, llvm::ValueTypeByHwMode, llvm::TreePattern&) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.h:896:0
#11 0x000055e966535034 llvm::TreePatternNode::UpdateNodeTypeFromInst(unsigned int, llvm::Record*, llvm::TreePattern&) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:1533:0
#12 0x000055e96653a7c6 llvm::TreePatternNode::ApplyTypeConstraints(llvm::TreePattern&, bool) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:2402:0
#13 0x000055e96653dea3 llvm::TreePattern::InferAllTypes(llvm::StringMap<llvm::SmallVector<llvm::TreePatternNode*, 1u>, llvm::MallocAllocator> const*) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:2760:0
#14 0x000055e966542fe2 llvm::CodeGenDAGPatterns::parseInstructionPattern(llvm::CodeGenInstruction&, llvm::ListInit*, std::map<llvm::Record*, llvm::DAGInstruction, llvm::LessRecordByID, std::allocator<std::pair<llvm::Record* const, llvm::DAGInstruction> > >&) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:3585:0
#15 0x000055e96654353b llvm::CodeGenDAGPatterns::ParseInstructions() <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:3633:0
#16 0x000055e96653ebfc llvm::CodeGenDAGPatterns::CodeGenDAGPatterns(llvm::RecordKeeper&, std::function<void (llvm::TreePattern*)>) <dir>/llpc//llvm/utils/TableGen/CodeGenDAGPatterns.cpp:2864:0
#17 0x000055e96668fe96 (anonymous namespace)::GlobalISelEmitter::GlobalISelEmitter(llvm::RecordKeeper&) <dir>/llpc//llvm/utils/TableGen/GlobalISelEmitter.cpp:2832:0
#18 0x000055e96669b6ed llvm::EmitGlobalISel(llvm::RecordKeeper&, llvm::raw_ostream&) <dir>/llpc//llvm/utils/TableGen/GlobalISelEmitter.cpp:4218:0
#19 0x000055e9667661f3 (anonymous namespace)::LLVMTableGenMain(llvm::raw_ostream&, llvm::RecordKeeper&) <dir>/llpc//llvm/utils/TableGen/TableGen.cpp:206:0
#20 0x000055e96686ede1 llvm::TableGenMain(char*, bool (*)(llvm::raw_ostream&, llvm::RecordKeeper&)) <dir>/llpc//llvm/lib/TableGen/Main.cpp:109:0
#21 0x000055e966766324 main <dir>/llpc//llvm/utils/TableGen/TableGen.cpp:229:0
#22 0x00007fe0bb4c92b1 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x202b1)
#23 0x000055e9664bc02a _start (<dir>/llpc//bld/bin/llvm-tblgen+0x26302a)
Stack dump:
0.      Program arguments: <dir>/llpc//bld/bin/llvm-tblgen -gen-global-isel -I <dir>/llpc//llvm/lib/Target/AMDGPU -I <dir>/llpc//llvm/include -I <dir>/llpc//llvm/lib/Target <dir>/llpc//llvm/lib/Target/AMDGPU/AMDGPUGISel.td -o lib/Target/AMDGPU/AMDGPUGenGlobalISel.inc.tmp -d lib/Target/AMDGPU/AMDGPUGenGlobalISel.inc.d
[1573/1987] Building X86GenDAGISel.inc...
ninja: build stopped: subcommand failed.
Command exited with non-zero status 1
7126.86user 433.95system 3:12.85elapsed 3920%CPU (0avgtext+0avgdata 1451428maxresident)k
13760inputs+29478488outputs (10589major+123218555minor)pagefaults 0swaps
```

JacobHeAMD, 2018-06-05:
It's supposed to be fixed by commit: 0a811b8a5adf08602ef94f064dd0eb9a8aa9d546, please try the latest llvm.

dnovillo, 2018-06-07:
It's fixed now.  Thanks!

closed 2018-06-07 by dnovillo