GPUOpen-Drivers/AMDVLK #38: Wayland WSI is deadlocking in FIFO mode

Themaister, 2018-06-03:
I have a problem where vkAcquireNextImageKHR can indefinitely hang on Wayland.
This happens spuriously, and there is no obvious pattern to it.

When it happens, it's hanging on a semaphore wait indefinitely, waiting for an image to be fetched from the present thread.

There is a way to unblock the thread, which is to move the mouse, or any kind of windowing event seems to unblock it. This of course will never happen when you're playing with a joypad ... My theory is that the present thread is not polling for present complete events from Wayland or similar, and it only gets flushed handling on a mouse event or something like that.

I cannot seem to reproduce this with MAILBOX, so the logic might be a bit different.

To reproduce:
- Compile and run RetroArch with Wayland and Vulkan enabled.
- Play something using a gamepad.
- Make sure vsync is enabled.
- Find spurious hang in the game.
- Move the mouse and see that it unblocks the game.

To get a GDB backtrace of it, you'll need to keep the GDB terminal window in focus while playing, and when it hangs, ctrl+c it right away. Losing focus on the game window also unblocks the game apparently.

jfactory07, 2018-06-28:
Hi

What is the OS did you run RetroArch  ?

Themaister, 2018-06-28:
Arch Linux 64-bit, running Gnome 3.

jfactory07, 2018-07-26:
The issue should be fixed in latest wsa and pal, Please have a try.

Themaister, 2018-07-27:
Cool! I'll check it out when I can.

Themaister, 2018-08-06:
Using the latest drivers on Wayland (amdvlk + wsa) seems to take the entire system down. The entire screen freezes, and I need to hard reboot.

jfactory07, 2018-08-07:
Did you see any error information from dmesg when your screen freezes ?

Themaister, 2018-08-07:
[  118.606893] amdgpu 0000:26:00.0: GPU fault detected: 147 0x03488401
[  118.606898] amdgpu 0000:26:00.0:   VM_CONTEXT1_PROTECTION_FAULT_ADDR   0x070B0C69
[  118.606901] amdgpu 0000:26:00.0:   VM_CONTEXT1_PROTECTION_FAULT_STATUS 0x04084001
[  118.606904] amdgpu 0000:26:00.0: VM fault (0x01, vmid 2, pasid 32788) at page 118164585, read from 'TC7' (0x54433700) (132)

jfactory07, 2018-08-08:
It is GPU hang. It is different with the hang you got in vkAcquireNextImageKHR . How long did you run the RetroArch when you get this hang ? I will try to reproduce it by running RetroArch.

jfactory07, 2018-08-13:
I forgot to ask you which GPU did you use  ?

jfactory07, 2018-08-13:
I can reproduce the issue. I will fix it soon.

jfactory07, 2018-08-13:
hi,
The new GPU hang issue may have the root cause as issue 43 which is fixed in latest AMDVLK.
Without the fix, I can see random GPU hang and random segfault when start RetroArch.
Please help to verify it.

Themaister, 2018-08-19:
Seems to work now.

closed 2018-08-19 by Themaister