GPUOpen-Drivers/AMDVLK #39: dxvk: F.E.A.R. 3 black screen in menu

zenyd, 2018-06-09:
F.E.A.R 3 is only showing a black screen for the menu. But one can get into the story mode, where it only displays the environment, no characters, or HUD. The log shows "Failed to compile pipeline" errors.

I have taken two traces showing the problem: one captured with wine d3d and one with dxvk. The first can be replayed with amdvlk, but the second can't. It shows the menu correctly when replayed with radv.

[Trace 1 (wine d3d)](https://drive.google.com/open?id=1Nnz5UUBNOILHhPOyUlPHOJFNj4qUTfQM)
[Trace 2 (dxvk)](https://drive.google.com/open?id=1VSlUZ5oolm4vTaff8FufKkJaTMijSIiV)

Hardware is a Mullins based chip.

Logs:
[d3d11.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/2087393/d3dretrace_d3d11.log)
[dxgi.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/2087394/d3dretrace_dxgi.log)

AnzhongHuang, 2018-07-13:
@zenyd Thanks, it's a driver's bug. It has been fixed internally and the fix will be released later.

jinjianrong, 2018-07-20:
Please check the fix with https://github.com/GPUOpen-Drivers/llpc/commit/c0c713b36bbadfb2acccb996cc5d2c892eb05b1a

closed 2018-07-26 by jinjianrong