GPUOpen-Drivers/AMDVLK #42: vkcube segfault

zenyd, 2018-07-29:
vkcube segfaults when run:

`vkcube[1337]: segfault at 5577bd1a4018 ip 00007f5007953518 sp 00007ffcf1c4dbb8 error 4 in amdvlk64.so[7f50076e8000+22dd000]`

coredump:
[core.vkcube.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/2238840/core.vkcube.zip)

not only vkcube segfaults - dolphin emulator and mpv also segfault at the same offset.

vkcube r119.0d57034
amdvlk r40.48d46fa

jfactory07, 2018-07-31:
Hi, 
what is the GPU that you run vkcube ?

Zakhrov, 2018-08-01:
I'm having the same problem in vkcube and vkgears on a HD 7770 and a kabini apu

jfactory07, 2018-08-01:
Mullins is not supported by amdgpu KMD by default. To enable AMDVLK on Mulins, you need rebuild the kernel:
1. build the kernel with CONFIG_DRM_AMDGPU_SI enabled. 
2. add amdgpu.si_support=1 to the kernel parameters

Zakhrov, 2018-08-01:
Attaching screenshots
[image]
[image]

Command line output:
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/inc/util/palMemTrackerImpl.h:76:~MemTracker)
AMD-PAL: Warn: ================ List of Leaked Blocks ================ (/home/aaron/Projects/AMDVLK/drivers/pal/inc/util/palMemTrackerImpl.h:338:MemoryReport)
AMD-PAL: Warn: ClientMem = 0x0x7f659ce1b040, AllocSize =     1424, MemBlkType = New, File = /home/aaron/Projects/AMDVLK/drivers/pal/src/core/os/lnx/lnxVamMgr.cpp, LineNumber =      421, AllocNum =        1 (/home/aaron/Projects/AMDVLK/drivers/pal/inc/util/palMemTrackerImpl.h:349:MemoryReport)
AMD-PAL: Warn: ================ End of List =========================== (/home/aaron/Projects/AMDVLK/drivers/pal/inc/util/palMemTrackerImpl.h:352:MemoryReport)
1 physical devices
vendor id 1002, device name AMD Radeon HD 7700 Series
AMD-PAL: Info: Vulkan error: VK_TIMEOUT(2), from Pal error: Pal::Result::Timeout(3) (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/vk_conv.cpp:1010:PalToVkError)
AMD-PAL: Error: Assertion failed: (pDri3IdleFence == nullptr) || (m_dri3Procs.pfnXshmfenceQuery(pDri3IdleFence->ShmFence()) == 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/os/lnx/dri3/dri3WindowSystem.cpp:717:Present)
[vulkaininfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2248167/vulkaininfo.txt)

jfactory07, 2018-08-01:
Did you see the crash is caused by :
PAL_ASSERT((pDri3IdleFence == nullptr) || (m_dri3Procs.pfnXshmfenceQuery(pDri3IdleFence->ShmFence()) == 0)); at dri3WindowSystem.cpp:717 ?

If yes, please try disable debug assert:
add "HardAssert,0" to /etc/amd/amdPalSettings.cfg

In the second screenshot vkcube rander correctly. Is there any issue ?

zenyd, 2018-08-01:
For me a RelWithDebInfo build segfaults, but a Debug build runs the demo. Will try with a Release build later today.

Zakhrov, 2018-08-01:
vkcube renders the cube but it doesn't spin. I am away from my main system now but this issue is on my Kabini based work laptop. 
failed to initialize wayland, falling back to xcb
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/aaron/Projects/C++/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:76:~MemTracker)
AMD-PAL: Warn: ================ List of Leaked Blocks ================ (/home/aaron/Projects/C++/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:338:MemoryReport)
AMD-PAL: Warn: ClientMem = 0x0x7f36f9e19040, AllocSize =     1424, MemBlkType = New, File = /home/aaron/Projects/C++/vulkandriver/drivers/pal/src/core/os/lnx/lnxVamMgr.cpp, LineNumber =      421, AllocNum =        1 (/home/aaron/Projects/C++/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:349:MemoryReport)
AMD-PAL: Warn: ================ End of List =========================== (/home/aaron/Projects/C++/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:352:MemoryReport)
AMD-PAL: Error: Unconditional Assert | Reason: Unknown (/home/aaron/Projects/C++/vulkandriver/drivers/pal/src/core/os/lnx/lnxDevice.cpp:1036:TranslateMemoryType)
AMD-PAL: Error: Unconditional Assert | Reason: Unknown (/home/aaron/Projects/C++/vulkandriver/drivers/pal/src/core/os/lnx/lnxDevice.cpp:1036:TranslateMemoryType)

zenyd, 2018-08-01:
My vkcube does also not spin, but it's also not showing any text!
[image]

debug output with backtrace:
```
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/inc/util/palMemTrackerImpl.h:76:~MemTracker)
AMD-PAL: Warn: ================ List of Leaked Blocks ================ (/home/zenyd/Downloads/amdvlk/drivers/pal/inc/util/palMemTrackerImpl.h:338:MemoryReport)
AMD-PAL: Warn: ClientMem = 0x0x7fffefc19040, AllocSize =     1424, MemBlkType = New, File = /home/zenyd/Downloads/amdvlk/drivers/pal/src/core/os/lnx/lnxVamMgr.cpp, LineNumber =      421, AllocNum =        1 (/home/zenyd/Downloads/amdvlk/drivers/pal/inc/util/palMemTrackerImpl.h:349:MemoryReport)
AMD-PAL: Warn: ================ End of List =========================== (/home/zenyd/Downloads/amdvlk/drivers/pal/inc/util/palMemTrackerImpl.h:352:MemoryReport)
AMD-PAL: Error: Unconditional Assert | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/src/core/os/lnx/lnxDevice.cpp:1036:TranslateMemoryType)
AMD-PAL: Error: Unconditional Assert | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/src/core/os/lnx/lnxDevice.cpp:1036:TranslateMemoryType)
1 physical devices
vendor id 1002, device name Unknown AMD GPU
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:483:ExpandColor)
[New Thread 0x7fffee899700 (LWP 31179)]
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:483:ExpandColor)
AMD-PAL: Info: Vulkan error: VK_TIMEOUT(2), from Pal error: Pal::Result::Timeout(3) (/home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/vk_conv.cpp:1010:PalToVkError)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:483:ExpandColor)
AMD-PAL: Error: Assertion failed: (pDri3IdleFence == nullptr) || (m_dri3Procs.pfnXshmfenceQuery(pDri3IdleFence->ShmFence()) == 0) | Reason: Unknown (/home/zenyd/Downloads/amdvlk/drivers/pal/src/core/os/lnx/dri3/dri3WindowSystem.cpp:717:Present)
^C
Thread 1 "vkcube" received signal SIGINT, Interrupt.
0x00007ffff6ea4e47 in sched_yield () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff6ea4e47 in sched_yield () from /usr/lib/libc.so.6
#1  0x00007ffff2501e49 in Util::YieldThread ()
    at /home/zenyd/Downloads/amdvlk/drivers/pal/src/util/lnx/lnxMutex.cpp:162
#2  0x00007ffff252f11f in Pal::Queue::WaitIdle (this=0x555558e88310)
    at /home/zenyd/Downloads/amdvlk/drivers/pal/src/core/queue.cpp:610
#3  0x00007ffff11427a6 in vk::Queue::WaitIdle (this=0x555558cf14f0)
    at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/vk_queue.cpp:405
#4  0x00007ffff114440d in vk::entry::vkQueueWaitIdle (queue=0x555558cf14e8)
    at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/vk_queue.cpp:1189
```

The Release build is causing a segmentation fault:
`Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3bc49c8 in vk::ShaderOptimizer::ApplyProfileToShaderCreateInfo(vk::PipelineProfile const&, vk::PipelineOptimizerKey const&, vk::ShaderStage, vk::PipelineShaderOptionsPtr) () from /usr/lib/amdvlk64.so
`

jfactory07, 2018-08-02:
I see. Will fix it soon.

jfactory07, 2018-08-03:
it is a bug in vkcube.  I am trying to submit a commit to fix it.
https://github.com/krh/vkcube/pull/31

zenyd, 2018-08-03:
I tested your commit and it helped with the frozen cube. It's spinning now for the Debug build at least. The Release build still is segfaulting though:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3cc0c95 in vk::ShaderOptimizer::ApplyProfileToShaderCreateInfo (this=<optimized out>, profile=..., pipelineKey=..., 
    shaderStage=<optimized out>, options=...) at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/app_shader_optimizer.cpp:323
323	bool ShaderOptimizer::ProfilePatternMatchesPipeline(
(gdb) where
#0  0x00007ffff3cc0c95 in vk::ShaderOptimizer::ApplyProfileToShaderCreateInfo (this=<optimized out>, profile=..., pipelineKey=..., 
    shaderStage=<optimized out>, options=...) at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/app_shader_optimizer.cpp:323
#1  0x00007ffff3d06e70 in vk::ShaderOptimizer::OverrideShaderCreateInfo (options=..., shaderStage=vk::ShaderStageVertex, 
    pipelineKey=..., this=0x555556f97570) at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/app_shader_optimizer.cpp:136
#2  vk::PipelineCompiler::ApplyProfileOptions (pProfileKey=0x7fffffffd128, pShaderInfo=0x7fffffffcf00, pShaderModule=<optimized out>, 
    stage=vk::ShaderStageVertex, pDevice=0x555556f970c8, this=0x5555558d5ec0)
    at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/pipeline_compiler.cpp:1085
#3  vk::PipelineCompiler::ConvertGraphicsPipelineInfo (pVbInfo=0x7fffffffcdd0, pCreateInfo=0x7fffffffcee0, pIn=0x7fffffffddc0, 
    pDevice=0x555556f970c8, this=0x5555558d5ec0) at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/pipeline_compiler.cpp:921
#4  vk::GraphicsPipeline::Create (pDevice=<optimized out>, pPipelineCache=<optimized out>, pCreateInfo=<optimized out>, 
    pAllocator=<optimized out>, pPipeline=<optimized out>)
    at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/vk_graphics_pipeline.cpp:625
#5  0x00007ffff3c92706 in vk::Device::CreateGraphicsPipelines (pPipelines=<optimized out>, pAllocator=0x55555586f5c8, 
    pCreateInfos=<optimized out>, count=<optimized out>, pipelineCache=0x0, this=0x555556f970c8)
    at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/vk_device.cpp:1674
#6  vk::entry::vkCreateGraphicsPipelines (device=<optimized out>, pipelineCache=<optimized out>, createInfoCount=<optimized out>, 
    pCreateInfos=<optimized out>, pAllocator=<optimized out>, pPipelines=<optimized out>)
    at /home/zenyd/Downloads/amdvlk/drivers/xgl/icd/api/vk_device.cpp:2468
#7  0x000055555555bc53 in init_cube (vc=0x7fffffffe1f0) at ../cube.c:147
#8  0x0000555555558be8 in init_vk_objects (vc=0x7fffffffe1f0) at ../main.c:222
#9  0x000055555555a072 in init_xcb (vc=0x7fffffffe1f0) at ../main.c:857
#10 0x000055555555b113 in init_display (vc=0x7fffffffe1f0) at ../main.c:1345
#11 0x000055555555b3bd in main (argc=1, argv=0x7fffffffe728) at ../main.c:1416
```

jfactory07, 2018-08-06:
I have tried Release and RelWithDebInfo. But I can't reproduce the segment fault. Did you try to build it on a clean directory. It seems there is old .obj file when you build AMDVLK. So I think your build is incorrect.

On the other hand,  vk::ShaderOptimizer::ApplyProfileToShaderCreateInfo should return at the begin of it since profile.entryCount is 0. Can you try to add return at the begin of  vk::ShaderOptimizer::ApplyProfileToShaderCreateInfo.

Zakhrov, 2018-08-07:
It works successfully if we add return statements to the begining of
 `void ShaderOptimizer::ApplyProfileToShaderCreateInfo`
and 
`void ShaderOptimizer::ApplyProfileToGraphicsPipelineCreateInfo`
I'm guessing that vkcube doesn't like the optimized shader profiles. I'm no sure why it works with the debug build though. Maybe our shaders are not generating properly?

zenyd, 2018-08-07:
@Zakhrov those changes help workaround the crashes not only in vkcube, but also in dolphin emulator and mpv.

zenyd, 2018-08-08:
Hi, yes that fixes the problem!
But there remains further segmentation faults even with this fix. That is when running mpv (it used to segfault on start) it segfaults on closing the program. This also only happens with release builds but not with debug builds. To reproduce do `mpv --no-config --gpu-api vulkan <some video or image>`.

```
Thread 10 "mpv/vo" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd77fe700 (LWP 10132)]
0x00007fffd4f38b80 in ?? ()
(gdb) where
#0  0x00007fffd4f38b80 in ?? ()
#1  0x00007ffff0c5f1d1 in __nptl_deallocate_tsd.part.8 () from /usr/lib/libpthread.so.0
#2  0x00007ffff0c601c0 in start_thread () from /usr/lib/libpthread.so.0
#3  0x00007fffef75753f in clone () from /usr/lib/libc.so.6
```

jfactory07, 2018-08-13:
I can see mpv get segfault when it exit on both Debug and Release AMDVLK driver. I also checked the back trace when getting segfault. I don't see it happens in AMDVLK so I have to say it's mpv's issue instead of a bug in AMDVLK.

zenyd, 2018-08-13:
OK thanks for taking a look. I will check this with the mpv devs. Closing this issue as it's resolved.

closed 2018-08-13 by zenyd