GPUOpen-Drivers/AMDVLK #43: Failed to build AMDVLK with a missing file error

dnovillo, 2018-07-31:
I followed the instructions at https://github.com/GPUOpen-Drivers/AMDVLK#build-instructions. Specifically:

``` shell
cd <vulkandriver_path>/drivers/xgl
cmake -H. -Bbuilds/Release64
cd builds/Release64
make -j$(nproc)
```

Near the end of the build, I get the error:

```
[ ... ]
[ 99%] Building CXX object icd/CMakeFiles/xgl.dir/api/renderpass/renderpass_builder.cpp.o
[ 99%] Building CXX object icd/CMakeFiles/xgl.dir/api/renderpass/renderpass_logger.cpp.o
[ 99%] Building CXX object icd/CMakeFiles/xgl.dir/api/utils/json_reader.cpp.o
[ 99%] Building CXX object icd/CMakeFiles/xgl.dir/api/strings/strings.cpp.o
c++: error: /..../drivers/xgl/icd/api/strings/strings.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
icd/CMakeFiles/xgl.dir/build.make:1413: recipe for target 'icd/CMakeFiles/xgl.dir/api/strings/strings.cpp.o' failed
make[2]: *** [icd/CMakeFiles/xgl.dir/api/strings/strings.cpp.o] Error 1
CMakeFiles/Makefile2:17409: recipe for target 'icd/CMakeFiles/xgl.dir/all' failed
make[1]: *** [icd/CMakeFiles/xgl.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

Is there another repo that I should be getting?

Thanks.

JacobHeAMD, 2018-08-01:
The issue is not seen at my end. strings.cpp is generated by xgl/icd/api/strings/generate_strings.py while "make". Could you please help to check if the generate_strings.py running successfully at your end?

dnovillo, 2018-08-01:
Please see my last update.  The build only fails if you use the '-d' or '--debug' flag to 'make'.  Could you try reproducing by doing a fresh build with 'make -d'?

JacobHeAMD, 2018-08-04:
Sure, I've built it with "make -j64 --debug=j --debug=i" but the issue is not seen.
I checked the CMakeLists.txt again and looks like there is a bug in xgl/icd/CMakeLists.txt, could you please help to try the following patch?

diff --git a/icd/CMakeLists.txt b/icd/CMakeLists.txt
index 096b5fd..a7dd7bf 100644
--- a/icd/CMakeLists.txt
+++ b/icd/CMakeLists.txt
@@ -250,7 +250,7 @@ add_custom_command(

 add_custom_target(
     RunVKStringsGenerator
-    DEPENDS ${ICD_GEN_STRINGS_FILES} ${ICD_STRING_SOURCE_FILES}
+    DEPENDS ${ICD_GEN_STRINGS_FILES} ${ICD_STRING_SOURCE_FILES} ${ICD_STRING_DIR}/strings.cpp
     COMMENT "Checking if re-generation is required for strings"
 )

JacobHeAMD, 2018-08-07:
Hi @dnovillo, could you please reproduce this issue with the clean source code. Just realize that the content of strings.cpp is generated during compiling but string.cpp itself. I suspect that the strings.cpp is removed from your workspace by mistake. Could you please help to check it?

closed 2018-08-09 by JacobHeAMD