GPUOpen-Drivers/AMDVLK #44: Rise oft the tomb raider causes ring gfx timeout

Zakhrov, 2018-08-02:
launching Rise of the Tomb Raider with AMDVLK causes the following error in dmesg:

```
[ 1792.044917] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, last signaled seq=60221, last emitted seq=60223
[ 1792.044926] [drm] No hardware hang detected. Did some blocks stall?
```
I managed to get the following errors in the debug build in AMDVLK
```
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:237:GetDepthStencilBltPerSubres)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:237:GetDepthStencilBltPerSubres)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:237:GetDepthStencilBltPerSubres)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:237:GetDepthStencilBltPerSubres)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:237:GetDepthStencilBltPerSubres)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_po
```

My GPU is an AMD Radeon HD7770 / R7 250X (GFX6 Cape Verde) running with the AMDGPU driver using AMDVLK update 26/07/2018
Once the error is triggered, the computer stops responding and only a hard reboot works

Zakhrov, 2018-08-07:
The release build fails to launch the game. Steam reports that it is running and the process shows up in the task manager but I get no game window. Attached Screenshots:
[image]
[image]

jfactory07, 2018-08-07:
Would you mind to debug this issue ? It is strange  that release build fails to launch the game but debug build will have a gfx ring timeout issue.

jfactory07, 2018-08-20:
Thanks for your update ! It seems the latest code has fixed the issue you got when using release driver. I will check the new issue.

Zakhrov, 2018-09-01:
The code as of 31-08-2018 still stalls the GPU on both the debug and release drivers. The debug driver gives the following output

```
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:237:GetDepthStencilBltPerSubres)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:931:GetLayouts)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:931:GetLayouts)

```
And dmesg gives me the following:

```
[  326.026713] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=71543, emitted seq=71545
[  326.026723] [drm] GPU recovery disabled.

```

Zakhrov, 2018-10-27:
The newest code (2018-10-26) didn't lock up my system but it did cause the game to crash with the following errors:
```
AMD-PAL: Error: Assertion failed: pInfo->interpolatorCount >= 1 | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6PipelineChunkVsPs.cpp:124:EarlyInit)
AMD-PAL: Error: Assertion failed: endRegAddr >= startRegAddr | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/pal/src/core/hw/gfxip/gfx6/gfx6CmdUtil.cpp:2929:BuildSetSeqContextRegs)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:934:GetLayouts)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:934:GetLayouts)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:934:GetLayouts)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:934:GetLayouts)
RiseOfTheTombRaider: /home/aaron/Projects/AMDVLK/drivers/llpc/patch/llpcCodeGenManager.cpp:113: virtual bool Llpc::LlpcDiagnosticHandler::handleDiagnostics(const llvm::DiagnosticInfo&): Assertion `diagInfo.getSeverity() != DS_Error' failed.
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:934:GetLayouts)
AMD-PAL: Error: Assertion failed: (results[0].usages != 0) && (results[1].usages != 0) && (results[2].usages != 0) | Reason: Unknown (/home/aaron/Projects/AMDVLK/drivers/xgl/icd/api/barrier_policy.cpp:934:GetLayouts)

```
It seems to only affect GFX6 GPUs and RADV seems to work fine

closed 2020-08-19 by JacobHeAMD