GPUOpen-Drivers/AMDVLK #45: Dolphin with Super Mario Sunshine hits assertion

zenyd, 2018-08-09:
When running SMS in dolphin with a release build the rendering is incomplete (other games seem to work fine). The debug build hits an assertion :
```dolphin-emu: /home/zenyd/Downloads/amdvlk/drivers/llvm/lib/Target/AMDGPU/SIISelLowering.cpp:8368: llvm::SDNode* llvm::SITargetLowering::adjustWritemask(llvm::MachineSDNode*&, llvm::SelectionDAG&) const: Assertion `NewOpcode != -1 && NewOpcode != static_cast<int>(Node->getMachineOpcode()) && "failed to find equivalent MIMG op"' failed.```

To reproduce you need to run dolphin with [sms_fifo.dff.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/2275289/sms_fifo.dff.zip) and the following command:
`dolphin-emu -b -e sms_fifo.dff`

The game is working fine with radv.

[Backtrace](https://github.com/GPUOpen-Drivers/AMDVLK/files/2275301/sms_dolphin_bt.txt) of assert
Dolphin: dolphin-emu-git-5.0.r8518.d425732837
Hardware:Mullins

crystalJinamd, 2018-08-20:
@zenyd Can you help to dump pipelines and shader since I cannot reproduce the issue on my desktop ? Thanks!

The way to dump pipelines:
1. vi /etc/amd/amdPalSettings.cfg
2. add following lines:
EnablePipelineDump,1
PipelineDumpDir,<dump_dir_path>  (PipelineDumpDir is a sub-path relative to $AMD_DEBUG_DIR or $TMPDIR or /var/tmp/)

zenyd, 2018-08-20:
Hello @crystalJinamd, I am somehow not able to reproduce this issue myself anymore! I think it was something on my end, maybe a broken shader cache or something. Sorry for the noise!

closed 2018-08-20 by zenyd