GPUOpen-Drivers/AMDVLK #47: [Linux] AMVLK build fails with missing strings.cpp

shmerl, 2018-08-19:
I'm trying to build current AMDVLK on Debian testing. It gets to the point of building llpc, and somewhere in the middle of it, produces this:

```
[ 97%] Building CXX object icd/CMakeFiles/xgl.dir/api/app_profile.cpp.o
[ 97%] Building CXX object icd/CMakeFiles/xgl.dir/api/barrier_policy.cpp.o
[ 97%] Building CXX object icd/CMakeFiles/xgl.dir/api/app_shader_optimizer.cpp.o
[ 97%] Building CXX object icd/CMakeFiles/xgl.dir/api/strings/strings.cpp.o
[ 97%] Building CXX object icd/CMakeFiles/xgl.dir/api/gpu_event_mgr.cpp.o
c++: error: /home/user/build/amdvlk/drivers/xgl/icd/api/strings/strings.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [icd/CMakeFiles/xgl.dir/build.make:809: icd/CMakeFiles/xgl.dir/api/strings/strings.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:17377: icd/CMakeFiles/xgl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
Any idea what can be wrong?

shmerl, 2018-08-19:
OK, I noticed that git status shows strings.cpp as deleted, so something went wrong in your build scripts. I was able to work around it by doing:

```git checkout -- icd/api/strings/strings.cpp```

JacobHeAMD, 2018-08-20:
The issue is already fixed internally, will submit it to github soon. You can fix this by following patch for the time being,
diff --git a/icd/CMakeLists.txt b/icd/CMakeLists.txt
index eab15a96ad..33491383e1 100644
--- a/icd/CMakeLists.txt
+++ b/icd/CMakeLists.txt
@@ -371,7 +371,7 @@ endif()
 #endif

 add_custom_command(
-    OUTPUT ${ICD_STRING_OUTPUT_FILES} ${ICD_STRING_DIR}/strings.cpp
+    OUTPUT ${ICD_STRING_OUTPUT_FILES}
     COMMAND ${PYTHON_CMD} ${ICD_GEN_STRINGS} ${ICD_GEN_STRINGS_OPT}
     DEPENDS ${ICD_GEN_STRINGS_FILES} ${ICD_STRING_SOURCE_FILES}
     COMMENT "Generating Vulkan api strings ${ICD_GEN_STRINGS_OPT}"

Please checkout the strings.cpp first and then apply the patch.

JacobHeAMD, 2018-08-28:
Already fixed.

closed 2018-08-28 by JacobHeAMD