GPUOpen-Drivers/AMDVLK #48: Wayland WSI in FIFO is extremely jittery

Themaister, 2018-08-19:
I've tried to play some games with the Wayland backend, but it is not usable, the jitter is too extreme.
Very regularly, I see frame times of 30+ ms, only to be followed by a 2-3 ms frame time. It's like the acquire next image is held back through two vblanks for some reason. I suspect this has something to do with the locking in the present thread, which can end up holding for way too long.

This works fine in RADV.

To reproduce, try RetroArch and apply this patch:
```
diff --git a/gfx/common/vulkan_common.c b/gfx/common/vulkan_common.c
index 31c7f3d620..68a59eec0d 100644
--- a/gfx/common/vulkan_common.c
+++ b/gfx/common/vulkan_common.c
@@ -29,6 +29,7 @@
 
 #include "vulkan_common.h"
 #include "../../libretro-common/include/retro_timers.h"
+#include "../../libretro-common/include/features/features_cpu.h"
 #include "../../configuration.h"
 #include "../include/vulkan/vulkan.h"
 #include "../../libretro-common/include/retro_assert.h"
@@ -2351,7 +2352,9 @@ void vulkan_present(gfx_ctx_vulkan_data_t *vk, unsigned index)
 #ifdef HAVE_THREADS
    slock_lock(vk->context.queue_lock);
 #endif
+   RARCH_LOG("Queue present begin\n");
    err = vkQueuePresentKHR(vk->context.queue, &present);
+   RARCH_LOG("Queue present end\n");
 
 #ifdef WSI_HARDENING_TEST
    trigger_spurious_error_vkresult(&err);
@@ -2489,13 +2492,26 @@ retry:
 
    vkCreateFence(vk->context.device, &fence_info, NULL, &fence);
 
+   RARCH_LOG("Acquire image begin\n");
+   static double last_acquire;
    err = vkAcquireNextImageKHR(vk->context.device,
          vk->swapchain, UINT64_MAX,
          VK_NULL_HANDLE, fence, &vk->context.current_swapchain_index);
+   double current = cpu_features_get_time_usec() / 1000.0;
+   RARCH_LOG("Acquired image %u at: %12.3f ms\n", vk->context.current_swapchain_index, current);
 
    if (err == VK_SUCCESS)
       vkWaitForFences(vk->context.device, 1, &fence, true, UINT64_MAX);
 
+   current = cpu_features_get_time_usec() / 1000.0;
+   double delta = current - last_acquire;
+   if (last_acquire != 0.0 && delta < 4.0)
+      RARCH_WARN("[Vulkan]: Frame time is less than 4 ms! Weird jitter?\n");
+   else if (last_acquire != 0.0 && delta > 30.0)
+      RARCH_WARN("[Vulkan]: Frame time is more than 30 ms! Weird jitter?\n");
+   RARCH_LOG("Acquired image %u completed at: %12.3f ms\n", vk->context.current_swapchain_index, current);
+   last_acquire = current;
+
 #ifdef WSI_HARDENING_TEST
    trigger_spurious_error_vkresult(&err);
 #endif
```
I also hacked WSA for some more logging:
```
diff --git a/wayland/src/wsa_wayland.c b/wayland/src/wsa_wayland.c
index aa106f1..4f3db24 100644
--- a/wayland/src/wsa_wayland.c
+++ b/wayland/src/wsa_wayland.c
@@ -395,12 +395,30 @@ static void WaylandDestroyImage(
     pthread_mutex_unlock(&s_mutex);
 }
 
+static double get_time(void)
+{
+   struct timespec ts;
+   clock_gettime(CLOCK_MONOTONIC, &ts);
+   return ts.tv_nsec * 1e-9 + ts.tv_sec;
+}
+
+static void log_time(const char *str, ...)
+{
+   va_list va;
+   va_start(va, str);
+   char buffer[64];
+   vsprintf(buffer, str, va);
+   fprintf(stderr, "Time %10s: %12.3f ms\n", buffer, 1000.0 * get_time());
+   va_end(va);
+}
+
 // Present.
 static WsaError WaylandPresent(
     int32          hWsa,
     int32          hImage,
     WsaRegionList* presentRegions)
 {
+   log_time("Present enter %d", hImage + 3 - s_surfaceSlot);
     WSA_ASSERT((hWsa >= 0) && (hWsa < MAX_INSTANCE_NUMBER) && s_instanceList[hWsa].initialized);
     WSA_ASSERT((hImage < MAX_SURFACE_NUMBER) && (s_surfaceList[hImage].pSurface != NULL));
 
@@ -423,6 +441,7 @@ static WsaError WaylandPresent(
     wl_display_flush(s_instanceList[hWsa].pDisplay);
 
     pthread_mutex_unlock(&s_mutex);
+   log_time("Present leave");
 
     return Success;
 }
@@ -431,6 +450,7 @@ static WsaError WaylandPresent(
 static WsaError WaylandWaitForLastImagePresented(
     int32 hWsa)
 {
+   log_time("WaitForLastImage enter");
     WSA_ASSERT((hWsa >= 0) && (hWsa < MAX_INSTANCE_NUMBER) && s_instanceList[hWsa].initialized);
 
     int32 ret = 0;
@@ -450,6 +470,7 @@ static WsaError WaylandWaitForLastImagePresented(
     }
     pthread_mutex_unlock(&s_mutex);
 
+   log_time("WaitForLastImage leave");
     return Success;
 }
 
@@ -458,11 +479,14 @@ static WsaError WaylandImageAvailable(
     int32 hWsa,
     int32 hImage)
 {
+   int32 img = hImage + 3 - s_surfaceSlot;
+   log_time("ImageAvailable enter %d", img);
     WSA_ASSERT((hWsa >= 0) && (hWsa < MAX_INSTANCE_NUMBER) && s_instanceList[hWsa].initialized);
     WSA_ASSERT((hImage >= 0) && (hImage < MAX_SURFACE_NUMBER));
 
     if (s_surfaceList[hImage].busy == false)
     {
+   log_time("ImageAvailable leave (exit 0)");
         return Success;
     }
 
@@ -473,6 +497,7 @@ static WsaError WaylandImageAvailable(
     if (ret < 0)
     {
         pthread_mutex_unlock(&s_mutex);
+   log_time("ImageAvailable leave (exit 1)");
         return UnknownFailure;
     }
 
@@ -481,21 +506,25 @@ static WsaError WaylandImageAvailable(
     if (ret < 0)
     {
         pthread_mutex_unlock(&s_mutex);
+   log_time("ImageAvailable leave (exit 2)");
         return UnknownFailure;
     }
     ret = wl_display_dispatch_queue_pending(s_instanceList[hWsa].pDisplay, s_instanceList[hWsa].pQueue);
     if (ret < 0)
     {
         pthread_mutex_unlock(&s_mutex);
+   log_time("ImageAvailable leave (exit 3)");
         return UnknownFailure;
     }
     if (s_surfaceList[hImage].busy == false)
     {
         pthread_mutex_unlock(&s_mutex);
+   log_time("ImageAvailable leave (exit 4)");
         return Success;
     }
     pthread_mutex_unlock(&s_mutex);
 
+   log_time("ImageAvailable leave (exit 5)");
     return ResourceBusy;
 }
```
Some log output is:
```
[INFO] Acquired image 0 at:  8233988.892 ms
[INFO] Acquired image 0 completed at:  8233988.920 ms
[INFO] Queue present begin
[INFO] Queue present end
Time WaitForLastImage leave:  8234005.227 ms
[INFO] Acquire image begin
Time Present enter 2:  8234005.249 ms
Time ImageAvailable enter 1:  8234005.253 ms
Time Present leave:  8234005.261 ms
Time WaitForLastImage enter:  8234005.264 ms
Time ImageAvailable leave (exit 4):  8234005.337 ms
[INFO] Acquired image 1 at:  8234005.351 ms
[INFO] Acquired image 1 completed at:  8234005.553 ms
[INFO] Queue present begin
Time WaitForLastImage leave:  8234021.872 ms
[INFO] Queue present end
[INFO] Acquire image begin
Time Present enter 0:  8234021.895 ms <-- Two presents?
Time ImageAvailable enter 2:  8234021.895 ms
Time Present leave:  8234021.906 ms
Time WaitForLastImage enter:  8234021.909 ms
Time WaitForLastImage leave:  8234038.447 ms
Time Present enter 1:  8234038.477 ms <-- ?!
Time ImageAvailable leave (exit 4):  8234038.505 ms
Time Present leave:  8234038.513 ms
Time WaitForLastImage enter:  8234038.519 ms
[INFO] Acquired image 2 at:  8234038.546 ms
[WARN] [Vulkan]: Frame time is more than 30 ms! Weird jitter? <-- App is held back way too long.
[INFO] Acquired image 2 completed at:  8234038.778 ms
[INFO] Queue present begin
[INFO] Queue present end
[INFO] Acquire image begin
Time ImageAvailable enter 0:  8234039.068 ms
Time ImageAvailable leave (exit 0):  8234039.073 ms
[INFO] Acquired image 0 at:  8234039.086 ms
[WARN] [Vulkan]: Frame time is less than 4 ms! Weird jitter? <-- Next frame is way too fast.
[INFO] Acquired image 0 completed at:  8234039.113 ms
[INFO] Queue present begin
[INFO] Queue present end
[INFO] Acquire image begin
Time WaitForLastImage leave:  8234055.658 ms
Time Present enter 2:  8234055.683 ms
Time ImageAvailable enter 1:  8234055.687 ms
Time Present leave:  8234055.695 ms
Time WaitForLastImage enter:  8234055.698 ms
Time ImageAvailable leave (exit 4):  8234055.788 ms
[INFO] Acquired image 1 at:  8234055.804 ms
[INFO] Acquired image 1 completed at:  8234056.061 ms
Time WaitForLastImage leave:  8234071.781 ms
Time Present enter 0:  8234071.797 ms
Time Present leave:  8234071.806 ms
Time WaitForLastImage enter:  8234071.809 ms
```

Themaister, 2018-08-19:
To clarify, the GPU load of this application is virtually zero. It's just blitting a small frame to screen.

jfactory07, 2018-08-21:
hi,

I have played FloppyBird (https://github.com/RobLoach/ChaiLove-FloppyBird)  and I don't see the game is jittery. 

Which game did you try ? The issue may be specific to the game .

Themaister, 2018-08-21:
It's hard to notice any stuttery video because of the triple buffering, but audio underruns very often because of the jitter. Did you verify behavior with log or just visual?

jfactory07, 2018-09-03:
After trying your patch for RetroArch , I can see the jitter in log. But there is a small issue in the patch:
+   else if (last_acquire != 0.0 && delta > 30.0)
+      RARCH_WARN("[Vulkan]: Frame time is more than 30 ms! Weird jitter?\n");

I think the threshold should be 34, since 30 FPS is about 33.33 ms per frame.

I have commits for this issue, please help to verify it:
WSA: https://github.com/jfactory07/wsa/commit/b465901354e594bc4ec420d2983889648d5405d9
PAL: https://github.com/jfactory07/pal/commit/1f60b8f57dc62bdf324caac91bc68f77ddf4130e

Themaister, 2018-09-03:
- AMDVLK with your branches:
```
[INFO] [Video]: Average monitor Hz: 59.952038 Hz. (2.115 % frame time deviation, based on 2048 last samples).
```
- RADV:
```
[INFO] [Video]: Average monitor Hz: 59.952038 Hz. (2.262 % frame time deviation, based on 2048 last samples).
```
We have parity. Seems to run great now, both in windowed and fullscreen.

jfactory07, 2018-09-04:
Thanks for your test. We will merge the patch soon.

closed 2018-11-19 by jfactory07