GPUOpen-Drivers/AMDVLK #53: on tahiti doing a 32x16384 copy image on transfer queue can cause hangs

airlied, 2018-09-12:
I was looking into SDMA and Transfer queues and wrote some hacky crucible tests.

One of the tests creates a 32x16384 2D tiled image, and uses the transfer queue to copy it to another 32x16384 2D tiled image. On my tahiti GPU this seems to get some VM faults and a subsequent GPU hang.

airlied, 2018-09-12:
Even doing image->buffer copies appear to hang.

kbiElude, 2018-10-10:
Is that a fully-resident image? Any additional create flags that are specified for the image in question?

DominikWitczakAMD, 2018-10-12:
I can't reproduce the issue, I'm afraid. What I tried doing was an image->image copy of a R8_UNORM 32x16384 2D image using the transfer queue. Everything works fine.

Can you share an api dump / source code showing how to trigger the error, please? Thanks.

airlied, 2018-10-12:
I was using test code from

https://gitlab.freedesktop.org/airlied/crucible/commits/miptree-rework

The code is a bit rough still I'll try and retest and see if I can still reproduce

DominikWitczakAMD, 2018-10-12:
Did you mean this: https://gitlab.freedesktop.org/airlied/crucible/blob/miptree-rework/src/tests/func/transfer/transfer_buffer.c ? :-) Thanks.

DominikWitczakAMD, 2018-10-12:
In specific, please make sure no validation errors are being reported.

DominikWitczakAMD, 2018-10-12:
Hang on, I'm seeing something interesting. Which HW generation are you using?

airlied, 2018-10-12:
Yeah gcn 1 and 1.1 si/cik hwm and the miptree tests that us transfer queye

closed 2021-03-05 by jinjianrong