GPUOpen-Drivers/AMDVLK #54: Broken window content with Radeon R7 M440 and Linux 4.18.6

VadikLeshy, 2018-09-12:
This problem apears when I try to start any vulkan application.

Screenshots of vkmark:
https://imgur.com/a/bj985tC
https://imgur.com/a/DmSrb6B
https://imgur.com/a/VDPLp8e
https://imgur.com/a/2GzfkE0

Also amdvlk detects this GPU as "AMD Radeon (TM) R7 M340", but am not sure, if it's related.

jinjianrong, 2018-09-20:
@VadikLeshy, is amdgpu.si_support enabled in  kernel?

VadikLeshy, 2018-09-20:
Yes. It's enabled.

$ lspci | grep -i radeon
00:01.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Wani [Radeon R5/R6/R7 Graphics] (rev c9)
03:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445] (rev 83)
$ lspci -s 03:00.0 -k
03:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445] (rev 83)
        Subsystem: Acer Incorporated [ALI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445]
        Kernel driver in use: amdgpu
        Kernel modules: amdgpu

Also RADV works fine with my R7 M440.

VadikLeshy, 2018-09-20:
My hardware config is AMD A12 9700P (Carrizo APU) + Radeon R7 M440 (AMD Iceland). I'm not sure, but can this bug be related with PRIME offloading (eDP connected to Carrizo)? Cause APU works fine with amdvlk.

thirdeyefunction, 2018-10-07:
I also have the problem in a dual (AMD + AMD) GPU solution. The output GPU renders correctly, but the secondary GPU does not. It works fine with RADV, as expected.

YangtseSu, 2018-11-30:
> @VadikLeshy, is amdgpu.si_support enabled in kernel?

Yes,R5 M230

$ dmesg |grep amdgpu
[    0.000000] Command line: BOOT_IMAGE=/@/boot/vmlinuz-4.19-x86_64 root=UUID=f606d546-d214-465b-98d4-7f0f2384f49e rw rootflags=subvol=@ quiet radeon.si_support=0 amdgpu.si_support=1 amdgpu.dc=1
[    0.230521] Kernel command line: BOOT_IMAGE=/@/boot/vmlinuz-4.19-x86_64 root=UUID=f606d546-d214-465b-98d4-7f0f2384f49e rw rootflags=subvol=@ quiet radeon.si_support=0 amdgpu.si_support=1 amdgpu.dc=1

MojoJojoDojo, 2019-01-08:
My GPU is Iceland too (Radeon 530) .
 I get wrong graphics like in the pictures as well.

panic2k, 2019-07-09:
I've the issue here. Radeon 8690M as dGPU. the radeon module is blacklisted and kernel boot option radeon.si_support=0 amdgpu.si_support=1 are set. RADV runs fine so far.

01:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Sun XT [Radeon HD 8670A/8670M/8690M / R5 M330 / M430 / Radeon 520 Mobile]
	Subsystem: Dell Sun XT [Radeon HD 8670A/8670M/8690M / R5 M330 / M430 / Radeon 520 Mobile]

With RADV i got the same looking issue when choosing higher resolutions as my display supports (at downscaling 1080p to 900p in rpcs3). Maybe this can help anyone to figure it out

JacobHeAMD, 2019-10-28:
Could you please try the latest AMDVLK driver? It's supposed to be fixed now.

VadikLeshy, 2019-11-06:
Yes. This problem not apears with 2019.Q4.1

closed 2019-11-06 by VadikLeshy