GPUOpen-Drivers/AMDVLK #56: Segfault with O3 and O2

inferrna, 2018-10-04:
```
Ubuntu 18.04
linux kernel 4.19-rc6
mesa 18.21
GPU polaris11 (rx460)
CPU xeon 2665 (8/16)
```
Default build, with O3
```bash
(gdb) r
Starting program: /home/inferno/bin/vlksamples/gears 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3cecc4c in vk::Instance::LoadAndCommitSettings(unsigned int, Pal::IDevice**, vk::RuntimeSettings*, vk::AppProfile*) ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
(gdb) bt
#0  0x00007ffff3cecc4c in vk::Instance::LoadAndCommitSettings(unsigned int, Pal::IDevice**, vk::RuntimeSettings*, vk::AppProfile*) ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#1  0x00007ffff3cf7ccc in vk::PhysicalDeviceManager::UpdateLockedPhysicalDeviceList() ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#2  0x00007ffff3b3d27d in vk::Instance::Init(VkApplicationInfo const*) [clone .constprop.538] ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#3  0x00007ffff3ce90c7 in vk::Instance::Create(VkInstanceCreateInfo const*, VkAllocationCallbacks const*, VkInstance_T**) ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#4  0x00007ffff7992c72 in ?? () from /usr/lib/x86_64-linux-gnu/libvulkan.so.1
#5  0x00007ffff79a1fd2 in ?? () from /usr/lib/x86_64-linux-gnu/libvulkan.so.1
#6  0x00007ffff79a5f48 in vkCreateInstance () from /usr/lib/x86_64-linux-gnu/libvulkan.so.1
#7  0x0000555555572941 in VulkanExampleBase::createInstance(bool) ()
#8  0x0000555555574e34 in VulkanExampleBase::initVulkan() ()
#9  0x0000555555566929 in main ()
(gdb) quit
```
O2 seems different
```
(gdb) r
Starting program: /home/inferno/bin/vlksamples/gears 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3c7f630 in vk::PhysicalDeviceManager::UpdateLockedPhysicalDeviceList() () from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
(gdb) bt
#0  0x00007ffff3c7f630 in vk::PhysicalDeviceManager::UpdateLockedPhysicalDeviceList() ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#1  0x00007ffff3a227ca in vk::Instance::Init(VkApplicationInfo const*) [clone .constprop.545] ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#2  0x00007ffff3c6cde5 in vk::Instance::Create(VkInstanceCreateInfo const*, VkAllocationCallbacks const*, VkInstance_T**) ()
   from /home/inferno/dev/vulkandriver/drivers/xgl/builds/Release64/icd/amdvlk64.so
#3  0x00007ffff7992c72 in ?? () from /usr/lib/x86_64-linux-gnu/libvulkan.so.1
#4  0x00007ffff79a1fd2 in ?? () from /usr/lib/x86_64-linux-gnu/libvulkan.so.1
#5  0x00007ffff79a5f48 in vkCreateInstance () from /usr/lib/x86_64-linux-gnu/libvulkan.so.1
#6  0x0000555555572941 in VulkanExampleBase::createInstance(bool) ()
#7  0x0000555555574e34 in VulkanExampleBase::initVulkan() ()
#8  0x0000555555566929 in main ()
(gdb) 
```

Release build with O1 and Debug runs fine. Debug build produces some additional output:
```bash
Debug64$ VK_ICD_FILENAMES=./amd_icd64.json ~/bin/vlksamples/gears 
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:76:~MemTracker)
AMD-PAL: Warn: ================ List of Leaked Blocks ================ (/home/inferno/dev/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:338:MemoryReport)
AMD-PAL: Warn: ClientMem = 0x0x55fe5508e0b0, AllocSize =     1424, MemBlkType = New, File = /home/inferno/dev/vulkandriver/drivers/pal/src/core/os/lnx/lnxVamMgr.cpp, LineNumber =      440, AllocNum =        1 (/home/inferno/dev/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:349:MemoryReport)
AMD-PAL: Warn: ================ End of List =========================== (/home/inferno/dev/vulkandriver/drivers/pal/inc/util/palMemTrackerImpl.h:352:MemoryReport)
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/os/lnx/lnxDevice.cpp:617:EarlyInit)
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/os/lnx/lnxDevice.cpp:617:EarlyInit)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
AMD-PAL: Warn: Alert triggered: earlyPhase == false | Reason: Unknown (/home/inferno/dev/vulkandriver/drivers/pal/src/core/hw/gfxip/gfx6/gfx6Barrier.cpp:485:ExpandColor)
```

My firmware:
```bash
$ md5sum /lib/firmware/amdgpu/polaris11_*
c7081692d6367d0090af9bdc52d51e0d  /lib/firmware/amdgpu/polaris11_ce_2.bin
d2586cba8f25c8d50bc8bb2a5919bbcb  /lib/firmware/amdgpu/polaris11_ce.bin
58632a3ab9edbc1b1b08397fa4be087f  /lib/firmware/amdgpu/polaris11_k_smc.bin
153136ef03712cbae3c782ffeeeb68a7  /lib/firmware/amdgpu/polaris11_mc.bin
10631a2dce918053e726f63b95cc7904  /lib/firmware/amdgpu/polaris11_me_2.bin
9e0286e905fadc3ed597199f7caf5af3  /lib/firmware/amdgpu/polaris11_me.bin
79c4b1a52e99563e5f3ea46f2bd6349c  /lib/firmware/amdgpu/polaris11_mec2_2.bin
fc267c85534d31d44b7085dd111bf090  /lib/firmware/amdgpu/polaris11_mec_2.bin
198736549193bc377acd51493fb06677  /lib/firmware/amdgpu/polaris11_mec2.bin
198736549193bc377acd51493fb06677  /lib/firmware/amdgpu/polaris11_mec.bin
b8912a056a5cefc64afb860510fdb18e  /lib/firmware/amdgpu/polaris11_pfp_2.bin
af00efbff2f718f5edd479d59877ce5d  /lib/firmware/amdgpu/polaris11_pfp.bin
f5359801e8381f369a4fb9952716b189  /lib/firmware/amdgpu/polaris11_rlc.bin
58f4605527d2a00b078770c3ed030787  /lib/firmware/amdgpu/polaris11_sdma1.bin
2f4cd546ad0bd4e719a0ec03bd2e6f39  /lib/firmware/amdgpu/polaris11_sdma.bin
1b03e5d9f108b35059726e221249aa50  /lib/firmware/amdgpu/polaris11_smc.bin
5ceb324fb62937ef9dbc6aeb795d6a2f  /lib/firmware/amdgpu/polaris11_smc_sk.bin
55a9110bf8f76a9857d001d9fa670ab7  /lib/firmware/amdgpu/polaris11_uvd.bin
6c19732fa7d1b1d1a0c024da9d63e472  /lib/firmware/amdgpu/polaris11_vce.bin
```

skeevy420, 2019-05-14:
It built fine for me today from master with -O2 and then -Ofast on Manjaro with gcc 8.3.0 and the -Ofast build plays the Hitman 2 DX11 version, using DXVK 1.2, within 1.5 FPS of RADV according to the Miami benchmark (both using a custom proton build based on wine-staging 4.8).  In game it plays great and I can't tell a difference from earlier RADV play-throughs.  On the benchmark, RADV reports 37.1 while AMDVLK reports 35.75.

I feel I should mention that I've never had an issue compiling with O2 and I've been doing that for over a year.  Just had a weird Ofast itch today and happened to see this issue.

Manjaro
kernel 5.1.1
gcc 8.3.0
amdvlk (5475d7df)
RX 580 4gb
Mesa 19.0.3
dual Xeon x5687 (8/16)
48gb ddr3 1333 ecc

JaxLinAMD, 2020-07-24:
looks fine :)

closed 2020-07-24 by JaxLinAMD