GPUOpen-Drivers/AMDVLK #58: amdvlk64.so: undefined symbol: _ZN2vk24OverrideDispatchTable_NDEPNS_13DispatchTableE

haasn, 2018-10-20:
I get this error from the loader when trying to load the built AMDVLK driver. Dunno what's going on here. This happens for a plain old in-tree build on current `master`, using the instructions from the README.

I tried the older tagged versions but they exhibit the exact same failure.

Full build log: https://0x0.st/sEGQ.txt
I'm using `-GNinja` and `ninja` in this log, but omitting this and using `make` instead does not solve the issue.

haasn, 2018-10-20:
Using `LD_PRELOAD` on a shared library that just exports this symbol appears to solve the issue.

JacobHeAMD, 2018-12-06:
It's expected to be fixed by the latest driver (Q3.4), please have a try.

closed 2018-12-06 by JacobHeAMD

haasn, 2018-12-06:
Confirmed working, thanks.