GPUOpen-Drivers/AMDVLK #59: Invalid identification

APakrohk, 2018-10-28:
### **System Details:**
* Laptop model(If you use a laptop): **==>>** ASUS N551ZU
* Operating System: ? **==>>** Gnu/Linux
* Distro(For GNU/Linux and BSD users): ? **==>>** Manjaro
* Desktop Environment(For GNU/Linux and BSD users): ? **==>>** KDE Plasma5
* AMDVLK Version: ? **==>>** 2.51.0830.g3775c7f-1
* How do you install AMDVLK ? (Repositories, source, ...) **==>>** Arch Repo (Extra) Sudo pacman -Sy amdvlk
----------------------------------------------------------------------------------------------------------
### VULKANINFO
[vulkaninfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2522387/vulkaninfo.txt)

neVERberleRfellerER, 2018-10-28:
Your AMDVLK package is not from Arch Repo (Extra) but from some Manjaro repo or other 3rd party repo. Arch have AMDVLK only in AUR as of now.
Problem with your vulkaninfo is probably caused by 32 bit AMDVLK being loaded into 64 bit vulkaninfo. Try adjusting VK_ICD_FILENAMES, e.g:
`VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/amd_icd64.json" vulkaninfo`
to force 64 bit AMDVLK. If it works, wrong default was selected for some reason.

JacobHeAMD, 2018-11-26:
@neVERberleRfellerER is right! Close it.

closed 2018-11-26 by JacobHeAMD