GPUOpen-Drivers/AMDVLK #60: Errors with multiple monitors and direct display

robwyatt, 2018-11-07:
vkGetPhysicalDeviceDisplayPropertiesKHR() only fills the resulting array for the first monitor but it returns the correct number of monitors. 

The problem is in the code below, 'count' is never incremented so on the second or later times around the loop 'count' is still zero and the code will always overwrite the first monitor in the result array. 

pAttachedScreens is indexed by 'i', the output array is indexed by 'count' and count is alwasy zero.

VkResult PhysicalDevice::GetDisplayProperties(
    uint32_t*                                   pPropertyCount,
    utils::ArrayView<VkDisplayPropertiesKHR>    properties)
{
    uint32_t  screenCount   = 0;
    uint32_t  count         = 0;

    Pal::IScreen*   pScreens      = nullptr;
    uint32_t        propertyCount = *pPropertyCount;

    if (properties.IsNull())
    {
        VkInstance()->FindScreens(PalDevice(), pPropertyCount, nullptr);
        return VK_SUCCESS;
    }

    Pal::IScreen* pAttachedScreens[Pal::MaxScreens];

    VkResult result = VkInstance()->FindScreens(PalDevice(), &propertyCount, pAttachedScreens);

    uint32_t loopCount = Util::Min(*pPropertyCount, propertyCount);

    for (uint32_t i = 0; i < loopCount; i++)
    {
        Pal::ScreenProperties props = {};

        pAttachedScreens[i]->GetProperties(&props);

        properties[count].display = reinterpret_cast<VkDisplayKHR>(pAttachedScreens[i]);
        properties[count].displayName = nullptr;
        properties[count].physicalDimensions.width  = props.physicalDimension.width;
        properties[count].physicalDimensions.height = props.physicalDimension.height;
        properties[count].physicalResolution.width  = props.physicalResolution.width;
        properties[count].physicalResolution.height = props.physicalResolution.height;
        properties[count].supportedTransforms       = VK_SURFACE_TRANSFORM_IDENTITY_BIT_KHR;
        properties[count].planeReorderPossible      = false;
        properties[count].persistentContent         = false;
    }

    *pPropertyCount = loopCount;

    return result;
}

JacobHeAMD, 2018-11-12:
Yes, you are right! Will fix it soon.

JacobHeAMD, 2018-11-12:
Thanks! @robwyatt

beniwtv, 2019-05-04:
Is that why direct display doesn't work in SteamVR with AMDVLK? If so, I can still reproduce this.

JacobHeAMD, 2019-05-05:
I don't think so if you are talking about https://github.com/GPUOpen-Drivers/AMDVLK/issues/3 . Please try to enable VK_EXT_display_surface_counter by AMDVLK_ENABLE_DEVELOPING_EXT="VK_EXT_display_surface_counter" for the steamvr issue

closed 2019-05-05 by JacobHeAMD