GPUOpen-Drivers/AMDVLK #61: ELEX with DXVK freeze at startup

nadrolinux, 2018-11-12:
Hello,
When I tried to run ELEX with DXVK game freeze at splash screen (it's really easy to reproduce, just run a game). I tested it on both Ubuntu 18.10 x64:
- ELEX from GoG + latest wine + DXVK 0.8-0.9
- ELEX from Steam + Proton 3.16-4
and Windows 10 x64:
- ELEX from GoG + DXVK 0.8-0.9 + Radeon Software Adrenalin Edition 18.10.x
- ELEX from Steam + DXVK 0.8-0.9 + Radeon Software Adrenalin Edition 18.10.x
but all the time with the same results. With RADV this game works without problems, but I want to check how performance looks with AMDVLK and use faster driver.
My PC: AMD Ryzen 5 1600X + 16GB RAM + Radeon RX580 8GB.
Best regards,

crystalJinamd, 2018-11-26:
@nadrolinux 
The hang issue should be fixed internally. You can try latest closed driver on amd.com. If you want to relace with AMDVLK open source driver, you can try it, there may be a crash issue that has been fixed and may be not public yet.

BTW, can you tell me how to do benchmark for this game ? Thanks!

nadrolinux, 2018-11-27:
@crystalJinamd 
This is great news. I'll download new drivers when those will be available. For a benchmark purposes I use this save game (paste it at 'STEAM_GAMES_PATH/steamapps/compatdata/411300/pfx/drive_c/users/steamuser/Saved Games/ELEX/SaveGames' on Linux (Proton) and 'C:\Users\username\Saved Games\ELEX\SaveGames' on Windows 10):
https://www.sendspace.com/file/yfex1m
Wait few seconds and observe a fps when a 'doctor' stay near a table (I have 55 FPS on Linux and 64 FPS on Windows in this place on ultra settings + FXAA T2X + 1920x1080) and next just move around a Tavar or any other place (this is a save game with completed all main quests and unlocked teleports) and check fps.

nadrolinux, 2018-11-30:
I don't see improvements in Radeon Software Adrenalin Edition 18.12.1 drivers and hope that next version will contain those fixes.

nadrolinux, 2019-01-08:
Today I updated AMDVLK to v-2019.Q1.1, however screen corruptions issue still exist.

nadrolinux, 2019-01-16:
With AMDVLK v-2019.Q1.2 almost everything works fine and performance is very good. I just found that eg. when you look partially on the sky and fast look around there are some glitches over a sky (like white triangles). It's not a bug which prevent gaming, however without them a would say that game works perfectly fine with this driver.

crystalJinamd, 2019-01-17:
@nadrolinux Could you provide a picture about the glitches issue, then we can try to reproduce it ? Thanks!

nadrolinux, 2019-01-17:
@crystalJinamd I prepared a video with those bugs: https://youtu.be/i2nQ5O9Gu3A

nadrolinux, 2019-02-26:
@crystalJinamd I tested v-2019.Q1.6 and this bug still exists

crystalJinamd, 2019-02-28:
@nadrolinux Thanks for your response, this glitches issue is under working.

nadrolinux, 2019-04-11:
@crystalJinamd Any progress about this issue?

crystalJinamd, 2019-04-12:
@nadrolinux It is fixed internally, not sure if it is synced to github. Maybe you can try with latest codes in GPUOpen-Drivers/xgl.

jinjianrong, 2019-04-12:
It might be fixed by https://github.com/GPUOpen-Drivers/xgl/commit/9795e006e9adb7b49b8abd32cd47c49b290fa3fc

nadrolinux, 2019-04-12:
Thanks for info, I'll check it today.

nadrolinux, 2019-04-13:
I tested v2019.Q2.1, but after that I checked that 9795e00 commit is missing in this release, so I'll wait for new release and check that again.

nadrolinux, 2019-04-22:
@crystalJinamd @jinjianrong With v2019.Q2.2 all seems to work fine. I added 'solved' prefix to issue title. Ticket should be closed. Thanks!

closed 2019-04-22 by nadrolinux