GPUOpen-Drivers/AMDVLK #62: Mainline FreeSync / VRR / Adaptive Sync support

ernstp, 2018-11-20:
Add support to AMDVLK for the new mainline FreeSync / VRR api?

Reference:
https://patchwork.freedesktop.org/series/49486/

https://patchwork.freedesktop.org/series/49491/

https://patchwork.freedesktop.org/series/51388/

shmerl, 2019-03-26:
+1. FreeSync support so far is missing both in radv and amdvlk. Also Wayland use case might need special considerations.

debianxfce, 2019-04-08:
There is a patch for the RADV freesync support, but non existent gtk4 applications prevents mainlining.
https://gitlab.freedesktop.org/mesa/mesa/merge_requests/596

viggy96, 2019-06-15:
RADV 19.1 now has FreeSync. Does AMDVLK have this yet?

shmerl, 2019-12-20:
Just tested with latest amdvlk - adaptive sync still doesn't work.

RarogCmex, 2020-02-17:
What about now?

shmerl, 2020-02-17:
I haven't test it lately. Does it work for you?

JacobHeAMD, 2020-08-19:
Please try with adding "EnableAdaptiveSync,1" to /etc/amd/amdPalSettings.cfg.

closed 2020-08-19 by JacobHeAMD