GPUOpen-Drivers/AMDVLK #64: GTA V flickering light at night and the whole system freeze at the benchmark

Alexander88207, 2018-11-24:
AMDVLK : 2.0.63
  Treiber: ATI Technologies Inc. Radeon RX 570 Series
    Treiberversion: 4.5.13540 Compatibility Profile Context 18.40.2.15
    OpenGL-Version: 4.5
    Desktopfarbtiefe: 24 Bit per Pixel
    Monitor-Aktualisierungsrate: 144 Hz

I cant provide an apitrace

During benchmark i have this issues and then the jet part comes the system starts to freeze. 
[image]
[image]
[image]

jinjianrong, 2018-11-26:
@Alexander88207, is this an OpenGL game but you run it on top of Vulkan? what middleware do you use?

Alexander88207, 2018-11-26:
Yeah, i use vulkan (dxvk) & proton 3.16-4 to play this game. i can test it later it in vanilla wine too if this is necessary.

jinjianrong, 2019-03-11:
@Alexander88207 looks like the hang issue can't be reproduced with latest driver any more, could you help confirm it?

Alexander88207, 2019-04-06:
https://www.youtube.com/watch?v=5mtKFQIBSvM

closed 2019-04-06 by Alexander88207

Alexander88207, 2019-04-07:
it is fixed