GPUOpen-Drivers/AMDVLK #67: vulkaninfo fails for AMDVLK with Vega M GL, working with RADV

rstrube, 2018-12-03:
Hello AMD!

I've been trying to get the AMDVLK driver working with my Vega M GL (Kaby Lake G i7-8705G) on my Dell XPS 9575.  I've installed a prepacked version of the library (for Manjaro Linux) which was built against commit 81fd878 (https://github.com/GPUOpen-Drivers/AMDVLK/commit/81fd87809330b5f4cb1f4fa81a3e80e3c0ba3605) which is slightly newer than your 2018.4.2 release.

I've completely uninstalled the other vulkan libraries (RADV and Intel) so I only have the following JSON files in my `/usr/share/vulkan/icd.d` directory:
- `amd_icd32.json`
- `amd_icd64.json`

When running `vulkaninfo` I get the following error:

```
==========
VULKANINFO
==========

Vulkan Instance Version: 1.1.85

ERROR: [Loader Message] Code 0 : /usr/lib32/amdvlk32.so: wrong ELF class: ELFCLASS32
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
Cannot create Vulkan instance.
/build/vulkan-tools/src/Vulkan-Tools/vulkaninfo/vulkaninfo.c:825: failed with VK_ERROR_INCOMPATIBLE_DRIVER
```
If I uninstall AMDVLK and reinstall RADV and Intel's vulkan driver `vulkaninfo` works correctly, and I can also use the `VK_ICD_FILENAMES` environment variable to specify if I want to use the Intel iGPU or the discrete Vega M GL.  The `DRI_PRIME` variable also affects which GPU is used, so starting an application with `DRI_PRIME=1` will cause the Vega M GL to be used.  Does AMDVLK not yet support PRIME (aka Hybrid) GPUs like the Vega M?

Thanks!

hustwarhd, 2018-12-03:
Could you please provide the dmesg?

rstrube, 2018-12-07:
Hi!

Sorry for the late reply.  Here's the dmesg for my system.  Note that I ran `vulkaninfo` before dumping dmesg.

```
[    0.000000] Linux version 4.19.6-1-MANJARO (builduser@development) (gcc version 8.2.1 20180831 (GCC)) #1 SMP PREEMPT Sat Dec 1 12:21:26 UTC 2018
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.19-x86_64 root=UUID=3b2133d2-3156-4ac2-ae14-df7dfbc1c769 rw quiet rd.udev.log-priority=3
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000386f7fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000386f8000-0x00000000386f8fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000386f9000-0x00000000386f9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000386fa000-0x0000000047e59fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000047e5a000-0x000000004821cfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000004821d000-0x0000000048269fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000004826a000-0x0000000048bb6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000048bb7000-0x0000000049531fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000049532000-0x00000000495fefff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000495ff000-0x00000000495fffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000049600000-0x000000004f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000004ae7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ACPI=0x4822c000  ACPI 2.0=0x4822c000  SMBIOS=0xf0000  SMBIOS 3.0=0xf0020  ESRT=0x49254018  MEMATTR=0x4565f018 
[    0.000000] SMBIOS 3.1.1 present.
[    0.000000] DMI: Dell Inc. XPS 15 9575/0C32VW, BIOS 1.2.0 10/10/2018
[    0.000000] tsc: Detected 3100.000 MHz processor
[    0.001125] tsc: Detected 3096.000 MHz TSC
[    0.001125] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001126] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001131] last_pfn = 0x4ae800 max_arch_pfn = 0x400000000
[    0.001135] MTRR default type: write-back
[    0.001135] MTRR fixed ranges enabled:
[    0.001136]   00000-9FFFF write-back
[    0.001137]   A0000-BFFFF uncachable
[    0.001137]   C0000-FFFFF write-protect
[    0.001138] MTRR variable ranges enabled:
[    0.001139]   0 base 0080000000 mask 7F80000000 uncachable
[    0.001139]   1 base 0060000000 mask 7FE0000000 uncachable
[    0.001140]   2 base 0050000000 mask 7FF0000000 uncachable
[    0.001140]   3 base 004C000000 mask 7FFC000000 uncachable
[    0.001141]   4 base 004B000000 mask 7FFF000000 uncachable
[    0.001141]   5 base 2000000000 mask 7000000000 uncachable
[    0.001142]   6 disabled
[    0.001142]   7 disabled
[    0.001142]   8 disabled
[    0.001143]   9 disabled
[    0.001916] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002064] last_pfn = 0x49600 max_arch_pfn = 0x400000000
[... 967 lines cut ...]
[   14.785902] wlp2s0: RX AssocResp from 88:3d:24:75:b2:67 (capab=0x1011 status=0 aid=1)
[   14.788434] wlp2s0: associated
[   14.788578] ath: EEPROM regdomain: 0x8348
[   14.788581] ath: EEPROM indicates we should expect a country code
[   14.788584] ath: doing EEPROM country->regdmn map search
[   14.788587] ath: country maps to regdmn code: 0x3a
[   14.788590] ath: Country alpha2 being used: US
[   14.788592] ath: Regpair used: 0x3a
[   14.788595] ath: regdomain 0x8348 dynamically updated by country element
[   14.824306] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[   14.877495] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 88:3d:24:75:b2:67
[   15.428677] NET: Registered protocol family 38
[   16.181375] input: MX Anywhere 2 Keyboard as /devices/virtual/misc/uhid/0005:046D:B018.0008/input/input33
[   16.181572] input: MX Anywhere 2 Mouse as /devices/virtual/misc/uhid/0005:046D:B018.0008/input/input34
[... cut ...]
[   26.009534] audit: type=1131 audit(1544157769.690:61): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   32.419204] audit: type=1131 audit(1544157776.100:62): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   62.704880] [drm] PCIE GART of 256M enabled (table at 0x000000F400000000).
[   62.907829] [drm] UVD and UVD ENC initialized successfully.
[   63.018765] [drm] VCE initialized successfully.
[   68.783416] amdgpu 0000:01:00.0: GPU pci config reset
```

akien-mga, 2019-09-24:
I have a similar issue on similar hardware, also Vega M GL on an i7-8705G:
```
$ inxi -CG
CPU:       Topology: Quad Core model: Intel Core i7-8705G bits: 64 type: MT MCP L2 cache: 8192 KiB 
           Speed: 2819 MHz min/max: 800/4100 MHz Core speeds (MHz): 1: 2819 2: 2490 3: 2045 4: 2211 5: 2466 6: 2214 7: 1994 
           8: 2662 
Graphics:  Device-1: Intel HD Graphics 630 driver: i915 v: kernel 
           Device-2: Advanced Micro Devices [AMD/ATI] Polaris 22 XL [Radeon RX Vega M GL] driver: amdgpu v: kernel 
           Display: x11 server: Mageia X.org 1.20.4 driver: amdgpu,intel FAILED: ati unloaded: fbdev,modesetting,vesa tty: N/A 
           OpenGL: renderer: Mesa DRI Intel HD Graphics 630 (Kaby Lake GT2) v: 4.5 Mesa 19.1.7
```

I'm running amdvlk on Mageia 7 x86_64 using the Fedora 30 SRPM from https://copr.fedorainfracloud.org/coprs/tkov/amdvlk/package/amdvlk-vulkan-driver/ which I rebuilt for my system (amdvlk-vulkan-driver-2.105-0.20190829.gita14d42d.fc30).

I get the following error:
```
$ VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd.x86_64.json vulkaninfo 
/home/iurt/rpmbuild/BUILD/Vulkan-Tools-sdk-1.1.106.0/vulkaninfo/vulkaninfo.c:923: failed with VK_ERROR_OUT_OF_HOST_MEMORY
```

The error is similar to #18, and the related #17 hints at a firmware issue, but that's for a different card.

I run:
```
$ uname -r
5.2.16-desktop-2.mga7
$ rpm -qa | grep kernel-firmware
kernel-firmware-nonfree-20190720-1.mga7.nonfree
kernel-firmware-20190603-1.mga
```

amdgpu otherwise seems to work decently with RADV, I've been running Vulkan applications successfully with it.

`dmesg` doesn't list much relevant (same output as @rstrube on the part where vulkaninfo is run):
```
[ 2211.591020] [drm] PCIE GART of 256M enabled (table at 0x000000F400000000).
[ 2211.705961] [drm] UVD and UVD ENC initialized successfully.
[ 2211.806889] [drm] VCE initialized successfully.
[ 2218.203060] amdgpu 0000:01:00.0: GPU pci config reset
```

Any advice on how to debug further?

akien-mga, 2019-09-24:
I think it might indeed be a firmware issue, but probably on the downstream side. I don't see `vegam*` firmware either in Mageia's `kernel-firmware` nor `kernel-firmware-nonfree` (both generated from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git). I'll investigate further.

akien-mga, 2019-09-24:
I extracted the `vegam*` firmware files from AMDGPU-PRO 19.30 (RHEL 8 package):
```
$ ll usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam*
-rw-r--r-- 1 akien akien   8852 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_ce.bin
-rw-r--r-- 1 akien akien  17044 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_me.bin
-rw-r--r-- 1 akien akien 262824 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_mec2.bin
-rw-r--r-- 1 akien akien 262824 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_mec.bin
-rw-r--r-- 1 akien akien  17044 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_pfp.bin
-rw-r--r-- 1 akien akien  16660 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_rlc.bin
-rw-r--r-- 1 akien akien  12692 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_sdma1.bin
-rw-r--r-- 1 akien akien  12692 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_sdma.bin
-rw-r--r-- 1 akien akien 130964 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_smc.bin
-rw-r--r-- 1 akien akien 375424 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_uvd.bin
-rw-r--r-- 1 akien akien 166816 Jul 25 08:52 usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_vce.bin
```

Their binary size is the same as those on `linux-firmware.git/HEAD` which I have installed already. (BTW `vegam` firmware has never been updated on `linux-firmware` since its initial commit: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/amdgpu)

The SHA256 sums also seem to be the same:
```
$ sha256sum usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam*
c0d4acb3f1bd1def58cee933103069ce2b6b6d9b6a63591bd811b51a05ac0d67  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_ce.bin
c63ac86b770df364dc23c50ed6360a4823ad0e37120df12e4fb74bf574e3b536  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_me.bin
97bd777ed7ad8cf6f2825dfdde1ed508c9f4b824db0c336b4e1d4ccc6c1136a5  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_mec2.bin
dbbe48008aaa667b64e2fbd7e93975200ae89b62aadaa6387b5ef14259701c20  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_mec.bin
3c088e0cf4dadfe067c525a63337387659e1d81d8a5896ba6b8cbc5b36762950  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_pfp.bin
c729f89ade4fcd05c390f7bbf1fcb93eda47ac6911027c707add844181004b47  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_rlc.bin
477f1eb3a796c1248f38fb98ff389e9bb49d7cae746556055484a60668f509a2  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_sdma1.bin
9d964f4896b6713a1e41b8eca5ef665598290e4cc801d58fb2257845b7112040  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_sdma.bin
d6d422d2717b59bb7e23597157c2d6978b066475b4872045aff4702f2fee95eb  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_smc.bin
e05664b587a1585d98db3097cf4a55119ee4af2da68e65d3d43b3f57065a95fb  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_uvd.bin
b063be1f8093b41f527eecc7c592ff92c06d1c9bd5d2b4560e9dcd065f49898a  usr/src/amdgpu-19.30-855429.el8/firmware/amdgpu/vegam_vce.bin

$ sha256sum /lib/firmware/amdgpu/vegam*
c0d4acb3f1bd1def58cee933103069ce2b6b6d9b6a63591bd811b51a05ac0d67  /lib/firmware/amdgpu/vegam_ce.bin
c63ac86b770df364dc23c50ed6360a4823ad0e37120df12e4fb74bf574e3b536  /lib/firmware/amdgpu/vegam_me.bin
97bd777ed7ad8cf6f2825dfdde1ed508c9f4b824db0c336b4e1d4ccc6c1136a5  /lib/firmware/amdgpu/vegam_mec2.bin
dbbe48008aaa667b64e2fbd7e93975200ae89b62aadaa6387b5ef14259701c20  /lib/firmware/amdgpu/vegam_mec.bin
3c088e0cf4dadfe067c525a63337387659e1d81d8a5896ba6b8cbc5b36762950  /lib/firmware/amdgpu/vegam_pfp.bin
c729f89ade4fcd05c390f7bbf1fcb93eda47ac6911027c707add844181004b47  /lib/firmware/amdgpu/vegam_rlc.bin
477f1eb3a796c1248f38fb98ff389e9bb49d7cae746556055484a60668f509a2  /lib/firmware/amdgpu/vegam_sdma1.bin
9d964f4896b6713a1e41b8eca5ef665598290e4cc801d58fb2257845b7112040  /lib/firmware/amdgpu/vegam_sdma.bin
d6d422d2717b59bb7e23597157c2d6978b066475b4872045aff4702f2fee95eb  /lib/firmware/amdgpu/vegam_smc.bin
e05664b587a1585d98db3097cf4a55119ee4af2da68e65d3d43b3f57065a95fb  /lib/firmware/amdgpu/vegam_uvd.bin
b063be1f8093b41f527eecc7c592ff92c06d1c9bd5d2b4560e9dcd065f49898a  /lib/firmware/amdgpu/vegam_vce.bin
```

So the issue might not be with the firmware version as in #18 - or if it is, it affects AMGPU PRO too and means that Vega M is not properly supported.

akien-mga, 2019-09-25:
For the reference, I updated my Vulkan SDK to 1.1.121.0, but same error:
```
VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd.x86_64.json vulkaninfo 
/home/akien/Projects/mageia/Checkout/_vulkan/vulkan-tools/BUILD/Vulkan-Tools-sdk-1.1.121.0/vulkaninfo/vulkaninfo.c:967: failed with VK_ERROR_OUT_OF_HOST_MEMORY
```

JacobHeAMD, 2020-08-19:
@akien-mga , can you check if "radeon" is loaded with "lsmod | grep radeon"? Can you paste the output of "ls /dev/dri/"?

akien-mga, 2020-08-19:
I just tried again with amdvlk-2020.Q3.3-el.x86_64.rpm:
```
$ VK_ICD_FILENAMES=/etc/vulkan/icd.d/amd_icd64.json vulkaninfo 
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
/home/iurt/rpmbuild/BUILD/Vulkan-Tools-sdk-1.2.131.1/vulkaninfo/vulkaninfo.h:371: failed with ERROR_INCOMPATIBLE_DRIVER
```
I didn't uninstall my working Mesa drivers, but I wouldn't expect them to conflict as I'm specifying the amdvlk ICD manually.

>  can you check if "radeon" is loaded with "lsmod | grep radeon"? Can you paste the output of "ls /dev/dri/"?

```
$ sudo lsmod | grep radeon
$ ls /dev/dri/
by-path/  card0  card1  renderD128  renderD129
```

```
$ inxi -vv -CSG
System:    Host: cauldron Kernel: 5.7.12-desktop-1.mga8 x86_64 bits: 64 Desktop: KDE Plasma 5.19.4 Distro: Mageia 8 mga8 
CPU:       Topology: Quad Core model: Intel Core i7-8705G bits: 64 type: MT MCP L2 cache: 8192 KiB 
           Speed: 1714 MHz min/max: 800/3100 MHz Core speeds (MHz): 1: 1714 2: 1891 3: 1915 4: 1453 5: 2784 6: 2467 7: 2645 
           8: 2183 
Graphics:  Device-1: Intel HD Graphics 630 driver: i915 v: kernel 
           Device-2: Advanced Micro Devices [AMD/ATI] Polaris 22 XL [Radeon RX Vega M GL] driver: amdgpu v: kernel 
           Device-3: Cheng Uei Precision Industry (Foxlink) HP Wide Vision FHD Camera type: USB driver: uvcvideo 
           Display: x11 server: Mageia X.org 1.20.8 driver: intel,v4l resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa Intel HD Graphics 630 (KBL GT2) v: 4.6 Mesa 20.1.5
```

jinjianrong, 2020-08-19:
No plan to support Vega M

closed 2020-08-19 by jinjianrong