GPUOpen-Drivers/AMDVLK #70: The Witcher 3 (Wine+dxvk) hangs the system with admvlk

shmerl, 2018-12-05:
Just tested The Witcher 3 using latest amdvlk release, since now it should support `VK_EXT_transform_feedback`. It hangs the system in the start menu (I had to do Alt+SysRq / REISUB).

OS: Debian testing, kernel 4.19.5
amdvlk: v-2018.Q4.3
GPU: Sapphire Pulse Vega 56.
Wine: 3.21
dxvk: latest master (post 0.93).

AnzhongHuang, 2018-12-05:
Do you means the VK_EXT_transform_feedback causes GPU hang? What about running this game without transform feedback?

shmerl, 2018-12-05:
I don't know what is causing the hang. It was hanging for me in the past as well, so the hang is not necessarily related to `VK_EXT_transform_feedback`. I just tested it now because of `VK_EXT_transform_feedback`.

AnzhongHuang, 2018-12-05:
Strange, the game works very well with transform feedback on my local, also on Vega, I guess some bugs were fixed internally.

shmerl, 2018-12-05:
The hang is erratic. One time I was able to load into the game and on second run it hang. Also it's random and happens in different places each time but pretty early on.

shmerl, 2018-12-12:
@AnzhongHuang : do you know when VK_EXT_transform_feedback will be released? Latest amdvlk update still doens't have it.

AnzhongHuang, 2018-12-13:
@shmerl It depends on CTS tests. The CTS tests for transform feedback are not released officially, before we release this extension we need to pass those CTS tests first.

shmerl, 2019-01-15:
Is there any ETA on release of the related tests?

shmerl, 2019-01-30:
Is VK_EXT_transform feedback now in the dev branch? I see this commit:
https://github.com/GPUOpen-Drivers/xgl/commit/6a629e8286ed9d2a8a3799131a274ee58590765d
> Implement VK_EXT_transform_feedback

But building amdvlk including the dev branch still produces it without the extension.

jinjianrong, 2019-01-31:
We are still working on some issues, so the extension has not been enabled by default. We will provide a way soon to enable the extension through environment variable

shmerl, 2019-02-01:
With latest release the game doesn't hang the system anymore, but after trying to load a save, it just freezes (the game itself). It's still possible to Alt+Tab out of it.

nadrolinux, 2019-02-06:
@shmerl 
I also noticed a game freeze few seconds after load a saved game. I have Radeon RX 580.

shmerl, 2019-02-06:
Not sure if it's dxvk or amdvlk issue at that point.

nadrolinux, 2019-02-06:
I think that this is AMDVLK issue, because with RadV all works fine.

shmerl, 2019-02-06:
It could be something in the way dxvk is using amdvlk, though I don't expect some specific incorrect usage.

shmerl, 2019-03-17:
The hang is gone in the latest release, so this can be closed.

closed 2019-03-17 by shmerl