GPUOpen-Drivers/AMDVLK #72: Garbled graphics

MojoJojoDojo, 2019-01-02:
GPU : Radeon 530
OS : Ubuntu 18.10 Budgie
Using the latest AMDVLK GPU driver (12.12 release) I get garbled graphics.
Here are the examples(I mostly use LInux for emulators):
Dolphin:
[image]
RPCS3:
[image]
Only the PPSSPP emulator worked correctly with the driver.
Thanks.

Alexander88207, 2019-01-03:
@MojoJojoDojo Mesa works?

MojoJojoDojo, 2019-01-08:
Yes RADV works but I wanted to do a comparison between them.

Cx64, 2019-01-24:
[image]

I've the very same problem with vulkan-smoketest :
04:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Mars LE [Radeon HD 8530M / R5 M240] (prog-if 00 [VGA controller])
	Subsystem: ASUSTeK Computer Inc. Mars LE [Radeon HD 8530M / R5 M240]
	Flags: bus master, fast devsel, latency 0, IRQ 44
	Memory at e0000000 (64-bit, prefetchable) [size=256M]
	Memory at f7c00000 (64-bit, non-prefetchable) [size=256K]
	I/O ports at d000 [size=256]
	Expansion ROM at f7c40000 [disabled] [size=128K]
	Kernel driver in use: amdgpu
Kernel 4.20.3-042003-generic #201901171122 SMP Thu Jan 17 16:25:26 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux.
What to do next ?
Thanks in advance!

MojoJojoDojo, 2019-01-24:
I wonder if thats why Vulkan stopped working on Windows as well.
I can only use it with my Intel GPU.

jinjianrong, 2019-01-25:
@MojoJojoDojo we can't reproduce the issue with RPCS3 emulator-  jojo bizarre adventure. could you provide renderdoc trace?

MojoJojoDojo, 2019-01-25:
@jinjianrong First time I do this i hope i did it right.
[jojo.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/2797398/jojo.zip)

Cx64, 2019-01-25:
@jinjianrong here my renderdoc trace for vulkan-smoketest, hope this help.
Thanks
luca
[vulkan-smoketestlogs.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/2797803/vulkan-smoketestlogs.zip)

jinjianrong, 2019-01-30:
@Cx64 does the vulkan-smoketestlogs.zip reproduce the issue on your system? it runs well on our system. could you check if the firmware needs to be updated on your system as talked here: https://github.com/GPUOpen-Drivers/AMDVLK/issues/50#issuecomment-441206916 ?

jinjianrong, 2019-01-30:
@Cx64 we see a black image with the jojo.zip trace, seems not same with the issue you reported. what do you see with the trace on your system?

MojoJojoDojo, 2019-01-30:
[image]

Cx64, 2019-01-30:
> @Cx64 does the vulkan-smoketestlogs.zip reproduce the issue on your system? it runs well on our system. could you check if the firmware needs to be updated on your system as talked here: [#50 (comment)](https://github.com/GPUOpen-Drivers/AMDVLK/issues/50#issuecomment-441206916) ?

Yes I'm able to reproduce the same problem as you can see from the screenshot.
I'm on Ubuntu 18.04 with kernel 4.20.5-042005-generic, amdvlk_2019.Q1.4, latest firmware, more detail on info.txt and kernel.log .
What I can do next ? 
Thanks in advance!
[image]
[info.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2811543/info.txt)
[kernel.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/2811544/kernel.log)

MojoJojoDojo, 2019-01-30:
I will give more info on my GPU as well:
lspci -vnn | grep -i AMD
01:00.0 Display controller [0380]: Advanced Micro Devices, Inc. [AMD/ATI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445] [1002:6900] (rev c1)
	Kernel driver in use: amdgpu
	Kernel modules: amdgpu

vulkaninfo | grep -i AMD
		GPU id       : 0 (AMD RADV ICELAND (LLVM 9.0.0))
		GPU id       : 2 (Unknown AMD GPU)
		GPU id       : 0 (AMD RADV ICELAND (LLVM 9.0.0))
		GPU id       : 2 (Unknown AMD GPU)
		GPU id       : 0 (AMD RADV ICELAND (LLVM 9.0.0))
		GPU id       : 2 (Unknown AMD GPU)
		GPU id       : 0 (AMD RADV ICELAND (LLVM 9.0.0))
		GPU id       : 2 (Unknown AMD GPU)
		GPU id       : 0 (AMD RADV ICELAND (LLVM 9.0.0))
		GPU id       : 2 (Unknown AMD GPU)
GPU id       : 0 (AMD RADV ICELAND (LLVM 9.0.0))
GPU id       : 2 (Unknown AMD GPU)
			AMD RADV ICELAND (LLVM 9.0.0) (ID: 0)
			Unknown AMD GPU (ID: 0)
	deviceName     = AMD RADV ICELAND (LLVM 9.0.0)
	VK_AMD_draw_indirect_count          : extension revision  1
	VK_AMD_gcn_shader                   : extension revision  1
	VK_AMD_shader_core_properties       : extension revision  1
	VK_AMD_shader_info                  : extension revision  1
	VK_AMD_shader_trinary_minmax        : extension revision  1
	deviceName     = Unknown AMD GPU
	VK_AMD_buffer_marker                : extension revision  1
	VK_AMD_draw_indirect_count          : extension revision  1
	VK_AMD_gcn_shader                   : extension revision  1
	VK_AMD_gpa_interface                : extension revision  1
	VK_AMD_memory_overallocation_behavior: extension revision  1
	VK_AMD_mixed_attachment_samples     : extension revision  1
	VK_AMD_negative_viewport_height     : extension revision  1
	VK_AMD_rasterization_order          : extension revision  1
	VK_AMD_shader_ballot                : extension revision  1
	VK_AMD_shader_core_properties       : extension revision  1
	VK_AMD_shader_explicit_vertex_parameter: extension revision  1
	VK_AMD_shader_fragment_mask         : extension revision  1
	VK_AMD_shader_image_load_store_lod  : extension revision  1
	VK_AMD_shader_info                  : extension revision  1
	VK_AMD_shader_trinary_minmax        : extension revision  1
	VK_AMD_texture_gather_bias_lod      : extension revision  1

Cx64, 2019-02-28:
Hi, I've just installed the latest version amdvlk_2019.Q1.6_amd64.deb but the problem still persist, mesa works ok. @jinjianrong any hope that this will be fixed anytime soon ?
Thanks!

MojoJojoDojo, 2019-03-01:
I think they won't fix it that easily because its specific to certain GPU's.
But i narrowed it down to Iceland or Radeon™ M200/M300/M400 Series.

jinjianrong, 2019-03-04:
Unfortunately, we have not reproduced the issue yet. 
@MojoJojoDojo according to the vulkaninfo you provided, looks like your system is running with both RADV and AMDVLK, could you remove RADV's JSON file and try again? 

@Cx64 could you provide the vulkaninfo log?

Cx64, 2019-03-04:
> Unfortunately, we have not reproduced the issue yet.
> @MojoJojoDojo according to the vulkaninfo you provided, looks like your system is running with both RADV and AMDVLK, could you remove RADV's JSON file and try again?
> 
> @Cx64 could you provide the vulkaninfo log?

Hi @jinjianrong 
you can find all the log here:
https://github.com/GPUOpen-Drivers/AMDVLK/issues/72#issuecomment-457719634

Let me if I can help further !
Thanks,
luca

MojoJojoDojo, 2019-03-07:
I think this issue is the same as issue #54

bobeick87, 2019-03-13:
Hi all,
I have a Radeon HD8670m (Southern Island, Hainan/Mars) and the AMDVLK produces the same kind of garbled graphics described here by you guys. It seems most of the driver works fine but just something messes up when the images get rendered producing a mess. In windows Vulkan works great with this card. Back to linux with Mesa the vulkan (intel) works ok but it's slow.

jinjianrong, 2019-03-14:
@bobeick87 are there two GPUs in your system: one intel's integrated GPU + one Radeon HD8670m? Does the issue happen when rendering with HD8670m but present on Intel's integrated GPU?

bobeick87, 2019-03-15:
Hi jinjianrong, yes I have the integrated intel 4400 HD (Core i5 4200U Haswell) and the discrete radeon HD8670M. I have compiled and build AMDVLK via the available instructions and seems fine. My system is Ubuntu 18.10 Kernel is 5.0.1, and am using AMDGPU kernel module. I tested Vulkan with the emulator Dolphin, and vulkan-cube (vulkan sdk x.101) and both render strange graphics and give no error messages. The rendering is fine if I install and run tests on the mesa-vulkan driver (which uses the intel integrated). It is a shame not to be able to use vulkan on the discrete.

MojoJojoDojo, 2019-03-15:
I am here to report that the latest Windows driver fixed Vulkan on my AMD GPU.
But now I cant access my Intel GPU's Vulkan capabilities  (not a big loss).
Now that its fixed on Windows I dont have to relay on Linux for the RADV driver but I will check if it works there later.

seesturm, 2019-03-28:
I have the same problem (garbled picture) on my host with two Vega56 GPUs. If I remove one of them vulkan-smoketest works.

Pik-a-boo, 2019-04-01:
Same problem as others - corrupted graphics .
I have a dual Intel-AMD GPU system.
Looks like the problem is with Hybrid Graphics?
From the last comment and other comments here.
RADV driver works as intended.

seesturm, 2019-04-11:
I managed to get AMDVLK working for me by means of putting into ~/.config/amdPalSettings.cfg:
`EnumPreferredDeviceIndex,0`
Don't know if this could help the users with hybrid graphics here since I have two identical cards (Vega56).

Pik-a-boo, 2019-04-12:
@seesturm putting what where?
Can you give more details?

seesturm, 2019-04-12:
@Pik-a-boo: I created a config file "amdPalSettings.cfg" in "/home/seesturm/.config" with a single line "EnumPreferredDeviceIndex,0". If there already happens to be a config file with the same name in your ".config" folder then you could add the line to the existing config file.

jozefkucia, 2019-05-05:
The same issue on machine with Intel HD Graphics 630 and Radeon RX Vega 56.

00:02.0 VGA compatible controller: Intel Corporation HD Graphics 630 (rev 04)
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XL/XT [Radeon RX Vega 56/64] (rev c3)

VadikLeshy, 2019-08-07:
I also have the same issue on my Laptop with AMD APU A12 + Rareon R7 M440.

$ lspci | grep -i radeon
00:01.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Wani [Radeon R5/R6/R7 Graphics] (rev c9)
03:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445] (rev 83)
$ lspci -s 03:00.0 -k
03:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445] (rev 83)
Subsystem: Acer Incorporated [ALI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445]
Kernel driver in use: amdgpu
Kernel modules: amdgpu

https://github.com/GPUOpen-Drivers/AMDVLK/issues/54

Pik-a-boo, 2019-10-01:
I cant believe this issue is still a thing.
You think it got fixed but than you find out its only for Navi GPUs.
No legacy support huh.

closed 2019-11-13 by MojoJojoDojo