GPUOpen-Drivers/AMDVLK #73: Artifacts with DXVK and Project Cars

random2324, 2019-01-06:
Trying Proton 3.16-6 with latest amdvlk from git produces this strange artifacts:
[image]

Vendor: X.Org (0x1002)
    Device: AMD Radeon HD 7900 Series (TAHITI, DRM 3.27.0, 4.20.0-2-amd-staging-drm-next-git, LLVM 8.0.0) (0x6798)
    Version: 19.0.0
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 4.5
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 2608 MB, largest block: 2608 MB
    VBO free aux. memory - total: 3014 MB, largest block: 3014 MB
    Texture free memory - total: 2608 MB, largest block: 2608 MB
    Texture free aux. memory - total: 3014 MB, largest block: 3014 MB
    Renderbuffer free memory - total: 2608 MB, largest block: 2608 MB
    Renderbuffer free aux. memory - total: 3014 MB, largest block: 3014 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 3072 MB
    Total available memory: 6144 MB
    Currently available dedicated video memory: 2608 MB

Radv just shows me a black screen btw.

random2324, 2019-02-02:
I tried out some more games. Hitman and Dirt 3. All of those including Project Cars show artifacts when I select  MSAA. This look like a general problem for me on those GCN 1.0 cards.

closed 2020-08-19 by JacobHeAMD

procedural, 2020-08-19:
> @JacobHeAMD added wontfix

Why is that? I saw MSAA artifacts on Tahiti on a recent version of AMDVLK too.

random2324, 2020-08-19:
Switched to Vega so I cant test anymore. Still should be fixed now that GCN1.0 is nearly ready for AMDGPU as default.