GPUOpen-Drivers/AMDVLK #74: Artifacts with DXVK and The Witcher 3

guglovich, 2019-01-08:
Proton 3.16-6 + AMDVLK
[https://ibb.co/6R0khXR](url)

Distr: Arch
Kernel: 4.20.0-arch1-1
GPU: RX460

shmerl, 2019-03-17:
Looks Polaris specific, since it doesn't happen on Vega 56 for me. Does it still occur with latest amdvlk?

jiaolu, 2019-03-18:
> Looks Polaris specific, since it doesn't happen on Vega 56 for me. Does it still occur with latest amdvlk?

the problem i met is rough edges of shadow miss-casted on the trees on vega.

aufkrawall, 2019-04-23:
There is still an issue with Polaris and v-2019.Q2.2 with certain foliage before some background objects like distant buildings, the foliage than shows what looks like blocky shadow corruption.

nadrolinux, 2019-05-27:
Today I tested The Witcher 3 with AMDVLK 2019.Q2.4 and finally it seems that all issues are gone. I have Radeon RX 580.

jinjianrong, 2019-05-28:
@nadrolinux thanks for the testing

closed 2019-05-28 by jinjianrong