GPUOpen-Drivers/AMDVLK #75: Path of Exile not rendering correctly

BrosephJ, 2019-01-09:
Xubuntu 18.10, Linux Kernel 4.20
Playing with Lutris, DXVK 0.94

RADV:

[image]

AMDVLK:

[image]

montvid, 2019-01-27:
got the same problem with the recent amdvlk 2019q1.3. Dxvk 0.96, mesa 19, kernel 4.20, ubuntu 18.04.

montvid, 2019-01-29:
Sure I can help out! It would be my first time generating a trace. Do you have a wiki on how to do it on Linux, Ubuntu?

It is a popular free game on steam as well on their site 20gb to download. Too dark rendering starts just after starting the game in the character selection screen so no need to play to debug.

montvid, 2019-01-29:
Oh and the game only starts only with tkg wine used by lutris.

Arvamer, 2019-03-10:
@jinjianrong  [RenderDoc capture for login screen](https://drive.google.com/open?id=1HkUXPrAR6LQF4gVcSokb326yjkxKiRkP).

crystalJinamd, 2019-03-12:
@Arvamer hi, Which renderdoc version and asic did you use to capture the trace ? Thanks.

Arvamer, 2019-03-12:
@crystalJinamd  I used renderdoc 1.2. Sorry I'm not sure what you meant by asic. I captured it on rx 580 using amdvlk built from git (GPUOpen-Drivers/xgl@f2af4b0c33963842f544107d005f0a6c82ea513f).

mflayer, 2019-03-19:
Hi! I'm having the same issue. It gets to the point where in the darker parts of the game like caves, everything becomes pitch black.
Relevant information: Arch on 5.0.2 kernel, latest amdvlk build from git (via AUR), Radeon R9 280 GPU, renderdoc 1.2-1, playing on Steam with Proton ver. 3.16-8 beta.

Here's a render doc capture from within one of those caves(first time doing it, let me know if you need anything else): [Link, 400MB compressed](https://drive.google.com/file/d/1uADXaAVMrafRw-dP-tnY1DGRLGDjiHH4/view?usp=sharing)

mflayer, 2019-03-19:
Figured this might be of some importance:
RenderDoc reports multiple issues (13 in my case), severity high. Description is always "Creating renderpass Resourceid::##### contains a subpass dependency that would allow writing indirect command arguments. Indirect command contents are read at the end of the render pass, so write-after-read overwrites will cause incorrect display of indirect arguments." where ##### is a number.
Don't know if this helps.

BrosephJ, 2019-03-20:
I did the heinous deed of going back to Windows 10 so I am unable to help with this. ;_; Sorry for late response!

ernstp, 2019-03-22:
Same issue here! Works quite well otherwise.

Arvamer, 2019-03-31:
I have updated to latest git and it looks like this issue is fixed now.

jinjianrong, 2019-04-01:
@Arvamer thanks for the update

closed 2019-04-01 by jinjianrong