GPUOpen-Drivers/AMDVLK #76: Problems with AMD Radeon Pro WX 5100 and blender (2.79b)

sosyco, 2019-01-12:
I have done a fresh installation of Kubuntu 18.04 and the drivers:
`wget https://github.com/GPUOpen-Drivers/AMDVLK/releases/download/v-2019.Q1.1/amdvlk_2019.Q1.1_amd64.deb
sudo dpkg -i amdvlk_2019.Q1.1_amd64.deb
sudo aptitude install mesa-opencl-icd clinfo`

the base-results shows little bit less performance with vkmark: (what shalls!)
[https://openbenchmarking.org/result/1901117-SK-VKMARK19264](https://openbenchmarking.org/result/1901117-SK-VKMARK19264)

but the blender-performance (performance?) 
source: bmps_27.blend
with: GPU and tiles: 400x400 : >20min
!!!the installed Radeon ProRender can not be activated!!
after changing back to the original from AMD:
with Radeon ProRender: < 30sec

JacobHeAMD, 2020-08-19:
@sosyco , is radv installed? can you please remove it if so?

jinjianrong, 2021-03-05:
Please open a new issue if the problem is still there

closed 2021-03-05 by jinjianrong