GPUOpen-Drivers/AMDVLK #82: Feature request: NV12 format support (VK_FORMAT_G8_B8R8_2PLANE_420_UNORM etc)

llyzs, 2019-01-31:
Our application requires to do some NV12 pre-processing before submitting the NV12 surface to AMF to do the H264 encoding. Currently the AMD Vulkan driver does not support any YUV format. vkGetPhysicalDeviceFormatProperties(VK_FORMAT_G8_B8R8_2PLANE_420_UNORM) reports 0 usage bits and any further image view creation crashes the Vulkan driver.

JacobHeAMD, 2019-02-01:
I tried with modified vulkaninfo, and it shows
[image]

llyzs, 2019-02-01:
Is this because the AMDGPU-PRO driver is behind the open-source one? I thought that the PRO driver is built on top of the open-source one. I am using the PRO driver 18.50 (RX480) and the YUV formats are in the unsupported formats list:

```
Unsupported formats:
FORMAT_UNDEFINED
FORMAT_R8G8B8_UNORM
FORMAT_R8G8B8_SNORM
FORMAT_R8G8B8_USCALED
FORMAT_R8G8B8_SSCALED
FORMAT_R8G8B8_UINT
FORMAT_R8G8B8_SINT
FORMAT_R8G8B8_SRGB
FORMAT_B8G8R8_UNORM
FORMAT_B8G8R8_SNORM
FORMAT_B8G8R8_USCALED
FORMAT_B8G8R8_SSCALED
FORMAT_B8G8R8_UINT
FORMAT_B8G8R8_SINT
FORMAT_B8G8R8_SRGB
FORMAT_R16G16B16_UNORM
FORMAT_R16G16B16_SNORM
FORMAT_R16G16B16_USCALED
FORMAT_R16G16B16_SSCALED
FORMAT_R16G16B16_UINT
FORMAT_R16G16B16_SINT
FORMAT_R16G16B16_SFLOAT
FORMAT_X8_D24_UNORM_PACK32
FORMAT_D24_UNORM_S8_UINT
FORMAT_ETC2_R8G8B8_UNORM_BLOCK
FORMAT_ETC2_R8G8B8_SRGB_BLOCK
FORMAT_ETC2_R8G8B8A1_UNORM_BLOCK
FORMAT_ETC2_R8G8B8A1_SRGB_BLOCK
FORMAT_ETC2_R8G8B8A8_UNORM_BLOCK
FORMAT_ETC2_R8G8B8A8_SRGB_BLOCK
FORMAT_EAC_R11_UNORM_BLOCK
FORMAT_EAC_R11_SNORM_BLOCK
FORMAT_EAC_R11G11_UNORM_BLOCK
FORMAT_EAC_R11G11_SNORM_BLOCK
FORMAT_ASTC_4x4_UNORM_BLOCK
FORMAT_ASTC_4x4_SRGB_BLOCK
FORMAT_ASTC_5x4_UNORM_BLOCK
FORMAT_ASTC_5x4_SRGB_BLOCK
FORMAT_ASTC_5x5_UNORM_BLOCK
FORMAT_ASTC_5x5_SRGB_BLOCK
FORMAT_ASTC_6x5_UNORM_BLOCK
FORMAT_ASTC_6x5_SRGB_BLOCK
FORMAT_ASTC_6x6_UNORM_BLOCK
FORMAT_ASTC_6x6_SRGB_BLOCK
FORMAT_ASTC_8x5_UNORM_BLOCK
FORMAT_ASTC_8x5_SRGB_BLOCK
FORMAT_ASTC_8x6_UNORM_BLOCK
FORMAT_ASTC_8x6_SRGB_BLOCK
FORMAT_ASTC_8x8_UNORM_BLOCK
FORMAT_ASTC_8x8_SRGB_BLOCK
FORMAT_ASTC_10x5_UNORM_BLOCK
FORMAT_ASTC_10x5_SRGB_BLOCK
FORMAT_ASTC_10x6_UNORM_BLOCK
FORMAT_ASTC_10x6_SRGB_BLOCK
FORMAT_ASTC_10x8_UNORM_BLOCK
FORMAT_ASTC_10x8_SRGB_BLOCK
FORMAT_ASTC_10x10_UNORM_BLOCK
FORMAT_ASTC_10x10_SRGB_BLOCK
FORMAT_ASTC_12x10_UNORM_BLOCK
FORMAT_ASTC_12x10_SRGB_BLOCK
FORMAT_ASTC_12x12_UNORM_BLOCK
FORMAT_ASTC_12x12_SRGB_BLOCK
FORMAT_G8B8G8R8_422_UNORM
FORMAT_B8G8R8G8_422_UNORM
FORMAT_G8_B8_R8_3PLANE_420_UNORM
FORMAT_G8_B8R8_2PLANE_420_UNORM
FORMAT_G8_B8_R8_3PLANE_422_UNORM
FORMAT_G8_B8R8_2PLANE_422_UNORM
FORMAT_G8_B8_R8_3PLANE_444_UNORM
FORMAT_R10X6_UNORM_PACK16
FORMAT_R10X6G10X6_UNORM_2PACK16
FORMAT_R10X6G10X6B10X6A10X6_UNORM_4PACK16
FORMAT_G10X6B10X6G10X6R10X6_422_UNORM_4PACK16
FORMAT_B10X6G10X6R10X6G10X6_422_UNORM_4PACK16
FORMAT_G10X6_B10X6_R10X6_3PLANE_420_UNORM_3PACK16
FORMAT_G10X6_B10X6R10X6_2PLANE_420_UNORM_3PACK16
FORMAT_G10X6_B10X6_R10X6_3PLANE_422_UNORM_3PACK16
FORMAT_G10X6_B10X6R10X6_2PLANE_422_UNORM_3PACK16
FORMAT_G10X6_B10X6_R10X6_3PLANE_444_UNORM_3PACK16
FORMAT_R12X4_UNORM_PACK16
FORMAT_R12X4G12X4_UNORM_2PACK16
FORMAT_R12X4G12X4B12X4A12X4_UNORM_4PACK16
FORMAT_G12X4B12X4G12X4R12X4_422_UNORM_4PACK16
FORMAT_B12X4G12X4R12X4G12X4_422_UNORM_4PACK16
FORMAT_G12X4_B12X4_R12X4_3PLANE_420_UNORM_3PACK16
FORMAT_G12X4_B12X4R12X4_2PLANE_420_UNORM_3PACK16
FORMAT_G12X4_B12X4_R12X4_3PLANE_422_UNORM_3PACK16
FORMAT_G12X4_B12X4R12X4_2PLANE_422_UNORM_3PACK16
FORMAT_G12X4_B12X4_R12X4_3PLANE_444_UNORM_3PACK16
FORMAT_G16B16G16R16_422_UNORM
FORMAT_B16G16R16G16_422_UNORM
FORMAT_G16_B16_R16_3PLANE_420_UNORM
FORMAT_G16_B16R16_2PLANE_420_UNORM
FORMAT_G16_B16_R16_3PLANE_422_UNORM
FORMAT_G16_B16R16_2PLANE_422_UNORM
FORMAT_G16_B16_R16_3PLANE_444_UNORM
```

mfleisz, 2019-02-14:
@JacobHeAMD Any info/update when this will be available in the pro stack? (we need support for VK_FORMAT_G8_B8R8_2PLANE_420_UNORM in the pro driver too).

JacobHeAMD, 2019-02-15:
@mfleisz , I'll look into this issue.

mfleisz, 2019-02-15:
@JacobHeAMD Thank you!

JacobHeAMD, 2019-02-25:
Hi @mfleisz , it will be supported soon but I'm not sure exact timeline when you can get the amdgpu-pro release driver with these formats enabled.

mfleisz, 2019-02-25:
Thanks for the update @JacobHeAMD

closed 2020-08-19 by JacobHeAMD