GPUOpen-Drivers/AMDVLK #83: Visual corruption in Heroes of the Storm

aufkrawall, 2019-02-03:
There are at least two issues:
Dark lines in ground texture (some sort of blending issue? Happens with radv as well, but NOT proprietary amdvlk 18.50):
https://abload.de/img/screenshot_20190203_1saj1z.png

Flashing of stars in the background of the quick match menu (doesn't happen with radv, but happens ALSO with proprietary amdvlk):
https://abload.de/img/screenshot_20190203_1xhjja.png

Flashing of the spawn location vortex (I suspect it is technically the same sort of sprites as the stars in the menu):
https://abload.de/img/screenshot_20190203_1omjp5.png

amdvlk_2019.Q1.5
dxvk 0.96.r10.ge5a06d3f

The game is free to play and the tutorial stuff etc. are skippable. Click the "Collection" tab and "Try" any hero to load a level.
You have to copy `d3dcompiler_47.dll` and `d3dx11_42.dll` from "Program Files (x86)/Heroes of the Storm/Support64" to "windows/system32" to make the game function correctly in Wine.

aufkrawall, 2019-02-09:
I also reported the issue to the radv developers:
https://bugs.freedesktop.org/show_bug.cgi?id=109394

But as it turned out on the DXVK bug tracker, this supposedly is a bug in LLVM, as it affects wined3d + radeonsi the same. It would be great if driver developers could report it to LLVM.

aufkrawall, 2019-03-15:
Both issues are still present with v-2019.Q1.8.

aufkrawall, 2019-04-23:
v-2019.Q2.2 fixed the flickering sprites, but the llvm terrain line corruption is unchanged.

JacobHeAMD, 2020-08-19:
@aufkrawall , is the terrain line corruption issue still seen with the latest amdvlk?

jinjianrong, 2020-08-19:
The line corruption seems to be an app issue

closed 2020-08-19 by jinjianrong

aufkrawall, 2023-06-19:
The issue is seen with both amdvlk 2023.Q2.2 and with RDNA3 also with Windows drivers, as LLPC apparently is now used for these cards on that platform as well.

Issue description by Rhys Perry:

> I think this is a game bug: it's using implicit derivatives when sampling textures in non-uniform control flow. In the GLSL and SPIR-V spec, the results are undefined. I think it's the same with HLSL.
> 
> Since the control flow is non-uniform and (IIRC) LLVM seems to move the operations creating the texture coordinates into it, I think it's causing discontinuities and messing up the LOD calculation.
https://bugs.freedesktop.org/show_bug.cgi?id=109394#c6

Something in RADV/ACO was fixed and it doesn't show the issue anymore (so this ticket could potentially get re-opened).